TrackMyIndoorWorkout/TrackMyIndoorWorkout #274: Use dependabot for keeping pub versions up to date

sebastiankutschbach, 2022-04-08:
Dependabot is now available for dart and the pub ecosystem:

https://github.blog/changelog/2022-04-05-pub-beta-support-for-dependabot-version-updates/

Even though it is still beta - do we want to give a try?

MrCsabaToth, 2022-04-11:
Maybe I'm too careful and we can just turn this on, and only case it barks too much we could defer it after #227. But we can try it even before that. I like dependabot.

closed 2022-05-30 by MrCsabaToth

MrCsabaToth, 2022-10-11:
I leverage dependabot for for my web/node/JS and also Java projects and I like it. In our case it seems like that for Flutter it wouldn't warn about security vulnerabilities, but essentially only newer available packages? I very often run "pub outdated", I think that's what they also do on their end and parse the output of that. Flutter CLI is pretty capable in itself. I'd really want to enable this, maybe we can ticket it, but I don't want to enable it until I/we can get to the https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/227 ticket, and upgrade to a community forked version of flutter_blue and get up-to-date with bluetooth_enable and also the permission_handler. These are all related to each other and Android 12. Back in the old days I was lenient and all packages were "any" version, essentially the latest. With sound null safety shenanigans and Flutter embedding API things and such for a long time I point to a specific version so all potential developers would see the exact same packages as I.
[https://opengraph.githubassets.com/baa332c7a40f49abee827f31af34c7d9e4ddc463f4bf18d59568e227de34bc63/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/227]<https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/227>
Be compatible with Android 12 permissions and build on M1 Mac · Issue #227 · TrackMyIndoorWorkout/TrackMyIndoorWorkout<https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/227>
This will need the upgrade of these plugins at least: bluetooth_enable flutter_blue permission_handler First as a quick start I&#39;ll: depend on bluetooth_enable_fork instead of bluetooth_enable, ...
github.com
As I mention flutter_blue, possibly I'll try boskosg's flutter_blue_plus or other fork, I even have a branch half way there. See this ticket as well:
https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/210
[https://opengraph.githubassets.com/e4d56b82da0e57ecd391ba72e06f86c2b269e12344fb57a94d5a0c962ddfdec8/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/210]<https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/210>
Look for another BLE library · Issue #210 · TrackMyIndoorWorkout/TrackMyIndoorWorkout<https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/210>
https://pub.dev/packages/flutter_blue is very inactive, the maintainer openly stated that since the library is working fine for his specific use he probably won&#39;t bother much with updating. The...
github.com
Also note that a proper new flutter_blue fork could solve foreground processing out of the box (or use one of the mentioned foreground processing helper plugins)
https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/16
[https://opengraph.githubassets.com/e892906fd6117cbd5f6dfaac95ee864986b986064d288ae13c23320c3fa9f662/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/16]<https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/16>
App goes cuckoo when the screen gets locked during exercise · Issue #16 · TrackMyIndoorWorkout/TrackMyIndoorWorkout<https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/16>
What happens is once the phone is unlocked, all the measurements which happened since the screen was locked starts to flow in and the app is in a craze for many seconds processing all the incoming ...
github.com
I cannot wait to get to these, but first I want to tick off some long promised features for users. I did the zone default changes, 5-6-7 zone support and configurable colors. (Let me know BTW if you have anything to modify or contribute anywhere.)
I already have average speed display on the leaderboard display. I'll also add a manually configurable pacer.
I'm also adding ability to omit GPS calculation and output from exports and uploads. That will open up the possibility to support stair trainers and step trainers (they are part of the FTMS standard, I cannot interpolate GPS for those onto a flat running track 🙂).

These are semantic and UX features, but the above mentioned permission and Bluetooth data processing tickets are more serious, technology stack related ones. I cannot wait to "unleash" dependabot after that. Oh BTW, I'm waiting for all of this, because dependabot will quickly find out that I pinned two packages and those hold behind some other transitive dependencies unfortunately.

https://github.blog/changelog/2022-04-05-pub-beta-support-for-dependabot-version-updates/

commit f9143333 (2022-05-23): Fixes #274, let's see how it flies