TrackMyIndoorWorkout/TrackMyIndoorWorkout #310: Strava login error on new installs

MrCsabaToth, 2022-06-30:
Dan Walkes has a new install and the OAuth to Strava seems to be broken:
[image]

MrCsabaToth, 2022-06-30:
It seems like the Strava OAuth is mingled with the Google OAuth, maybe somehow the dormant Google Fit integration merge interfered and broke the Strava (or other?) integrations?

MrCsabaToth, 2022-06-30:
Ref https://github.com/flutter/flutter/issues/66721

dwalkes, 2022-06-30:
Thanks @MrCsabaToth !
I just removed myself from the beta program, re-installed the non-beta and I on longer see this issue.  So it's either something specific to the beta or something about my previous install.

MrCsabaToth, 2022-06-30:
Actually the Google Fit integration is still separated in a separate feature branch. I suspect that this is caused by breaking changes in `url_launcher`, namely that it is now working as in-app WebView, inside the Flutter sandbox. However I don't know how that leads to this Google OAuth messup/mixup, what a shitshow!

MrCsabaToth, 2022-07-01:
Maybe the Google OAuth comes into play because Dan may use Google OAuth for logging in to Strava. So there's a chained OAuth going on: I kick off an OAuth for Strava and there's an OAuth within that for Google.

MrCsabaToth, 2022-07-01:
I'll reopen until confirmed by Dan

commit 0f8eca32 (2022-07-01): Attempting to fix #310 by enforcing external browser instead of WebView for OAuth
closed 2022-07-01 by MrCsabaToth