TrackMyIndoorWorkout/TrackMyIndoorWorkout #318: v3 calorie -> power logic for Sch 170 / 270 / 570

MrCsabaToth, 2022-07-24:
Extra follow-up of #305

Revise the calorie -> power logic which is currently influenced by `extra_mult` (https://github.com/ursoft/connectivity-samples/blob/da91a864a0ba2321aadc44d0ff01caf8c36d3594/BluetoothLeGatt/Application/src/main/java/com/example/android/bluetoothlegatt/BluetoothLeService.java#L202) to use this `coeffs` (https://github.com/ursoft/ANT_Libraries/blob/e122c007f5e1935a9b11c05e601a71f2992bad45/ANT_DLL/WROOM_esp32/WROOM_esp32.ino#L526)

According to Yuri Pershin (#240):
"extra_mult is 2nd algo, calculation by 'calories' and resistance with manual calibration - no power meter, only my feelings. Was very inaccurate (maybe, we got no calories at all in this field) - about 15% of power accuracy. coeffs is third algo (by resistance and cadence with power meter calibration). On the third iteration, I used the Favero Assioma Duo to calibrate each Schwinn resistance level at almost any cadence. I got about 5% of power accuracy, if you ignore the transients and flywheel inertia. As you can see, all the functions are not always linear in my case. I do not know why."

So the newest heuristics are backed by measurement of a Favero Assioma Duo power meter pedal.

MrCsabaToth, 2024-02-18:
Steve Hawkins was experiencing the app reported value differences compared tot he bike console.

MrCsabaToth, 2024-02-24:
Steve Hawkins and Michael Rife are both using Schwinn 170 and noticed the number not matching up with the console.

tkordenbrock, 2024-03-05:
Using the references above, I have implemented the new power logic based on resistance and cadence.  I am testing it with my Schwinn 270 this week and then I will create a PR for your review.

MrCsabaToth, 2024-03-05:
> Using the references above, I have implemented the new power logic based on resistance and cadence.  I am testing it with my Schwinn 270 this week and then I will create a PR for your review.
> 

This would be amazing! All Schwinn 170 / 270 / 570 owners could benefit.

MrCsabaToth, 2024-03-06:
> Using the references above, I have implemented the new power logic based on resistance and cadence. I am testing it with my Schwinn 270 this week and then I will create a PR for your review.

Let me know if you need any consultation we can have a video chat. On my app's side I'd expect the new v3 way would probably bring in different conversion table `resistancePowerFactor`. Maybe it could handle even more bytes. The v2 current implementation only estimates the watts from the difference of calories, and then estimates the speed from the watts, and then the covered distance can be calculated from the speed, however there are multiple levels of estimations. The v3 may not change that, but it might be closer to the console's readings out of the box.

tkordenbrock, 2024-03-06:
Thanks for the offer.  I brought in `calcPowerNewAlgo` from https://github.com/ursoft/ANT_Libraries/blob/e122c007f5e1935a9b11c05e601a71f2992bad45/ANT_DLL/WROOM_esp32/WROOM_esp32.ino#L525.  That is the bulk of the change.

You're right.  v3 is much closer to the console without any adjustments.  Using v2 I need to set the power factor to 133%.  That gives me a speed and distance that are very close to the console.  The power still gets recorded at the base value (the power factor is not applied). 

I am also fiddling with the cadence mix-in.  I have two issues:
* Occasionally, I get some bad cadence data.  I guess it is a glitch in the Bluetooth connection, but I don't know for sure.  The revolutions value goes from some reasonable series of values to something huge for a few seconds and then returns to normal.
* I am also trying to figure out why the x70 `stubRecord` method resets and returns an empty record every ~63 seconds.

MrCsabaToth, 2024-03-07:
@tkordenbrock Reply part 1

The Cadence Mixin was first introduced, because both the Cycling Speed and Cadence Sensor (https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.csc_measurement.xml) and the Power Meter profile (https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.cycling_power_measurement.xml) does not provide cadence data directly. Unfortunately they only provide cumulative crank revolution count and crank even time. This puts the burden on us to gather the revolution data in a LIFO queue (`ListQueue` in Dart) and compute the cadence using this two information.

Another catch is that both of these two data points can overflow independently!
1. "cumulative revolution" is only 2 bytes unsigned, so after 65535 revolutions it gonna overflow to 0. So while looking at the crank count and if we se a decrease we need to adjust by uint16max, because revolution count monotonically increases.
2. What's even worse is the last crank time. Because the standard I guess wants to provide high precision data so you don't have to accumulate too many data points in the queue, the supplied 2 byte unsigned value has a 1/1024 resolution. That's why it overflows every 64 seconds.

My mixin by default accumulates 10 seconds of data, because let's say you climb hard and pedal only 30 rpm, then within 10 seconds you only have 5 pedal strokes. The downside is that the mixin can have a 10 second delay regarding cadence. So maybe 10 seconds is too much, that could be just 5 (which is in the case of X70).

This overflow can happen at an independent time from the revolution overflow, but similarly time is also a monotone increasing measure (I wish I could pause or turn back time), so when the code detects a decrease, then it adjusts by the overflow span.

Looks like the adjustment happens differently. Since sometimes the code computes the distance from the revolutions (combined with the wheel circumference or the cadence) the number of overflows is actually remembered there. As for the time overflow that adjust only happens at the time of the computation from the queued values.

The X70 code uses the Cadence Mixin because that byte stream also contains cadence data, just somewhat different defaults. In the `SchwinnX70` class you can see `initCadence(5, 64, maxUint24);`. So we accumulate for 5 seconds, the time overflows in 64 seconds still, however the revolution counter is 3 bytes instead of just 2, so that takes longer to overflow.

If sometimes you see crazy cadence, maybe this part is not matching up. Maybe it's not unsigned 3 bytes but signed, or something else, some bytes misaligned? But only occasionally? Try to see if this happens every 64 seconds?

The revolution count should take a long time to overflow, since 65536 revolutions would take 655 minutes if you spin with 100 rpm. But since some cycling computers persist the counter, it can happen still.

MrCsabaToth, 2024-03-07:
@tkordenbrock Reply part 2
Regarding stubRecord reset:

```
    if (lastTime < 0 || duration.inSeconds > 63) {
      lastTime = time!;
      lastCalories = calories!;
      lastPower = 0.0;
      lastRecord = null;
      return RecordWithSport(sport: sport);
    } else if (time == lastTime) {
      return lastRecord ?? RecordWithSport(sport: sport);
    }
```

I see I introduced this knowingly with https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/commit/2d2aee88c138a106f3e8c0341dac7bdddb5f6e1a

Unfortunately I don't remember what triggered this. What I see in https://github.com/ursoft/connectivity-samples/blob/da91a864a0ba2321aadc44d0ff01caf8c36d3594/BluetoothLeGatt/Application/src/main/java/com/example/android/bluetoothlegatt/BluetoothLeService.java#L175

```
    public int currentCadence() {
        long t = System.currentTimeMillis();
        if(t - mLastCalcCadTime < 1000 && mLastCalcCad != 0)
            return mLastCalcCad;
        if(t - mLastRxTime > 5000)
            return 0;
        int cadPointer = mCadPointer;
        int maxRot = mCadRotations[cadPointer % 600];
        int minRot = maxRot;
        long timeDuration = 0;
        while(timeDuration < 60000) {
            cadPointer--;
            if(cadPointer < 0) cadPointer += 600;
            long tryTime = mCadTimes[cadPointer % 600];
            if(tryTime == 0 || t - tryTime > 61000) break;
            timeDuration = t - tryTime;
            minRot = mCadRotations[cadPointer % 600];
            if(maxRot < minRot) maxRot += 0x1000000;
            if(timeDuration > 15000 && maxRot - minRot > 20) break;
        }
        if(timeDuration == 0) mLastCalcCad = 0;
        else mLastCalcCad = (int)((60000.0 / timeDuration) * (maxRot - minRot) + 0.5);
        mLastCalcCadTime = t;
        return mLastCalcCad;
    }
```

At the beginning it has some shortcut ifs, one of them is `if (t - mLastRxTime > 5000)`. Maybe there's similar code somewhere else in the original code (for example `bool bSchwinnStopped = (int(nowTime - m_crankInfoCache.m_curSchwinnPowerIncomeTm) > 2000);`). Seemingly my rule is that after 64 seconds of no data I return back an empty record. The receiver code somewhere in the `FitnessEquipment` accounts for and handles empty Records as a signal.

Looking at the code I see some redundant code both with power and cadence related code, some of which is applied in the Mixins as well, so there's room to improve on separation of concerns for sure. BTW in the future I will refactor the code to modularize the `FitnessEquipment` logic better, but that's not in the immediate future, but I'm planning to write a version of the app which will be for local indoor races. That may push me to extract some of the logic into a library, which will be good for modularization and internal APIs overall, however will have a lot of code churn because lot of the logic is mixed (such as BT related, persistence). But this won't happen soon so you won't conflict with anything.

tkordenbrock, 2024-03-07:
Thanks for the detailed information.  This is very enlightening.  I didn't recognize the `time! / 1024` as a conversion of pseudo-milliseconds to seconds.   I'm surprised that the time value has such high resolution when the other values are low resolution.

The smoothing of the cadence data is further complicated by the fact that the revolutions is an integer value.  I am not an experienced cyclist so my RPMs avg 75-80.  Since the cadence mix-in sees ~2 data messages per second, that means there are a lot of repeated revolutions values in the list depending on when the packet was sent.  I'm still thinking about this too.

MrCsabaToth, 2024-03-08:
Since revolution is integer, that's why I collect 5 seconds of data (as the mixin configured) for X70. But theoretically the high resolution timer is so we don't have to queue as much. Try to keep using the mixing with the desired (chosen by you) parameters. There could be some bug in any of the processing code which can cause the occasional messup?

tkordenbrock, 2024-03-15:
Since implementing the v3 power algorithm, I have been tracking two unrelated cadence problems.

1) bogus cadence data from the bike

I thought this was rare, but I have seen it happen 3 times in the last week.  The packets from the bike look like this.

```
2024-03-09 06:17:40.353668,INFO,FITNESS_EQUIPMENT,_listenToData loop,key 8209 byteListPrep [17, 32, 0,  64, 36, 235,   0, 142, 191, 118, 160, 134, 17, 11, 0, 0, 7]
2024-03-09 06:17:40.438278,INFO,FITNESS_EQUIPMENT,_listenToData loop,key 8209 byteListPrep [17, 32, 0,  96, 36, 235,   0, 194,  27, 119,  96, 228, 17, 11, 0, 0, 7]
2024-03-09 06:17:40.528902,INFO,FITNESS_EQUIPMENT,_listenToData loop,key 8209 byteListPrep [17, 32, 0, 128, 36, 235,   0, 250, 125, 119, 208,  69, 18, 11, 0, 0, 7]
2024-03-09 06:17:40.654893,INFO,FITNESS_EQUIPMENT,_listenToData loop,key 8209 byteListPrep [17, 32, 0, 160, 36, 235, 128, 189, 222, 119,  96, 166, 18, 11, 0, 0, 7]
2024-03-09 06:17:40.732009,INFO,FITNESS_EQUIPMENT,_listenToData loop,key 8209 byteListPrep [17, 32, 0, 192, 36, 235,   0, 202,  65, 120,  96,   8, 19, 11, 0, 0, 7]
2024-03-09 06:17:40.850709,INFO,FITNESS_EQUIPMENT,_listenToData loop,key 8209 byteListPrep [17, 32, 0, 224, 36, 235, 128, 249, 159, 120,  96, 103, 19, 11, 0, 0, 7]
2024-03-09 06:17:40.964431,INFO,FITNESS_EQUIPMENT,_listenToData loop,key 8209 byteListPrep [17, 32, 0,   0, 37, 235, 128,  41,   2, 121, 208, 200, 19, 11, 0, 0, 7]
```
The Schwinn x70 protocol puts the total revolutions in bytes 4,5,6.  Converting the total revolutions bytes to an integer, you get this:
```
60196.0
60196.0
60196.0
8448804.0
60196.0
8448804.0
8448805.0
```
That single bit flip in the msb causes havoc in the RPM calculation.  When the bogus count is at the tail of the cadence mix-in queue and the head of the queue is correct, you get a huge RPM value which gets capped at `maxBytes`.  When the bogus count reaches the head of the queue and the tail of the queue is correct, you get a negative RPM value.

I have decided that this is a problem with my bike (or maybe all the x70 bikes).  I added checks in `addCadenceData()` to throw away any packets with a revDiff < 0.0 or revDiff > 5.0.  In both cases, the timestamp gets updated.

2) RPM swings of ~5 around the actual RPM

I have also observed some swings in the RPM when the console is reporting a consistent 75 RPM.  The swings happen when the following combination occurs:
1) the cadence data at the head of the queue has a time very close to the actual time that the revolutions counter clicked over
1) the cadence data at the tail of the queue occured at the beginning of the `ftmsDataThreshold` but isn't reported until the end of the `ftmsDataThreshold`

At my typical 75 RPM cadence, the worst case is when the queue head time is `X` and the queue tail time is `X+5.28` (6 * 480ms).  

75 RPM is 6 complete revolutions in the 5 second cadence mix-in window.  Rounding the 5 second window up to the nearest `ftmsDataThreshold` is 5.28 seconds.  6 revs / 5.28 sec * 60 sec/min = 68.18 RPM instead of 75 RPM.

The best solution I could come up with is to shorten the `ftmsDataThreshold` to 50ms.  That's small enough that each cadence event from the bike gets processed as it arrives.  This probably uses more battery due to higher CPU usage, but I don't know how to test that.

A side effect of shortening the `ftmsDataThreshold` is a huge difference between Moving Time and Elapsed Time.  With a short timer, `processRecord()` can be called with the same record twice which leads to a `dTMillis` of 0.  In this case, moving time doesn't advance.  I think this happens because the heart rate packet get delivered separately.  With a 480ms window, the heart rate packet gets combined with the cadence packets.  At 50ms, each packet gets processed individually.  I fixed this by ignoring duplicate records.

I have separate PRs about ready for all these changes.

tkordenbrock, 2024-03-15:
Shoot!  I didn't run the PRs through the flutter code formatter.  I'll do that.

Also, I'm a C/C++ programmer by trade, so this probably isn't good dart code.  Let me know if I can improve it.

MrCsabaToth, 2024-03-15:
> Shoot! I didn't run the PRs through the flutter code formatter. I'll do that.
> 
> Also, I'm a C/C++ programmer by trade, so this probably isn't good dart code. Let me know if I can improve it.

No worries, but the CI won't accept any PR which doesn't pass the formatting and more importantly the analyze phase. See the two commands in the https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout?tab=readme-ov-file#contribution-rules the analyze is the shorter command.

Once you do that and commit your changes your PRs should refresh. Looks like you have multiple branches so do the changes for all of those branches, and make sure the same changes otherwise they can collide with each other.

For the first glance we'll need to have several iterations, so I commend your patience in advance.

MrCsabaToth, 2024-03-16:
BTW I realized I could have been misleading with mentioning Git Flow and Git Town in the README. It's up to you, but you don't necessarily need to open 3 feature branches for your contribution. I won't squash the commits (to preserve history better) when I'll merge, I'm OK if your PR contains several (even many) commits. So if you like you can merge everything into one branch - still having all the commits.

tkordenbrock, 2024-03-16:
OK.  I was thinking that the three PRs aren't really related so if you wanted to accept the v3 power calculation and not the others because they are too specific to my bike then this would keep them separate.  If you want them all together, it easy to cherry pick them into a single branch.  That's what I'm doing to test all these changes.

MrCsabaToth, 2024-03-17:
Maybe you are right. We can keep it that way as well actually

MrCsabaToth, 2024-05-09:
@tkordenbrock I haven't forgot about you, but I'm busy with phone changes and conferences. However I had some changes recently and I'm familiarizing myself with some code which is closer to your PRs. Next two weeks will be super busy, but I'm still really want to continue with your contributions after that.

MrCsabaToth, 2024-12-26:
Hey @tkordenbrock, merry Christmas and holidays, I took the time to push some PRs to your fork, please merge them into your fork's branches and then we can merge your PRs.

Some formatting and analyzer changes were for the CI to accept the PRs. The only notable change is that instead of setting the FTMS Data Threshold to 50ms I made it configurable, because the 50ms is so drastic. I fear that decreasing that will result you in having too many Records recorded. 50ms is even lower than Flutter's generic 60ms UI refresh target.

With the configurability you can decrease it if you want, whereas for others it'll stay the standard 480ms (which is half a second minus a little for processing).

tkordenbrock, 2024-12-27:
Happy Holidays to you too!

I have cherry picked your commits to my combined testing branch and built successfully.  I'll test the new build tomorrow morning when I go for a ride.  I can merge your PRs after that.

The 50ms threshold fixes the large swings in RPM described in the second half of https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/318#issuecomment-1998831084.  Now that this is configurable, I can try other values.

I have also reduced the length of the cadenceData list from 5 to 3 (ie. `initCadence(3, 64, maxUint16)`).  That is still on a local branch because I can't prove that it is useful to other Schwinn X70 riders.  I think we discussed this previously but can't find it right now.

MrCsabaToth, 2024-12-28:
@tkordenbrock Thanks for your contribution. The #538 will make the sliding window configurable, at which point the default will jump back (for Schwinn) from 3 to 4. So you'd need to decrease that back one more time.

closed 2024-12-28 by MrCsabaToth