TrackMyIndoorWorkout/TrackMyIndoorWorkout #337: Stages SB20: display gear information

javimoya, 2022-09-04:
The stages bike is transmitting the gear information. Would be great to display that information.
Thanks

MrCsabaToth, 2022-09-04:
This would be great, sub tasks:
* Add this field to the Record entity and DAO
* Create schema migration to add the field to the database
* Enable this metric for the Indoor Bike FTMS protocol (and other capable FTMS protocols as well), see https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/bea03d863fa47a471c10cbe18a4d65475737f10b/lib/devices/device_descriptors/indoor_bike_device_descriptor.dart#L63 (indoor bike), https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/bea03d863fa47a471c10cbe18a4d65475737f10b/lib/devices/device_descriptors/cross_trainer_device_descriptor.dart#L63 (cross trainer), https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/bea03d863fa47a471c10cbe18a4d65475737f10b/lib/devices/device_descriptors/rower_device_descriptor.dart#L91 (Rower), inclination or ramp angle for treadmill https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/bea03d863fa47a471c10cbe18a4d65475737f10b/lib/devices/device_descriptors/treadmill_device_descriptor.dart#L60
* Enhance device codes so the new sensed metric is wired up and persisted. This is a non cumulative metric
* Add the metric to FIT export
* Add the metric to TCX export if possible
* Add the metric to migration CSV export
* Add if possible to JSON export (Under Armour)
* Find a place on the UI to display during measurements. This could also mean intelligent logic which emits if not present, same should be done for HR and cadence. The post-activity details already has an omit logic
* Find a place on the landscape measurement layout. Due to the possibility of combinations this will mean of more dynamic logic, which will be possible once the RiverPod UI update is in place (#127, #333). But once that's done the best would be if order was configurable #179

MrCsabaToth, 2022-09-05:
Also related #303, that's a cumulative metric

MrCsabaToth, 2024-02-27:
Since the ticket is for display only, then #458 is a duplicate of this ticket.

MrCsabaToth, 2024-04-29:
Releasing this with beta 155 (in some hours hopefully when it passes Google checks and scrunity), the resistance level can be turned on for display in the preferences.

closed 2024-04-29 by MrCsabaToth

MrCsabaToth, 2024-05-07:
@javimoya Were you able to test this by any chance?

MrCsabaToth, 2024-05-07:
[image]