TrackMyIndoorWorkout/TrackMyIndoorWorkout #372: Heart rate updates too infrequently

pockethook, 2022-10-28:
I've been hooking my Polar H7 up to the app and the HR appears to update quite infrequently. I've been doing a lot of rides where I attempt to keep a constant power and cadence and I'll often see a completely flat HR for a minute only to have it then jump by 4 in an instant.

pockethook, 2022-10-28:
I did the first 30 minutes of my ride today on your app:
[image]
[image]

I recorded my HR on the Polar app for the last 30 minutes of the ride:
[image]

MrCsabaToth, 2022-10-29:
Thanks for the report. Some questions:
1. Which build are you using?
2. Was this always like that or only after some update the app started to act like that?
3. Are you pairing the HRM directly to my app or the app receives the hr reading transitively via the fitness equipment console (so the HRM is paired to the console)? This matters so the connection might be laggy between the app or the console?
4. Does the reading act like that even before the workout start? (Do some squats and see if it reacts)
5. Could you record a short workout?

pockethook, 2022-10-29:
1. I'm using the latest version in the play store, version 1.1.125, build 125
2. I haven't been using the app for very long (a week or so) so it's always been like this
3. I'm pairing the HRM directly to the app, my HRM isn't compatible with the fitness equipment console unfortunately
4. I will check this in a few hours and report back
5. I have some logs from a previous session as well as a .fit file for that session: 
[LogExport27102022_200426_PM_all_[14].zip](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/9894196/LogExport27102022_200426_PM_all_.14.zip)

pockethook, 2022-10-29:
I see similar behaviour before I start recording the workout.

pockethook, 2022-10-29:
Sometimes I'm able to _force_ the app to update my HR by moving the HRM around or even disconnecting one of the metal clips and reattaching it, though the latter can also result in zeros being displayed.

MrCsabaToth, 2022-10-30:
The logs help and depict the bug. The FIT files are not helpful because they are binary and thus hard to debug, I only use them for upload or compatibility with other systems. But the logs are good.
BTW I see you use Precor Spinner Chrono Power or compatible bike. What fitness machine is it exactly? If it's a Chrono Power then the console supposed to be able to pair with you HRM. Any Bluetooth HRM.

MrCsabaToth, 2022-10-30:
But now it's my turn to debug

MrCsabaToth, 2022-10-31:
So this is definitely a case of mutability: the value which is hold for too long happened in the past, and comes back overriding the fresh HR reading for some reason. I haven't located the exact position yet. `FitnessEquipment` is the main class of interest in this case. When the recording starts a `FitnessEquipment` is instantiated with two main components: the `BluetoothDevice` provided by the `flutter_blue` package and a `DeviceDescriptor` which will tell which bytes correspond to what in the data packets coming in. On top of that `FitnessMachine` pays attention to the extra paired `HeartRateMonitor` and also now additional sensors like cadence, power, and speed sensors. The HRM is a dedicated sensor.

Here is the current relevant logic: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/181bcdc15c43bf309f4980de52b2c56d2f6dc879/lib/devices/gadgets/fitness_equipment.dart#L957

Here if there's no HR provided by the "main" equipment already the HR coming from the HRM is set into the return packet. Right after that comes an extra logic which would supplement the last known positive HR in case a special setting is set to bridge gaps in the reading. But that latter code wouldn't overwrite a positive HR reading coming from a HRM set right before it.

Right after this there's also an upper limit HR logic which was developed to deal with some unruly readings as well. I think that is not relevant, because if there was accidentally a too low HR limit set we would observe a different graph.

I see in the logs, that the HRM provides current HR readings, but somehow during the processing of the values it must be overwritten with older values. It's strange how sometimes it still changes.

There's also a logic here https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/181bcdc15c43bf309f4980de52b2c56d2f6dc879/lib/devices/gadgets/fitness_equipment.dart#L730 which makes sure that the HR stays current when the workout is paused (and thus a shortuct path returns to the UI, but we want to be sure the HR in that is recent). But I think pausing didn't happen in our case either.

Due to the fact of faulty HRMs, paused states and multiple data sources the logic has several parts and values are mutated. I already made refactors during the CSC and power meter support to decrease some of the hard to follow mutation coming back from `lastRecord`. Once I progress with the River Pod UI update branch I will try to keep this in mind and we'd need to revise all the data sources and the data flow anyway.

Most probably the stale HR reading haunts (pun intended due to Halloween time) from a `lastRecord`.

pockethook, 2022-10-31:
[This](https://www.precor.com/sites/default/files/304563-101%20rev%20B_Chrono_Console_Getting_Started%20Guide_072317_ENU_Legal%20Size_BW.pdf) is the bike I'm using, unfortunately I can only get HRMs to connect to the console over ANT+.

pockethook, 2022-10-31:
Yes, I don't think I paused the workout at all.

pockethook, 2022-10-31:
Happy Halloween :ghost:

pockethook, 2022-10-31:
I've hooked my HRM up to [nrfconnect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) and can see that the first byte is always 0x16 (assuming the HRM is properly attached), it sends out packets roughly every second and the payload size is either 4 bytes or 6. 0x16 means the 'Heart Rate Value Format is set to UINT8', and 'One or more RR-Interval values are present.'

If the RR interval bit is set then this line

https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/181bcdc15c43bf309f4980de52b2c56d2f6dc879/lib/devices/gadgets/heart_rate_monitor.dart#L43

assumes that that there's a single RR interval UINT16 value, but there can be multiple according to the [spec](https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.heart_rate_measurement.xml). If there are multiple this will return false:

https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/181bcdc15c43bf309f4980de52b2c56d2f6dc879/lib/devices/gadgets/heart_rate_monitor.dart#L60
 
Could this be related to what's going on?

MrCsabaToth, 2022-10-31:
> [This](https://www.precor.com/sites/default/files/304563-101%20rev%20B_Chrono_Console_Getting_Started%20Guide_072317_ENU_Legal%20Size_BW.pdf) is the bike I'm using, unfortunately I can only get HRMs to connect to the console over ANT+.

Oh! I have this bike too (in fact this bike was the original one which inspired me to write the app because the Wahoo app didn't pick up some metrics, that's why the icon of the app is an artwork of the Chrono Power). I don't know why I was in the belief that the HRM is paired with Bluetooth. I use wrist based Wahoo TICKRs and a Polar Verity Sense and both are dual BLE/ANT+ capable, so that's why I can pair with it. I'll find this bug though.

MrCsabaToth, 2022-10-31:
> I've hooked my HRM up to [nrfconnect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) and can see that the first byte is always 0x16 (assuming the HRM is properly attached), [...]

Oh, good that you mentioned this! Turns out that I'm interpreting the HRM data with multiple RR intervals wrong.
(BTW the first byte which is the flag (0x16 in this case, or decimal 22) can change without the data length change because two bits in the flag are contact status bits. That's kinda unusual, but I just mask out those bits to get to the ones commending the content length.)

I haven't had RR-Interval data streams yet. First glance of https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.heart_rate_measurement.xml showed simply `uint16` data for the flag C4. However it's a series of `uint16`s.

I have a `canMeasurementProcessed` pre-processing function: it just tells if the whole data packet conforms with what the feature flag suggests. If it does not I simply yeet the packet. Some fitness machines botch the FTMS protocol, or sometimes send fluke packets. The Precor Spinner Chrono Power sometimes it sends stuff like `[36, 8, 0]`, which I just yeet because the whole protocol is proprietary and I'm not sure what those short packets mean.

Unfortunately right now I yeet your HRM packets which have multiple RR data (there can be more than one). BTW in the future I really want to develop a HRV #191 Heart Rate Variability measurement feature, where these RR intervals will supply the required source data.

So now I know what's the problem and soon will devise a fix.

MrCsabaToth, 2022-10-31:
Not closing until @pockethook confirms the fix in the release

MrCsabaToth, 2022-10-31:
I deployed a new beta build 126 and it went through Google's check and now it's distributed. Please check if the fix solves your issue.

MrCsabaToth, 2022-11-01:
[2710202217-18.zip](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/9907654/2710202217-18.zip)

pockethook, 2022-11-01:
Brilliant! I tested the beta build through Google Play and it's working really smoothly. Thank you so much!

MrCsabaToth, 2022-11-01:
https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout.github.io/commit/d194ca5bca6fc1bd8e7436839bef8b4d5e230e7b

MrCsabaToth, 2022-11-01:
Happy to hear, closing

commit 060d07cf (2022-10-31): Fixes #372 by allowing multiple RR interval values (which happens with Polar H7)