TrackMyIndoorWorkout/TrackMyIndoorWorkout #380: Debug Garmin and Fitbit upload APIs

MrCsabaToth, 2022-11-30:
People are asked about these integrations but those ecosystems are closed off from uploads. Garmin has a workaround of manual upload via the website, but not the app.

Latest one was David Rudio asking for Garmin. We do know upload is possible since select few applications have the privilege to upload, so there is indeed an API. Example: (David Rudio mentions) SOLE treadmill can upload to Garmin.

We could peek into the exchange, but we'd need some enhanced MITM attack for that: set up a special wifi the treadmill connects to and being able to MITM into their TLS tunnel. Then we would obtain the secret API endpoint.

Note that this may not guarantee success because the endpoint may still fortified on the back-end but then we could try at least.

Note that Kinomap (500K+ download in Google Play store and even more in Apple store) even turned down Garmin's offer to caugh up $20K dough to open up that undocumented API for them. https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/5#issuecomment-1332410218 https://support.kinomap.com/hc/en-us/articles/115004861786?page=1#comment_360010812671

MrCsabaToth, 2022-12-03:
Ref #355 for FitBit

MrCsabaToth, 2023-04-20:
Péter Kelemen also requests Garmin upload

MrCsabaToth, 2023-12-28:
Check this out, tapiriik service has GarminCOnnect support supposedly: https://github.com/cpfair/tapiriik/tree/master/tapiriik/services/GarminConnect
It also seem to have support for NikePlus, RunKeeper, TrainerRoad and other services!

MrCsabaToth, 2024-01-08:
Ref Adidas Runtastic integration: #477

MrCsabaToth, 2024-01-08:
Ref Easics Runkeeper integration: #478

anthonywilliams, 2024-02-02:
the fitbit API is here: https://dev.fitbit.com/build/reference/web-api/activity/create-activity-log/

MrCsabaToth, 2024-02-02:
> the fitbit API is here: https://dev.fitbit.com/build/reference/web-api/activity/create-activity-log/

That only allows total uploads and I didn't find a way so far to upload time series somehow https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/355#issue-1402394170

MrCsabaToth, 2024-02-03:
> the fitbit API is here: https://dev.fitbit.com/build/reference/web-api/activity/create-activity-log/

But maybe I'm gonna add the integration with cumulative data (no detailed time series). Do you see maybe any way for time series data? BTW I looked at Google Health API and it's also not clear there how I could upload time series. But maybe final stats would be enough for users.

anthonywilliams, 2024-02-05:
I didn't see an API for uploading time series data, but overall data would be a useful feature on its own.

MrCsabaToth, 2024-02-06:
I'll try to get the FitBit simple upload done, because I'm using a FitBit watch (Google Pixel) every day for a health study (https://blog.google/products/fitbit/new-quest-fitbit-study-metabolic-health/). Before that I was alternating Samsung Galaxy Pro 5 on my right hand and the Pixel Watch, now I'm wearing the Pixel all the time. Besides also the SUUNTO 7.

MrCsabaToth, 2024-02-23:
@anthonywilliams In opened up #355 for your suggestion. This current ticket is to reverse engineer the FitBit and Garmin uploads to unlock a time series way. Until then I'll implement summary uploads in the realm of #355