TrackMyIndoorWorkout/TrackMyIndoorWorkout #399: Package dependendency problem

gilcu2, 2023-01-15:
Hello, I am trying "flutter pub get" on develop branch and have dependency problems:
```
Because every version of integration_test from sdk depends on vector_math 2.1.2 and track_my_indoor_exercise depends on vector_math ^2.1.4, integration_test from sdk is forbidden.
So, because track_my_indoor_exercise depends on integration_test from sdk, version solving failed.
Running "flutter pub get" in track_my_indoor_workout...                 
pub get failed (1; So, because track_my_indoor_exercise depends on integration_test from sdk, version solving failed.)
```
My flutter doctor says:
```
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.3.10, on EndeavourOS 6.1.6-arch1-1, locale en_US.utf8)
[✓] Android toolchain - develop for Android devices (Android SDK version 33.0.0)
[✓] Chrome - develop for the web
[✓] Linux toolchain - develop for Linux desktop
[✓] Android Studio (version 2021.3)
[✓] Android Studio
[✓] IntelliJ IDEA Community Edition (version 2022.3)
[✓] Connected device (2 available)
[✓] HTTP Host Availability

• No issues found!

```
How can I fix it?
Thanks

MrCsabaToth, 2023-01-15:
This is because you are on the stable channel and Flutter SDK has many pinned version dependencies inside. Take a look at this issue I opened about it https://github.com/flutter/flutter/issues/114199#issuecomment-1294263848. I'm on stable channel as well. To overcome this you need to execute `flutter update-packages --force-upgrade`. This has a side effect though: it modifies the SDK, so next time you upgrade Flutter version first you need to undo those changes (since the Flutter SDK is essentially a GitHub repo clone), then upgrade Flutter and re-execute `flutter update-packages --force-upgrade`. The other venue is to switch to the beta channel.
I'll modify the README to contain this extra information. It's also good guide to look at what the CI build does: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/develop/.github/workflows/flutter_test.yml
That also shows that if you want to compile the source you'd need to establish a dummy `secret.dart` with dummy API key values and secrets.

MrCsabaToth, 2023-01-15:
I reorganized the README partially to reflect all of this. Let me know if it's satisfactory.

gilcu2, 2023-01-16:
The changes in the Readme looks pretty good. I will check when I have time. Thanks (-:

MrCsabaToth, 2023-01-17:
Actually I decided to switch to the beta channel. I'll update the README again eventually.

MrCsabaToth, 2023-01-17:
I got tired of ditching several hundred yaml changes every time I upgrade flutter and the `flutter update-packages --force-upgrade`

gilcu2, 2023-01-27:
Hello MrCsabaToth. I see you changed to the flutter beta channel but Readme talks about stable. What is the easiest or recommended env? What about flutter 3.7?

MrCsabaToth, 2023-01-27:
I switched to the beta channel in the `beta_channel` feature branch only, which is not merged to develop yet. That branch has a modified README. The develop branch's README has not changed yet. However the merge will happen soon (in days) and then a beta channel will be required. This is to save everyone from the SDK version modifications (which need to be undone before Flutter version updates).
Not sure about 3.7, what do you think specifically? I'll embrace whatever is coming through beta. Flutter Forward was very promising!

gilcu2, 2023-01-27:
Thanks, I am clear now

MrCsabaToth, 2023-01-28:
Merged with beta release 135

MrCsabaToth, 2023-01-30:
I'm not liking the dependencies right now after switching to beta channel: https://stackoverflow.com/questions/75289298/how-to-bump-pref-package-to-version-2-7-1-needs-flutter-3-7-0-while-im-on-the
I reopen this issues.

MrCsabaToth, 2023-02-09:
More particularly I don't like that the lock file now ties every package to a specific SHA hash. The PRs what the Dependabot crafts produces the regular non SHA hash tied lock files. Maybe the hash will be the normal now, but why exact version is not enough?

closed 2023-02-10 by MrCsabaToth

MrCsabaToth, 2024-05-09:
Thinking about switching back to stable channel now. Hopefully there won't be too turbulent Flutter/Dart architectural upgrades, and this would possibly help with IDX IDE as well.

MrCsabaToth, 2024-08-07:
I switched back to stable channel for a while now

commit ba825edf (2023-01-18): Remove dormant unused function lefotver from the use_build_context_synchronously lin fix #399
See also https://github.com/rekariproject/datetime_picker_formfield/issues/1