TrackMyIndoorWorkout/TrackMyIndoorWorkout #404: Support for Virtufit rowing machine

roeland4, 2023-01-28:
I have a Virtufit rowing machine: https://virtufit.nl/product/virtufit-ultimate-pro-2-ergometer-roeitrainer/
It doesn't show up in the app. 
Should a device like this work or is this not supported?
Can I do something to add support?

[image]

MrCsabaToth, 2023-01-28:
Thanks for your report!

One thing to watch out for is that most of the consoles are single connection only. This means that if you connected to it before let's say with your tablet, and even if that app is now somewhere in the background the tablet can still hold the connection and that prevents the console to advertise itself for other devices. So make sure you fully exit / force close all related apps.

Otherwise your rower should show up because I see it's an FTMS (0x1826) and you have the 0x2AD1 Rower Data. Which app version are you using? Last night I pushed a 135 beta release, I wonder if this could be a fresh bug or not.

That latest build should handle cases when  the fitness machine (FTMS) doesn't hint the subtype (in your case it's a rower) in the advertisement service data. So I developed code which pre-connects the device in such cases and snoops the characteristics.

The app has a switch to not filter by perceived compatibility and list everything, it's under the Expert preferences, a switch almost at the top. If you turn that off what happens? Your device should show up and I wonder what the service data is under the details drawer in my app.

Let me know how that goes.

roeland4, 2023-01-28:
Ok, now I see the device. Maybe there was something else connected then. 

But I get this error now. 

[image]

roeland4, 2023-01-28:
I attached a log. Not sure If you can use it.

[2801202318.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/10527841/2801202318.txt)

MrCsabaToth, 2023-01-28:
Wow, I'm suspecting what's going on! This device is strange: it's not only FTMS, but it's manufactured by Wahoo and it has this 0xEE07 service. That's exactly the same service what my Precor Spinner Chrono Power has, and the code internals first check for that. So the code might not be well prepared for this ambiguity (both 0x1826 and 0xEE07 present) and I suspect that sometimes it might try to handle it as a Spinner Chrono Power and sometimes as a Generic FTMS Indoor bike.

I'll need some down time to revise code paths and resolve this (while leaving the Precor support and the generic Indoor Bike support intact). Essentially right now I first check for corner cases and proprietary protocols, such as the Precor Spinner Chrono Power which I reverse engineered. Also for some FTMS which are misimplemented. Only if there's no match for corner cases I fall back to generic FTMS support paths.

TL;DR: I can see a light at the end of the tunnel. I'm traveling, so it could take a couple of weeks.

roeland4, 2023-01-28:
No problem. Take all the time you need. Thanks for looking into this.

MrCsabaToth, 2023-01-29:
A few questions:
1. Do you have an 0xE01D characteristic inside that 0xEE07 service?
2. What characteristics does nRF Connect show under 0xEE07? (Similar way as you showed 1826 FTMS
3. Which app version are you using?

roeland4, 2023-01-29:
I'm using version 1.1.133 build 133

Doe you mean this?
uuid="a026**e01d**-0a7d-4ab3-97fa-f1500f9feb8b"

I exported the services to xml.

```xml
<server-configuration name="VIRTUFIT-UP2i-15782 - copy">
   <service uuid="a026ee01-0a7d-4ab3-97fa-f1500f9feb8b">
      <characteristic uuid="a026e002-0a7d-4ab3-97fa-f1500f9feb8b">
         <descriptor configure="CCCD"/>
         <permission name="WRITE"/>
         <property name="WRITE_WITHOUT_RESPONSE"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="a026e004-0a7d-4ab3-97fa-f1500f9feb8b">
         <descriptor configure="CCCD"/>
         <property name="NOTIFY"/>
      </characteristic>
   </service>
   <service uuid="0000180a-0000-1000-8000-00805f9b34fb">
      <characteristic uuid="00002a29-0000-1000-8000-00805f9b34fb">
         <permission name="READ"/>
         <property name="READ"/>
      </characteristic>
      <characteristic uuid="00002a24-0000-1000-8000-00805f9b34fb">
         <permission name="READ"/>
         <property name="READ"/>
      </characteristic>
      <characteristic uuid="00002a25-0000-1000-8000-00805f9b34fb">
         <permission name="READ"/>
         <property name="READ"/>
      </characteristic>
      <characteristic uuid="00002a27-0000-1000-8000-00805f9b34fb">
         <permission name="READ"/>
         <property name="READ"/>
      </characteristic>
      <characteristic uuid="00002a26-0000-1000-8000-00805f9b34fb">
         <permission name="READ"/>
         <property name="READ"/>
      </characteristic>
   </service>
   <service uuid="a026ee07-0a7d-4ab3-97fa-f1500f9feb8b">
      <characteristic uuid="a026e01f-0a7d-4ab3-97fa-f1500f9feb8b">
         <descriptor configure="CCCD"/>
         <permission name="READ"/>
         <property name="READ"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="a026e01e-0a7d-4ab3-97fa-f1500f9feb8b">
         <descriptor configure="CCCD"/>
         <permission name="READ"/>
         <property name="READ"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="a026e020-0a7d-4ab3-97fa-f1500f9feb8b">
         <descriptor configure="CCCD"/>
         <permission name="READ"/>
         <property name="READ"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="a026e01b-0a7d-4ab3-97fa-f1500f9feb8b">
         <descriptor configure="CCCD"/>
         <permission name="READ"/>
         <property name="READ"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="a026e01d-0a7d-4ab3-97fa-f1500f9feb8b">
         <descriptor configure="CCCD"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="a026e033-0a7d-4ab3-97fa-f1500f9feb8b">
         <permission name="READ"/>
         <property name="READ"/>
      </characteristic>
      <characteristic uuid="a026e016-0a7d-4ab3-97fa-f1500f9feb8b">
         <permission name="WRITE"/>
         <property name="WRITE_WITHOUT_RESPONSE"/>
      </characteristic>
      <characteristic uuid="a026e024-0a7d-4ab3-97fa-f1500f9feb8b">
         <descriptor configure="CCCD"/>
         <permission name="WRITE"/>
         <property name="WRITE_WITHOUT_RESPONSE"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="a026e028-0a7d-4ab3-97fa-f1500f9feb8b">
         <descriptor configure="CCCD"/>
         <permission name="WRITE"/>
         <property name="WRITE_WITHOUT_RESPONSE"/>
         <property name="NOTIFY"/>
      </characteristic>
   </service>
   <service uuid="00001826-0000-1000-8000-00805f9b34fb">
      <characteristic uuid="00002acc-0000-1000-8000-00805f9b34fb">
         <permission name="READ"/>
         <property name="READ"/>
      </characteristic>
      <characteristic uuid="00002ad3-0000-1000-8000-00805f9b34fb">
         <descriptor configure="CCCD"/>
         <permission name="READ"/>
         <property name="READ"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="00002ada-0000-1000-8000-00805f9b34fb">
         <descriptor configure="CCCD"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="00002ad1-0000-1000-8000-00805f9b34fb">
         <descriptor configure="CCCD"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="00002ad9-0000-1000-8000-00805f9b34fb">
         <descriptor configure="CCCD"/>
         <permission name="WRITE"/>
         <property name="WRITE_WITHOUT_RESPONSE"/>
         <property name="INDICATE"/>
      </characteristic>
   </service>
   <service uuid="4b480001-6e6f-7274-6870-6f6c65656e67">
      <characteristic uuid="4b480002-6e6f-7274-6870-6f6c65656e67">
         <descriptor configure="CCCD"/>
         <property name="NOTIFY"/>
      </characteristic>
      <characteristic uuid="4b480003-6e6f-7274-6870-6f6c65656e67">
         <permission name="WRITE"/>
         <property name="WRITE_WITHOUT_RESPONSE"/>
      </characteristic>
   </service>
</server-configuration>
```

MrCsabaToth, 2023-01-29:
Yes, that. You do have 0xE01D. Interesting. Those 0xE... services and characteristics are all proprietary and I reverse engineered it specifically for Spinner Chrono Power which is an indoor bike. I have to change the code to favor FTMS over the Precor corner case.

roeland4, 2023-01-29:
Yes, it seems both FTMS and the Wahoo Fitness Profile are available.
I found a document about the Wahoo profile. Maybe you can use it.

[WAHOO FITNESS EQUIPMENT PROFILE.pdf](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/10530688/WAHOO.FITNESS.EQUIPMENT.PROFILE.pdf)

MrCsabaToth, 2023-01-29:
Based on that
1. I need to take back labeling the protocol proprietary, because that looks like a proper documentation
2. Based on the documentation in the future I should add proper support for the whole family of devices #406
3. For you I'll need to figure out a quick fix

MrCsabaToth, 2023-02-08:
Hey @roeland4, could you test this APK https://drive.google.com/file/d/1JzZbpAIWQflFcTD4y8tzoiB8sY5BXo7U/view?usp=share_link ?
I was on a trip and haven't been able to work on this.
In case it doesn't work out I'll release another update shortly.

roeland4, 2023-02-08:
I installed build 134, but I get the same error.
The log is attached.

I also don't see a version number in the log, but maybe because I pressed _clear logs_ before starting?

[0802202308.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/10683188/0802202308.txt)

Let me know if I can do some other tests. Thx!

roeland4, 2023-02-08:
No problem. I tried again, but still the same error. Not sure if I do something wrong here.
Sometimes I get a generic error and than the app goes a step further, but everything remains zero.

[0802202310.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/10683904/0802202310.txt)

MrCsabaToth, 2023-02-08:
It's definitely not your fault. When the app scanned was it displaying your device as a rower now, or was it still a bike icon or a question mark icon?

roeland4, 2023-02-08:
[image]
[image]

MrCsabaToth, 2023-02-09:
We are very close to a solution, thanks for bearing with me. Seems like the icon is good now, but somehow the app still somehow pulls out a Generic Indoor Bike descriptor. Did the app ask you in any of the runs to pick a sport (from a list of water sports: rowing, kayaking, canoeing, swimming)? I want to be sure there's no stored device usage with the wrong sport (riding). Device Usages are to resolve situations when the sport is ambiguous and after the first pick the app remembers the sport this way (per device). To get to the Device Usages first navigate to the workout list, and then select the button on my screenshots:
[image]

Do you see any entries for the rower? If there is then please delete it.

[image]

MrCsabaToth, 2023-02-09:
I would have one more improvement: a change which will specifically recognize your rower.
Could you test this 4th version of build 136 pre-release https://drive.google.com/file/d/1XXNo6JhMnvU94wohGgYXLzxZdurhQuqd/view?usp=share_link ? Delete the Device Usage before testing it.

roeland4, 2023-02-10:
Strange the connection seems to fail now.

[1002202307.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/10704520/1002202307.txt)

MrCsabaToth, 2023-02-10:
So sorry, probably this will be the last required version: https://drive.google.com/file/d/1apZILHz6d6mjgLfG4k9fRquesSsgHmm9/view?usp=share_link
Based on your log I see that unexpectedly the manufacturer is not Wahoo Fitness, LLC, so I'm trying to match XEBEX which is the reported one by the logs. This is needed so I identify the device precisely.

roeland4, 2023-02-10:
ok thanks. It works again. But I'm not sure what information you need now to confirm it works? Happy to provide any information you need.

MrCsabaToth, 2023-02-10:
The log file was sufficient.

closed 2023-02-10 by MrCsabaToth