TrackMyIndoorWorkout/TrackMyIndoorWorkout #434: HR not showing when connected to Garmin FR 255 (broadcasting)

rjwonder, 2023-05-14:
The Android app is connected to the Garmin forerunner, which is broadcasting heart rate. When starting my Schwinn bike I can see actions coming from my bike, but heart rate stays empty. 

Is there anything I should do to display HR (Source: Garmin watch) while on my Schwinn bike?

MrCsabaToth, 2023-05-16:
Which Schwinn bike do you have? Usually (depending on the model) they are capable of pairing the HRM and usually my app can receive the heart rate via the console. This way you'd see heart rate both on the console or in the app.
There's also a possibility to pair the HRM directly to the app. That can be done before the workout on the generic scan screen: you can click on the HRM row's play button icon and that'd pair it with the app. Then when you click on the bike's play button the heart rate should be visible.
The other possibility is IF the instant measurement start option is NOT enabled (it's enabled by default, this causes the app to automatically start the workout recording once the navigation completes to the workout recording screen), then you can click the heart icon in the menu and you can pair HRM. This can theoretically be done even during the workout.

Recently I have reports that the application has problems with Schwinn 170/270/570 (#433). I haven't had opportunity to debug that yet, and it can happen that you may discovered a new or related bug. In that case the app has an internal logging capability: under the expert preferences you can turn on the logging to INFO level and then export a log. (Then don't forget to turn off the logging).

It's another story if the HRM doesn't show up in the scans. The heart rate monitor should be Bluetooth Low Energy Heart Rate Profile compatible.

MrCsabaToth, 2023-07-17:
Any news on this?

MrCsabaToth, 2023-08-23:
DC Rainmaker's guide on Garmin Wearable Heart Rate Broadcasting: https://www.dcrainmaker.com/2020/04/quick-how-to-garmin-wearable-heart-rate-broadcasting-to-apps.html

MrCsabaToth, 2023-12-11:
For Wear OS see https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/457 https://play.google.com/store/apps/details?id=lukas.the.coder.heartforbluetooth

MrCsabaToth, 2024-05-09:
I close this with the premise that Wear OS or Garmin native watch app which can transmit heart rate via standard HRM profile BLE protocol.

closed 2024-05-09 by MrCsabaToth