TrackMyIndoorWorkout/TrackMyIndoorWorkout #435: Android port in a disarray after AGP 8.0+ upgrade

MrCsabaToth, 2023-06-14:
The situation is a Deja Vu of the Dart nullability. Some package fixes require being up to date, however some packages are lagging behind. Several things going on: Dart major version upgrade, AGP 8.0+ upgrade, newer SDK versions.

Right now I'm working around and lagging plugins, it's a carnage. Will I need to yank out the protobuf from flutter_blue_plus to finally compile?

* https://github.com/flutter/flutter/issues/126403

MrCsabaToth, 2023-07-05:
I reverted to Flutter main (stable) channel, "stashed" the AGP8 carnage in a feature branch, and now trying it all over again without and AGP upgrade. This is a complete carnage, even worse than the one I had with null safety. I have many bugs to fix and the app doesn't compile. Not even for debug.

MrCsabaToth, 2023-07-05:
I hope that after I get through bugs and work commitments I'll be able to tackle https://github.com/boskokg/flutter_blue_plus/issues/338

MrCsabaToth, 2023-07-18:
I reverted to beta channel and right now debug is working without AGP8+ upgrade. However after trying the newest FlutterBluePlus 1.7.1 I ran into a permission issue: the debug version complains about missing location permission.

[Bluetooth apps on Android 12 and above would not need location permission for Bluetooth any more](https://www.androidpolice.com/2021/05/19/android-12-apps-wont-ask-for-location-permissions-when-all-they-want-is-bluetooth-scanning-which-yes-was-a-thing/), however FlutterBluePlus still required it (see https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/commit/f64f994c199ebe3724557e89c65a86263c5691d8).

Now with 1.7.1 even though I have explicit FINE and COARSE location permission use in my manifest, the plugin still crashes complaining about the lack of it. When I look at the app's permission settings in the OS it does not list location, so I'd be able to toggle it, and consequently I also cannot ask for it explicitly with the `permission_handler` plugin (instant denial). So we are in the pickle with this one as well.

MrCsabaToth, 2023-07-18:
The bluetooth permission problem seems to be noticed by others as well: https://github.com/boskokg/flutter_blue_plus/issues/370

MrCsabaToth, 2023-07-18:
Ref #378
Also want to mention for latest carnage: #443 #440

MrCsabaToth, 2023-07-30:
So I'm trying AGP 8.0+ upgrade again. Several dilapidated plugins making the AGP upgrade assistant fail:
* https://github.com/JayTWWM/Share-Files-And-Screenshot-Widgets-Flutter/issues/28
* https://github.com/ukasz123/soundpool/issues/117
* https://github.com/ukasz123/soundpool/issues/121
* https://github.com/avioli/uni_links/issues/187
* https://github.com/isar/isar/issues/1354

MrCsabaToth, 2023-07-30:
I also dropped the `bluetooth_enable_fork` plugin which was also one which blocked the AGP 8.0+. Isar is awesome and will pull through, but I need to look for alternatives for soundpool, uni_links, and the file sharing plugin.

MrCsabaToth, 2023-07-30:
Audio alternatives: all of them seems to have GUIs, which we don't need: https://fluttergems.dev/audio/

MrCsabaToth, 2023-07-30:
As for uni_links goes we only use it for it's `uriLinkStream.listen(` type streaming capability waiting for a return to confirm an upload. We don't leverage any deep linking right now. So maybe `dio` can be a good alternative? https://blog.logrocket.com/networking-flutter-using-dio/

MrCsabaToth, 2023-07-30:
For file share:
* https://pub.dev/packages/esys_flutter_share (discontinued)
* https://pub.dev/packages/share by flutter.dev, discontinued in favor of share_plus
* https://pub.dev/packages/share_extend
* https://pub.dev/packages/flutter_share_me
* https://pub.dev/packages/share_plus fluttercommunity.dev plugin

I'd try the last one because it's a fluttercommunity.dev maintained plugin with less likely to become abandoned.

closed 2023-07-31 by MrCsabaToth

MrCsabaToth, 2023-07-31:
There wasn't a problem with debug build, but with the release I got a
`~/repos/flutter/track_my_indoor_exercise/build/app/outputs/mapping/release/missing_rules.txt`
```
# Please add these rules to your existing keep rules in order to suppress warnings.
# This is generated automatically by the Android Gradle plugin.
-dontwarn com.google.android.play.core.splitcompat.SplitCompatApplication
-dontwarn com.google.android.play.core.splitinstall.SplitInstallException
-dontwarn com.google.android.play.core.splitinstall.SplitInstallManager
-dontwarn com.google.android.play.core.splitinstall.SplitInstallManagerFactory
-dontwarn com.google.android.play.core.splitinstall.SplitInstallRequest$Builder
-dontwarn com.google.android.play.core.splitinstall.SplitInstallRequest
-dontwarn com.google.android.play.core.splitinstall.SplitInstallSessionState
-dontwarn com.google.android.play.core.splitinstall.SplitInstallStateUpdatedListener
-dontwarn com.google.android.play.core.tasks.OnFailureListener
-dontwarn com.google.android.play.core.tasks.OnSuccessListener
-dontwarn com.google.android.play.core.tasks.Task
```

MrCsabaToth, 2023-12-08:
@dkaera Thanks, I'll try that. The situation kind of normalized as packages slowly catch up and upgrade as well, but you never know!

dkaera, 2023-12-08:
@MrCsabaToth 
yeah, I still have packages that don't use a namespace, so I've had to add this kind of workaround.