TrackMyIndoorWorkout/TrackMyIndoorWorkout #436: Add support for Topiom Rowing Machine

theaaron123, 2023-06-15:
The Bluetooth module is supported by KinoMap and FitShow apps and is displayed as `FS-4F3000` within them but does not show up as a supported device within TrackMyIndoorWorkout. 

I can help implement / provide any debug information if you point me in the right direction.

MrCsabaToth, 2023-06-15:
I mainly support FTMS compatible devices. The main question is if the machine exposes the 0x1826 Bluetooth Low Energy (BLE) service.

1. When trying my app, make sure that you exit any app on any other device you used the fitness machine with. Even if an application is in the background it may hold the Bluetooth connection to the machine and the console won't advertise itself so it won't be visible to any other app.
2. In case my app doesn't show the machine then you can turn off the compatibility filtering in the Expert preferences section. This will show every Bluetooth device which are advertising, not only the ones my app deems compatible. Does the machine show up? If yes, then click on the row (but not the play icon) and that will open an accordion section with details about the machine. What services does the machine advertise (Service UUIDs)?
3. In case of an unsupported protocol, or the machine not showing up you can also try the [NRF Connect app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp). That's a generic Bluetooth app which is capable of connection to any device and even logging traffic.
4. If the machine is FTMS compatible (or other BLE standard, like CSC cycling sensor, power meter sensor, RSC sensor) and I don't show it then I'd need to debug the situation. I developed an internal logging mechanism into my app which is also under the Expert preferences: you can set the logging level to INFO and then later export the logs.
5. In case of proprietary protocol the situation is hard. The best would be if I can use a similar machine. What make and model is it? It's not impossible to develop a support without having the machine, such as happened with Schwinn 170 / 270 / 570. However recently I learned that the support has problems (#433) and I was still not able to solve because I don't have those machines. Are you a developer by any chance?

MrCsabaToth, 2023-06-16:
Ok, so it's a rower. In this case my app expect the device to advertise 0x1826 FTMS Fitness Machine service support and within that a 0x2AD1 Rower Data characteristic (https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.rower_data.xml).

Maybe it's just that the console doesn't explicitly advertise that. However it's infeasible for my app to actually connect to everything it sees in order to determine if the support is fine or not. I'd say that you can try to disable the compatibility filtering and try to connect to the device (preferably also logging enabled).

With Kinomap stuff is more manual: you need to pick the type of your machine. One application I'm also curious about - and this is easier to use the NRF Connect because it's a normal user app - is [GymTrakr](https://play.google.com/store/apps/details?id=com.npe_inc.gem.gymtrakr). Please give that a try and report back if it can simply support the rower's console.

theaaron123, 2023-06-16:
The machine is advertising the following service UUIDs `180A, 180F, FFF0` and looking them up it may be that `FFF0` is a propriety protocol? 

GymTrakr doesn't connect.

The logs when trying to connect are:

`{FIND_DEVICES}  {startScan}  {Scan initiated}  {16 June 2023 12:12:46.136 PM}  {INFO}
{FIND_DEVICES}  {ScanResult}  {ScanResult{device: BluetoothDevice{id: CB:F0:01:00:30:4F, name: FS-4F3000, type: BluetoothDeviceType.le, isDiscoveringServices: false, _services: [], advertisementData: AdvertisementData{localName: FS-4F3000, txPowerLevel: null, connectable: true, manufacturerData: {1002: [115, 48, 79, 48, 0, 0, 0, 0, 0, 213]}, serviceData: {}, serviceUuids: [0000180a-0000-1000-8000-00805f9b34fb, 0000180f-0000-1000-8000-00805f9b34fb, 0000fff0-0000-1000-8000-00805f9b34fb]}, rssi: -72, timeStamp: 2023-06-16 12:12:46.247316}}  {16 June 2023 12:12:46.256 PM}  {INFO}
{logVersion}  {version}  {1.1.140 build 140}  {16 June 2023 12:12:49.819 PM}  {ERROR}
{FIND_DEVICES}  {goToRecording}  {Could not infer sport of the device}  {16 June 2023 12:12:49.824 PM}  {ERROR}
{FIND_DEVICES}  {ScanResult}  {ScanResult{device: BluetoothDevice{id: CB:F0:01:00:30:4F, name: FS-4F3000, type: BluetoothDeviceType.le, isDiscoveringServices: false, _services: [], advertisementData: AdvertisementData{localName: FS-4F3000, txPowerLevel: null, connectable: true, manufacturerData: {1002: [115, 48, 79, 48, 0, 0, 0, 0, 0, 213]}, serviceData: {}, 
serviceUuids: [0000180a-0000-1000-8000-00805f9b34fb, 0000180f-0000-1000-8000-00805f9b34fb, 0000fff0-0000-1000-8000-00805f9b34fb]}, rssi: -72, timeStamp: 2023-06-16 12:12:46.247316}}  {16 June 2023 12:12:49.835 PM}  {INFO}
{logVersion}  {version}  {1.1.140 build 140}  {16 June 2023 12:13:01.518 PM}  {ERROR}` 

I'll log some traffic with NRF Connect next and we may be able to determine if it's using a standard protocol but not advertising it as such.

MrCsabaToth, 2023-06-16:
I tried to determine Bluetooth support by web searches for the model, but the results are kind of inconclusive. Some reviews only mention Bluetooth support in general (which with some other machine only means capability to pear a BT heart rate monitor to the console). Some articles mention that apps support the console, but I never saw FTMS as a keyword mentioned. The fact that GymTrakr is not recognizing it also points towards a proprietary protocol. Sometimes it's hard to crack those (for example #275).

We have a few venues. One is to get help from the manufacturers, since if Kinomap supports it then there must be some proprietary documentation. The other is reverse engineering. For that the key is to remember what strokes per minute and what resistance level you were on and pair specific data packets to those. For example row with light resistance but high spm, and also high resistance lower spm. Also remember pace / speed. Then the specific bytes might be identifiable. That's how I reverse engineered the Precor Spinner Chrono Power as the first supported device when I started this app. But I was not able to crack the Bowflex #275 yet.

MrCsabaToth, 2023-06-16:
One more thing: does this machine have different versions with different consoles? When articles are inconclusive then sometimes this is the case. Sometimes the Bluetooth support require some extra pay for the "smarter" console. Sometimes console firmware updates could bring in new capabilities as well.

MrCsabaToth, 2023-06-16:
Shot the company an email, it could never hurt:
"I'm a mobile application developer who'd like to support your rowers. My application is https://trackmyindoorworkout.github.io/
Recently a customer of yours was asking for support: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/436
My searches were inconclusive, so I curious if your consoles are capable of Bluetooth FTMS Rower Data support?
Does your bluetooth support depend on console version, or firmware version?
Since Kinomap and a few applications seems to support your in case you are using a proprietary protocol would you be able to provide me a copy. I doesn't have to be fancy, but only enough so I can develop a support.
My application is free, free of ads, and open source."

In case of Bowflex I was never able to connect even after LinkedIn forensics, however this is a smaller company so they might be less bureaucratic and willing to talk to a tiny fish like me.

theaaron123, 2023-06-19:
There are at least two consoles one newer "TM3" and mine being the "TM2". 

Cheers for reaching out to the company, lets see if they reply.

In the mean time I've logged some traffic when connected with KinoMap.

As its a water rower, I believe the only transmitted value with be RPM, then all others will be derived? For example watts and distance calculations are wildly different in KinoMap compared to what the LCD on the rower shows.

Here's the log:
 
[Log 2023-06-19 10_48_50.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/11786879/Log.2023-06-19.10_48_50.txt)

MrCsabaToth, 2023-06-19:
I got a response back from the company:

> Dear Csaba,
> 
> All of our current monitors are Bluetooth-compatible, and we no longer offer the older version of displays. Rest assured that our current version is functioning properly. Thank you for your attention and continued support of our products.
>
> If you have any further questions or concerns, please feel free to reach out.
>
> Best regards,
> 
> Doris

I'm following up about FTMS or documentation. It's promising that they are communicating.

MrCsabaToth, 2023-06-19:
I'm able to calculate calories from heart rate (#154), however it can be extremely off especially between various sports (see for example #369). Calorie tune feature helps to apply a linear modification factor to that value.
Then I'm able to calculate watts from calorie delta. In case of cycling I have formulas to calculate speed based on watts. Different sports need different formulas though, because running, rowing or swimming won't be affected as much by wind resistance for example, but other physical factors could come in strong. See #398.
This wouldn't be able to tell spm (strokes per minute), and the transitive multi stage approximations would result in widely varying numbers. Although right now I have several machines which leave out some key measurements from the FTMS data for whatever reason (for example Schwinn IC4/IC8 is missing the calories. So far Schwinn 170 / 270 / 570 is the bike which has a proprietary protocol and I truly use multi stage transitive approximations (calories -> watts -> speed): #240

I'd still want to identify some data in the stream other than the heart rate. Another venue could be to look for console upgrade if Topium has any (for example water rower has S3 -> S4 upgrade https://www.waterrower.com/us/products/s3-to-s4-monitor-upgrade-2) or Bluetooth Com module (FTMS compatible, can plug in on the console https://www.waterrower.com/us/products/waterrower-commodule).

I've also seen some hardware projects:
* https://github.com/zpukr/ArduRower
* https://github.com/laberning/openrowingmonitor

I can co-operate to tune-in multi-stage approximation like with the the Schwinn 170, it sounded working fine (although it seems to have unrelated problems right now #433).

MrCsabaToth, 2023-06-19:
One more thing: in case the shaft of the resistance wheel is exposed, you might be able to attach a cycling cadence sensor (CSC Bluetooth standard) on it. This is how I support the Old Danube kayak ergometer: #384
This is Old Danube specific right now, but I can extend / modify it if it's applicable. That would be able to provide some kind of a data, although the spm is not correlated to the flywheel speed in a linear way, because there's also resistance setting.