TrackMyIndoorWorkout/TrackMyIndoorWorkout #440: Integration / portal Upload SVG screens are white out

MrCsabaToth, 2023-07-15:
This just happened suddenly after I upgraded to the latest Flutter. The white background of the buttons overflow and color the whole screen, leaving only the last button's logo visible and everything white.

People experience all kinds of strange things seem to be related to impeller:
* https://github.com/dnfield/flutter_svg/issues/938
* https://github.com/dnfield/flutter_svg/issues/942
* https://github.com/dnfield/flutter_svg/issues/961
* https://github.com/dnfield/flutter_svg/issues/962
* https://github.com/dnfield/flutter_svg/issues/965

One of the referenced impeller issue: https://github.com/flutter/flutter/issues/130078

MrCsabaToth, 2023-07-17:
Will be available with the next release

closed 2023-07-17 by MrCsabaToth

fisforfaheem, 2023-07-18:
please tell how to also fix this eror: I/flutter ( 7924): unhandled element filter; Picture key: Svg loader