TrackMyIndoorWorkout/TrackMyIndoorWorkout #449: The elapsed / moving time shows 0 while not pedaling and sets the elapsed time when peadling stops

MrCsabaToth, 2023-07-24:
Reported by Stuart Nealis:
1. While pedaling the time shows 00:00:00
2. When stopped pedaling the time display reveals the elapsed (or moving?) time, but the time still doesn't advance while stopped
3. When restarted pedaling the display zeroes out again

Questions:
1. What happens if Stuart stops again after that?
2. What type of machine is that?
3. Can we get a log about all of this?
4. What version of the app Stuart uses?
5. Does Stuart have HIIT mode turned on?
6. What happens if Stuart turns on Average, Maximum, or Alternating Stage Mode? This would divide a set of the metrics during workout and would display a separate moving and elapsed time.

Video recording: https://youtube.com/shorts/m8vDF8Tf7XE
Screenshot:
[image]

MrCsabaToth, 2023-07-24:
The device is a Pixel phone with Android 13 () and 5G support, so one of Pixel 4a (5G), 5, 5a (5G), 6, 6 Pro, 7, 7 Pro.

MrCsabaToth, 2023-07-27:
* Stuart doesn't have the HIIT mode on
* When he enabled the stat mode then the elapsed time was enhancing, but the moving time didn't
* I realized that right now both the production and the beta releases are at the same version 1.1.146
* The machine is a [Mekbelt from Amazon](https://a.co/d/3pqpZ74)
* When Stuart continues pedaling then stops again the moving time advances to a new value, but then holds that value
* Stuart provided a log as well

I suspect that somehow the logic doesn't consider the athlete moving, despite of having non zero metrics
[image]
[image]

MrCsabaToth, 2023-07-27:
The logs: [2607202310.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/12180026/2607202310.txt)

What is telling in the logs is that we can see a lot of cumulativeMovingTimeEnforcement violations. The values seem to probably correspond to the values which show through when the pedaling stopped. First is 1455, then 4843, ...

MrCsabaToth, 2023-08-23:
Gary Vellenzer experiences this as well as he is testing https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/457

MrCsabaToth, 2023-12-17:
> I encountered a variation of this issue when I stopped a workout and restarted it. It seemed like 2 different workouts were running simultaneously, the values shown would switch between the two.
> 
> While moving, the moving time would show 0 and calorie count would show the combined count from the old and new workout. While stationary, the moving time and calorie count would show the newer workout only. BPM would switch between the current and old value each refresh.
> 
> Screen recordings are from Samsung Dex, but I experienced this issue a couple times while using the phone normally without Dex. The time in the bottom right of the video can be matched to the logs.
> 
> 12:24 Workout start 12:44 I stopped the workout and started a new one since my HR monitor got disconnected (reconnect wouldn't work via bottom right menu)
> 
> [SamsungDeX_VF6Kglkj16.webm](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/assets/8077950/515228c4-c639-4687-8665-e4d098d6ac1a) [SamsungDeX_wvOyT2koR1.webm](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/assets/8077950/9263c39b-a534-4f70-9a7f-133b3bdffbf1) [logs.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/13687787/logs.txt)

Thanks for the recording, that is very interesting. I haven't thought it could be two workouts simultaneously. Several thoughts:
1. I notice the HRM glitching, this could happen if you get HRM data from the bike console, and also from an extra paired heart rate monitor. The two monitors feed into the same value and if they differ it'll glitch like that back-and forth. Did you manually pair an extra HRM via the app and also another to the bike console? What HRM are you using?
2. I see that you might be using a Schwinn IC4 or similar bike? IC8 or something else?
3. As for the workout / multiple simultaneous trainings: my thoughts are bellow

1. I can imagine one way for two workouts to happen simultaneously: in earlier version of Flutter I could actually prevent the user to leave the measurement screen while a workout was in progress. Since newer versions of Flutter I cannot fully prevent that any more, only warn the user when the workout is in progress and they are about to navigate away. This warning may not be shown if the user unintentionally or accidentally perform a back navigation gesture (such as on my OnePlus a simple flick from the left edge of the screen towards center (flick right). If this happens I still try to stop the workout, but the subscription routine which listens to the BLE data stream may not be unsubscribed and destroyed. I think this would case errors later, but maybe if someone starts a new workout after that the old subscribe and the new may run simultaneously. Can it happen that you navigated away accidentally and then started another workout? In case you sweat and it may cause unintended navigation or gestures there's a lock feature.
2. Only very specific cases allow that, such as circuit training when you switch between several machines, and I intentionally let two workout happen at the same time, and stop all of those once the whole circuit training is over. If you enable that the app simply lets you navigate back to the device finder screen. Have you maybe enabled circuit training by any chance? I don't think so, because the stop button turns into a pause button in that case.

I suspect this is still some other bug related to handling if the athlete is moving or not, but maybe it's simultaneous recording, thanks for the suggestion. I'm wondering about your answers.

CBlevings, 2023-12-18:
> > I encountered a variation of this issue when I stopped a workout and restarted it. It seemed like 2 different workouts were running simultaneously, the values shown would switch between the two.
> > While moving, the moving time would show 0 and calorie count would show the combined count from the old and new workout. While stationary, the moving time and calorie count would show the newer workout only. BPM would switch between the current and old value each refresh.
> > Screen recordings are from Samsung Dex, but I experienced this issue a couple times while using the phone normally without Dex. The time in the bottom right of the video can be matched to the logs.
> > 12:24 Workout start 12:44 I stopped the workout and started a new one since my HR monitor got disconnected (reconnect wouldn't work via bottom right menu)
> > [SamsungDeX_VF6Kglkj16.webm](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/assets/8077950/515228c4-c639-4687-8665-e4d098d6ac1a) [SamsungDeX_wvOyT2koR1.webm](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/assets/8077950/9263c39b-a534-4f70-9a7f-133b3bdffbf1) [logs.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/13687787/logs.txt)
> 
> Thanks for the recording, that is very interesting. I haven't thought it could be two workouts simultaneously. Several thoughts:
> 
> 1. I notice the HRM glitching, this could happen if you get HRM data from the bike console, and also from an extra paired heart rate monitor. The two monitors feed into the same value and if they differ it'll glitch like that back-and forth. Did you manually pair an extra HRM via the app and also another to the bike console? What HRM are you using?
> 2. I see that you might be using a Schwinn IC4 or similar bike? IC8 or something else?
> 3. As for the workout / multiple simultaneous trainings: my thoughts are bellow
> 4. I can imagine one way for two workouts to happen simultaneously: in earlier version of Flutter I could actually prevent the user to leave the measurement screen while a workout was in progress. Since newer versions of Flutter I cannot fully prevent that any more, only warn the user when the workout is in progress and they are about to navigate away. This warning may not be shown if the user unintentionally or accidentally perform a back navigation gesture (such as on my OnePlus a simple flick from the left edge of the screen towards center (flick right). If this happens I still try to stop the workout, but the subscription routine which listens to the BLE data stream may not be unsubscribed and destroyed. I think this would case errors later, but maybe if someone starts a new workout after that the old subscribe and the new may run simultaneously. Can it happen that you navigated away accidentally and then started another workout? In case you sweat and it may cause unintended navigation or gestures there's a lock feature.
> 5. Only very specific cases allow that, such as circuit training when you switch between several machines, and I intentionally let two workout happen at the same time, and stop all of those once the whole circuit training is over. If you enable that the app simply lets you navigate back to the device finder screen. Have you maybe enabled circuit training by any chance? I don't think so, because the stop button turns into a pause button in that case.
> 
> I suspect this is still some other bug related to handling if the athlete is moving or not, but maybe it's simultaneous recording, thanks for the suggestion. I'm wondering about your answers.

1. I don't use the HRM on the bike, and only have one monitor paired (FitBit Charge 6). I manually paired my Charge 6 to the app, but it lost connection during the workout. I couldn't re-pair it from the bottom right wheel menu for some reason, it wouldn't connect. I backed out of the workout and paired it from the main menu of the app.

2. I'm using a Bowflex C6, but it's basically a rebranded Schwinn IC4. They both use the same console unit as far as I'm aware.

3. -

4. I didn't use any gestures, but I may have hit the back arrow/button. I saw the popup about the current workout, not sure what option I chose though when that showed. I navigated back to the main screen of the app to re-pair my HRM as mentioned above.

5. I do not have circuit training enabled

MrCsabaToth, 2023-12-19:
Currently the bikes I regularly use are not FTMS Indoor bikes (the Stages SC3 is a Power Meter profile, and the Precor Spinner Power Chrono is Precor), so I'll get hold of my Schwinn IC4 to try to reproduce this.

MrCsabaToth, 2025-02-16:
I still haven't been able to reproduce this. In March when I possibly try to test #543 with an IC4 I should test this too.