TrackMyIndoorWorkout/TrackMyIndoorWorkout #458: How to change workout view to show active resistance?

lanes100, 2023-09-03:
Hi, I'm new to this app, but I'm trying to figure out a way to display the resistance being read on the app from my bike. I want to see this so I can use the app for doing peloton rides, and my bike has a 100 step resistance so I wanted to see how I can add this. When I go to the battery & extras page on the bottom right, it says the features are cadence, total distance, resistance leve, expended energy, elapsed time, and power measurement, but on the workout screen all i can see is my output, speed, and RPM. I'd love a way to add the resistance level the bike is showing being read by the app if that's possible.

MrCsabaToth, 2023-09-03:
The app doesn't have this capability right now to display the resistance level. Duplicate #337 #187

I looked for "resistance level" in the app and on the website but I don't see it mentioned like you quote. I guess you are talking about your bike's console or your bike's manual? What kind of bike do you have BTW?

If it's an FTMS bike then Indoor Bike standard has a place for resistance level https://github.com/oesmith/gatt-xml/blob/4fd2ede1d3da9365fdc6dec89290c346581a03f9/org.bluetooth.characteristic.indoor_bike_data.xml#L142, however it depends on the actual console if it transmits that data or not https://github.com/oesmith/gatt-xml/blob/4fd2ede1d3da9365fdc6dec89290c346581a03f9/org.bluetooth.characteristic.indoor_bike_data.xml#L49

As opposed to what steps I describe in #337, the easiest would be if we wouldn't persist the resistance level data, then we wouldn't have to deal with exporting it either or all the shenanigans. So if it's only for display purposes during the workout that would require less development.

Since the order of the metrics is not configurable right now (#179), where would you want to see the resistance level metric on the screen (between which current measurements?).

MrCsabaToth, 2023-09-03:
This ticket would cover Indoor Bikes with the first MVP. Maybe we can cover other sports if their FTMS protocol also has resistance level.

MrCsabaToth, 2023-09-03:
Regarding our codebase, one of the key places are: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/9720aebe2de3788fe0934f6200b167d5d101892c/lib/devices/device_descriptors/indoor_bike_device_descriptor.dart#L55

MrCsabaToth, 2023-09-07:
Hey @lanes100, where would you want the resistance level displayed during workout in order? Current order is: Time, Calories, Power, Speed / pace, RPM / SPM, HR, Distance.
I put down some foundations and I believe I just need to display it now along with a preferences settings which would enable it.

MrCsabaToth, 2024-02-16:
The core data capture code is there, we only need to display it now

MrCsabaToth, 2024-02-27:
Duplicate of #337

closed 2024-02-27 by MrCsabaToth

MrCsabaToth, 2024-04-29:
Releasing this with beta 155 (in some hours hopefully when it passes Google checks and scrunity), the resistance level can be turned on for display in the preferences.

MrCsabaToth, 2024-05-07:
@lanes100 Were you able to test this with the beta by any chance?

MrCsabaToth, 2024-05-07:
[image]