TrackMyIndoorWorkout/TrackMyIndoorWorkout #461: Add support for Technogym Run

MrCsabaToth, 2023-10-06:
This is not the MyRun, but the newest treadmill called simply "Run".
I came across one at SDC23. The trick is that it advertises itself as an FTMS Treadmill, however it never provides standard `Treadmill Data`. It does have the RSC companion sensor service exposed though and that transmits. So we'd need to cherry pick and distinguish this particular treadmill, and then use only that sensor. If the app tries to handle it as a treamdill it stays mute, nothing happens. Due to this behavior standard apps like GymTrakr also mute got tricked and remain mute - I guess waiting for the "main" data as me.

I've been able to test some code (although some Samsung folks were very fearful when they realized I was debugging, next time I shouldn't be such a smart ass and vocal, just do the debug as secretively as possible).

I must add that using the Treadmill with Samsung Health is suuuper easy: just touch the Samsung watch at the NFC location and then it automatically connects and everything is recorded. The FTMS way is not overly hard, but not as convenient for sure.

MrCsabaToth, 2023-10-07:
[image]
[image]

MrCsabaToth, 2023-10-07:
Some logs: [DebugLogs20231006T003451129937Z.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/12837003/DebugLogs20231006T003451129937Z.txt)

MrCsabaToth, 2023-10-07:
nRF Connect screenshots:
[image]

[image]

[image]

[image]

[image]

[image]

MrCsabaToth, 2023-10-13:
I realized I must check model to not collide with Technogym's MyRun trainer, because that's the same manufacturer, but its Treadmill Data service is not mute.

MrCsabaToth, 2023-10-23:
One last piece would be to add a preferences setting with which this workaround could be blocked or enforced. Name could be 'rsc_only_ftms_treadmill'.

MrCsabaToth, 2023-12-06:
Got released with the latest beta (build number 150)

closed 2023-12-06 by MrCsabaToth

MrCsabaToth, 2024-10-08:
After the `Google Cloud Live: Bay Area` (https://inthecloud.withgoogle.com/google-cloud-live-bay-area/register.html at the MTV campus's new Bay View 300 building) I stopped by the SDC24 (Samsung Developer Conference 24). This was possible because in 2024 it was not in the Moscone but rather the San Jose Convention Center (same as where the Intel Innovate was on 2023). Lot of ChargePoint there in the parking lot (at Google you cannot charge as a visitor, or it is extremely cumbersome) and participated in code labs.

Some were IoT / Smart Home. Had to install a driver on a smart hub and could control a Hue bulb. Another setup used Gen AI to phrase to create smart scenes: like if I press that smart button turn that bulb red, or if the window open sensor triggers then turn the bulb on. The Gen AI version deducted these (trigger event + target smart device + action on the device) from human text or voice. I spoke with the engineer of that system and they had two LLMs in that agentic system:
1. Large LLM to interpret and preprocess the human text
2. Small specialized LLM fine tuned to output JSON to write these which translates to the scene scripts.
3. They used sanitized (anonimized) data to generate synthetic training data.

Another set of labs were about Samsung Health. One set up a health study server in a Docker container setup, then deployed a mobile app and a wearable app. The Samsung Galaxy watch was measuring health metrics (heart rate, SpO2, ECG), before that the study was established, the user was enrolled via the mobile app. Then the wear app synced to the mobile app, which synced to the study server, where we could display graphs.

The lab engineers were from South Korea, and this engineer and his boss saw that my pulse was 60. I told them I'll run Tokyo marathon in 2025 and they were excited. Here are some photos:
[image]

ECG data:
[image]

[image]

MrCsabaToth, 2024-10-08:
So about SDC24 https://news.samsung.com/global/sdc24, I'm telling this is because after the labs closed and there was the party and raffles I went into the cordoned area to verify the Technogym Run support (and completed some company training videos while walking). Fortunately it works:
https://www.strava.com/activities/12577086872

[image]

[image]

cagnulein, 2024-10-08:
so they finally put ftms on their device by default?

MrCsabaToth, 2024-10-08:
> so they finally put ftms on their device by default?

Well, this was possible a year ago as well, when I came across the Technogym Run first time (at SDC23), maybe it was exactly this one. Note that the MyRun is different than the Run. Since the FTMS Treadmill Data doesn't have cadence metric, it is usually treated by also exposing an RSC (Running Cadence Sensor) profile. What is special with the Run is that it not only exposes an RSC profile, it doesn't really communicate via the FTMS Treadmill Data. However I can get what I want from the RSC. So a year ago that needed a change because I never considered the RSC as a "primary" or "main" profile / sensor, and I needed to prepare the logic to that scenario.

Maybe I should have double checked that it still doesn't send data through the FTMS Treadmill Data, because you never know when things change due to a firmware update. Over the years as I'm trying to tackle Life Fitness first their own app even wasn't able to handle their own machines, now that works.