TrackMyIndoorWorkout/TrackMyIndoorWorkout #462: Cannot record an FTMS compatible Lifefitness treadmill workout

MrCsabaToth, 2023-10-22:
Gary Vellenzer mentioned this first. Recently at the Hyatt Regency Sacramento I experienced this. Three treadmills advertised 0x1826 FTMS service's Treadmill Data. I was able to connect to them, but they haven't transmitted any data, and after a short while disconnected

MrCsabaToth, 2023-10-23:
Debug Log from my app: 
[DebugLogs20231020T171259375391Z.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/13065545/DebugLogs20231020T171259375391Z.txt)

MrCsabaToth, 2023-10-23:
[image]

Their own app stays zeroed out as well:
[image]

MrCsabaToth, 2023-10-23:
GymTrakr stays zeroed out as well:

[image]

nRF Connect signals well as the Treadmill forcefully disconnects after a short while even though I subscribe to the Treadmill Data stream:

[image]
-
[image]
-
[image]
-
[image]
-
[image]

MrCsabaToth, 2023-11-24:
I faced this issue in the Hyatt Regency Burlingame, CA @ ODSC West AI / Data Science conference. My app saw a bike and a cross trainer. However after many tries I was not able to figure out which machines were those. The hotel had two stair climbers and two ellipticals, but none of those seem to be the cross trainer. There was a Technogym recumbent bike but that didn't seem to be the bike either. I was dumbfounded. I was not able to record any workout with those machines. Screenshots follow.

MrCsabaToth, 2023-11-24:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MrCsabaToth, 2024-05-19:
At Google I/O I was at the Marriott (Santa Clara) and they had a whole bunch of LifeFitness machines in the gym. I made some progress but haven't got it working. I was able to record a few sessions where I connected to the machines via the Life Fitness Android app and did a BT snoop. Based on that I'll mimic what the application does.

1. There's a `0x4b91` proprietary service which is partially the clone of the `0x181c` BT standard User Data Service but with two extra proprietary characteristics (`4cf2` and `e85e`) and probably more proprietary behavior. The app sets several metrics (FN, LN, Email, Age, Gender, Weight, Height, and two proprietary values).
2. There's another service `0x4b90` which seems to be some kind of a Life Fitness Specific Control Point (`58dc` and `69e2` characteristics).
3. The `0x4b90` also has a status characteristic (possibly) `0x51de`
4. There are also communication about FTMS Training Status Client Characteristic Configuration

I have bluetooth snoops I'll attach
[image]
[image]

MrCsabaToth, 2024-05-19:
BT Snoops from Marriott Santa Clara / Google I/O 2024:
[marriott_santa_clara_google_io_2024_btsnoops.zip](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/15367292/marriott_santa_clara_google_io_2024_btsnoops.zip)

MrCsabaToth, 2024-06-21:
Only the Stair Climber has proper FTMS compatible data. The Bike, Elliptical (Cross Trainer), and the Treadmill has shim FTMS (looks like only contains elapsed time and HR optionally) and the meat is communicated in a side channel (characteristic).

For the Stair Climber we'll need to store the elevation gain somewhere, possibly even the floor numbers, maybe in an additional record (somewhat like we'll store extra data for the rowing). And we'd need to also upload that in FIT and other formats and think about what we'll do with the GPS. #163

MrCsabaToth, 2024-06-24:
All this preparation is for a September visit to Anaheim, where I'll stay at the Marriott. I CANNOT WAIT to test the app with the Life Fitness machines there, I already from photos, that they have stair climber, recumbent and upright bike, treadmill, elliptical. They even possibly have a step climber / stair stepper, however the that might not be Life Fitness.

My main secret goal along this visit will be code named "Project Life Fitness", to hone in the support.

Anaheim Marriott also have a Water Rower, so I'll also be able to test the [WaterRower Bluetooth ComModule](https://www.waterrower.com/us/products/waterrower-commodule) support.

cagnulein, 2024-06-25:
@MrCsabaToth did you check the QZ app? It's already working with non ftms devices http://qzfitness.com https://github.com/cagnulein/qdomyos-zwift/

MrCsabaToth, 2024-06-25:
> @MrCsabaToth did you check the QZ app? It's already working with non ftms devices http://qzfitness.com https://github.com/cagnulein/qdomyos-zwift/

qdomyos has extremely impressive list of supported devices! I'm not versed with WaterRower, I don't own one, and I'll only visit this place for a few days in September. However I'd like to prepare. Are you saying that if the S4 console doesn't seems to have the ComModule (does that provide FTMS standard Rower Data BTW?), then the built-in S4 Bluetooth smart protocol is proprietary? I'm looking at https://github.com/cagnulein/qdomyos-zwift/tree/master/src/devices but I don't see a separate folder for WaterRower. I have been planning to look at qdomyos for years now BTW: #238 

As far as Life Fitness goes I'll look at your version of treadmill support (https://github.com/cagnulein/qdomyos-zwift/tree/master/src/devices/lifefitnesstreadmill). From my Bluetooth snoop logs I determined that the Indoor Bike data of their upright and recumbent bikes also deviates from the standard FTMS, check out https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/develop/lib/devices/device_descriptors/life_fitness_bike_descriptor.dart (Instant Cadence is not indicated properly int he flags) although I must verify my findings again.

MrCsabaToth, 2024-06-26:
Code from qdomyos-zwift
```
    QBluetoothUuid _gattWriteChar1CustomService1(QStringLiteral("5da558dc-9cb2-11e5-8994-feff819cdc9f"));
    QBluetoothUuid _gattWriteChar2CustomService1(QStringLiteral("5da569e2-9cb2-11e5-8994-feff819cdc9f"));
    QBluetoothUuid _gattWriteChar3CustomService2(QStringLiteral("5da54cf2-9cb2-11e5-8994-feff819cdc9f"));
    QBluetoothUuid _gattWriteChar4CustomService2(QStringLiteral("ce78e85e-9cb6-11e5-8994-feff819cdc9f"));
    QBluetoothUuid _gattWriteCharControlPointId((quint16)0x2AD9);
    QBluetoothUuid _gattTreadmillDataId((quint16)0x2ACD);
    QBluetoothUuid _gattCrossTrainerDataId((quint16)0x2ACE);

    uint8_t initData1[1] = {0x01};
    uint8_t initData2a[20] = {0x38, 0x66, 0x65, 0x64, 0x61, 0x38, 0x38, 0x39, 0x31, 0x64,
                              0x62, 0x61, 0x34, 0x30, 0x31, 0x66, 0x38, 0x39, 0x39, 0x30};
    uint8_t initData2b[12] = {0x30, 0x32, 0x33, 0x30, 0x37, 0x39, 0x35, 0x66, 0x30, 0x38, 0x36, 0x30};
    uint8_t initData3[1] = {0x00};
    uint8_t initData4[7] = {0x00, 0x00, 0x00, 0x01, 0xb8, 0x5b, 0x5d};
    uint8_t initData5[1] = {0x02};

    if (gattWriteChar4CustomService2.isValid()) {

        writeCharacteristic(gattCustomService1, gattWriteChar1CustomService1, initData1, sizeof(initData1),
                            QStringLiteral("init"), false, false);
        writeCharacteristic(gattCustomService2, gattWriteChar3CustomService2, initData2a, sizeof(initData2a),
                            QStringLiteral("init"), false, false);
        writeCharacteristic(gattCustomService2, gattWriteChar3CustomService2, initData2b, sizeof(initData2b),
                            QStringLiteral("init"), false, false);
        writeCharacteristic(gattCustomService2, gattWriteChar4CustomService2, initData3, sizeof(initData3),
                            QStringLiteral("init"), false, false);
        writeCharacteristic(gattCustomService1, gattWriteChar2CustomService1, initData4, sizeof(initData4),
                            QStringLiteral("init"), false, false);
        writeCharacteristic(gattCustomService1, gattWriteChar1CustomService1, initData5, sizeof(initData5),
                            QStringLiteral("init"), false, false);
    }
```

MrCsabaToth, 2024-09-25:
Even though the Anaheim Marriott trip (AGU conference) didn't happen (#510), I'm at a DoubleTree now and I debugged some. The FTMS Service data based sport match logic (related to the `sportsMatch` boolean introduced for LifeFitness for this purpose) needed some TLC.

One more quirk: LifeFitness advertises standard manufacturer name characteristic, which we try to utilize because we don't get any manufacturer from the advertisement data. However this read after the connection happens errors out. So we'll need to introduce another boolean which signals the `FitnessMachine` to not touch that characteristic even if it's there.

With that I recorded workouts on the recumbent bike. The elliptical and the treadmill gets stuck during the connection phase.

A final note is that after the workouts were over on the recumbent the bike console told me that the workout is sent. However we initialize LN, FN, and email with dummy empty values. It'd be better to behave nicely otherwise LifeFitness may change stuff. So maybe we should indeed use the LN, FN, and Email configured in the preferences.

MrCsabaToth, 2024-09-26:
Trying various delays, full FN, LN, email. Maybe the other secret as well later?

When someone tries first time and the Android native Bluetooth pairing pops up (for audio) then the attempt fails at the last name step for some reason.
After that consecutive ones are successful.

The athlete should not move and should let the app configure all the way until the measurement screen is ready. Then I select quick start.

Between consecutive workouts someone should exit the app right now.

The treadmill data interpretation needs tuning. It won't have spm.

MrCsabaToth, 2024-09-26:
Trying various delays, full FN, LN, email. Maybe the other secret as well later?

When someone tries first time and the Android native Bluetooth pairing pops up (for audio) then the attempt fails at the last name step for some reason.
After that consecutive ones are successful.

The athlete should not move and should let the app configure all the way until the measurement screen is ready. Then I select quick start.

Between consecutive workouts someone should exit the app right now.

The treadmill data interpretation needs tuning. It won't have spm.

And this hotel doesn't have a stair stepper, hopefully one day I come across one