TrackMyIndoorWorkout/TrackMyIndoorWorkout #470: Ability to control speed and inclination of a Treadmill

MrCsabaToth, 2023-12-22:
As mentioned in https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/176#issuecomment-1684503451 now that we use the Control Point anyway to signal the start and stop of a workout, we might as well add support for speed and inclination. (In case of bike that would be resistance level).

There are two main things:
* Develop the UI, where and how the controls would be
* Liability! I don't want to get sued if someone falls

This is requested explicitly by Will Howell who has a Horizon Fitness 7.0AT treadmill.

MrCsabaToth, 2023-12-28:
We'll first need to query for the inclination and speed range to know the limits. We should also watch that constantly in case the athlete manually adjusts it

MrCsabaToth, 2023-12-31:
I don't have such a bike or treadmill which can be controlled by the FTMS control point or have servo motor to do adjustment upon command, so I'll need some help for this.

MrCsabaToth, 2024-01-28:
#458 will allow the display of the resistance level. Possibly we could implement resistance level up/down for bikes as well. Similarly to the resistance level the treadmill inclination should be a display only thing, we won't persist it for now because the generated GPS doesn't have any elevation change either, so it wouldn't match up with any inclination setting. Maybe one day if we have something if we add straight up climb GPS for stair and step machines #163, but until then we won't persist.

Ketec, 2024-04-02:
I was just looking into my Hammer Qvadis 7.0 to see if I can make it more useful. The default quick speed controls etc are not useful.

I connected with nRF Connect and found it does use the FTMS. I played around with the values according to the BT standards manual.
It supports speed and inclination control and reports the step, min and max for each.

However, I did run into a bit of an issue with speed control - the manual says it's uint16. But passing the lowest value of 1 sets the speed to 2.5 ( while it supports a step of 0.1 with a minimum of 0.3).

So I wonder if either they app has an issue or it's actually not using uint16 - but a bytearray with a smaller value?

Ideally the goal would be to able to add shortcuts for quick speed (like interval training) and/or with inclination to match.

MrCsabaToth, 2024-04-02:
@Ketec good info since I don't have such treadmill!

1. Can you see if your machine exposes the "Supported Speed Range" endpoint: https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.supported_speed_range.xml (I see you mention "while it supports a step of 0.1 with a minimum of 0.3" so you may have polled 0x2AD4? Never hurts to poll one more time.) Essentially someone cannot just willy-nilly set anything. It has to be within the supported boundary (also for liability purposes), and also the increments cannot be lower than the given. I assume the minimum increment will be 0.1 mph or similar.
2. When you tried to set it outside of a boundary (maybe some bad conversion or division) then did possibly get back a `0x09` or `0x0A` training status code indicating error (`0x09 Speed Outside of Control Region - Low (increase speed to return to controllable region)` or `0x0ASpeed Outside of Control Region - High (decrease speed to return to controllable region)`)? See Table 4.13: Training Status Field Definition. This code comes back in the 0x2AD3 Training Status characteristic 0x2AD3 (https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.training_status.xml). Try hook up to that one and perform the setting. I never tried to set by nRF Connect.
3. I also wonder what data you get back for the "Supported Inclination Range" endpoint 0x2AD5: https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.supported_inclination_range.xml

This will be beneficial once I put controls on the app. I see in my source code I don't even have the training status yet.

Ketec, 2024-04-02:
[image]
[image]
[image]

For config. Will log data separately with different interactions.

Ketec, 2024-04-02:
```
nRF Connect, 2024-04-02
FS-6041F1 (9F:55:37:54:96:98)
V	23:33:03.631	Reading characteristic 00002acc-0000-1000-8000-00805f9b34fb
D	23:33:03.631	gatt.readCharacteristic(00002acc-0000-1000-8000-00805f9b34fb)
I	23:33:03.735	Read Response received from 00002acc-0000-1000-8000-00805f9b34fb, value: (0x) 5C-16-00-00-03-00-00-00
A	23:33:03.735	"Fitness Machine Features:
Total Distance Supported
Inclination Supported
Elevation Gain Supported
Step Count Supported
Expended Energy Supported
Heart Rate Measurement Supported
Elapsed Time Supported

Target Setting Features:
Speed Target Setting Supported
Inclination Target Setting Supported" received
V	23:33:10.281	Reading descriptor 00002902-0000-1000-8000-00805f9b34fb
D	23:33:10.281	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
I	23:33:10.366	Read Response received from descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 00-00
A	23:33:10.366	"Notifications and indications disabled" received
V	23:33:13.631	Reading characteristic 00002ad3-0000-1000-8000-00805f9b34fb
D	23:33:13.631	gatt.readCharacteristic(00002ad3-0000-1000-8000-00805f9b34fb)
I	23:33:13.725	Read Response received from 00002ad3-0000-1000-8000-00805f9b34fb, value: (0x) 01-01
A	23:33:13.725	"Status: Idle
Details: " received
V	23:33:16.656	Reading descriptor 00002902-0000-1000-8000-00805f9b34fb
D	23:33:16.656	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
I	23:33:16.757	Read Response received from descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 00-00
A	23:33:16.757	"Notifications and indications disabled" received
D	23:33:16.761	gatt.setCharacteristicNotification(00002ada-0000-1000-8000-00805f9b34fb, false)
V	23:33:16.763	Notifications and indications disabled for 00002ada-0000-1000-8000-00805f9b34fb
V	23:33:19.633	Reading descriptor 00002902-0000-1000-8000-00805f9b34fb
D	23:33:19.633	gatt.readDescriptor(00002902-0000-1000-8000-00805f9b34fb)
I	23:33:19.724	Read Response received from descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 00-00
A	23:33:19.724	"Notifications and indications disabled" received
D	23:33:19.729	gatt.setCharacteristicNotification(00002ad9-0000-1000-8000-00805f9b34fb, false)
V	23:33:19.731	Notifications and indications disabled for 00002ad9-0000-1000-8000-00805f9b34fb
V	23:33:24.373	Reading characteristic 00002ad4-0000-1000-8000-00805f9b34fb
D	23:33:24.373	gatt.readCharacteristic(00002ad4-0000-1000-8000-00805f9b34fb)
I	23:33:24.467	Read Response received from 00002ad4-0000-1000-8000-00805f9b34fb, value: (0x) 1E-00-98-08-0A-00
A	23:33:24.467	"Minimum Speed: 0.29999998 km/h
Maximum Speed: 22.0 km/h
Minimum Increment: 0.099999994 km/h" received
V	23:33:26.483	Reading characteristic 00002ad5-0000-1000-8000-00805f9b34fb
D	23:33:26.483	gatt.readCharacteristic(00002ad5-0000-1000-8000-00805f9b34fb)
I	23:33:26.596	Read Response received from 00002ad5-0000-1000-8000-00805f9b34fb, value: (0x) 00-00-78-00-0A-00
A	23:33:26.596	"Minimum Inclination: 0.0
Maximum Inclination: 12.0
Minimum Increment: 1.0" received

```

Log of those data pulls

MrCsabaToth, 2024-04-02:
In my local dev environment I already have the resistance display (even though that's rather relevant to bikes, or certain other machines). I plan to add steps / revolutions / strokes #303.

For this ticket I'd need to add a few things:
0. Settings to turn on speed target setting feature
1. Settings to turn on inclination target setting feature
2. Settings to turn on resistance setting feature
3. Settings to turn on power setting feature
4. I already have a speed display
5. Speed up / down buttons. I guess either in a new display row or on top of the accordion section
6. Resistance up / down buttons. Just like for speed up / down: new display row added or buttons on top of the graph in the accordion. Builds on top of newly added #458
7. Inclination up / down buttons.
8. Power up / down buttons - this may need a special power controlled workout mode support, so maybe later
9. Heart rate up / down buttons - this may need a special ERG workout mode (heart rate controlled) support, so maybe later

Note: if the app is in simplified UI mode #33 then there's no accordion, so we'd need to handle that corner case - add accordion just make it empty without the graphs?

MrCsabaToth, 2024-04-02:
@Ketec nice, and do you see 0x09 or 0x0A coming in the Training Status when you try to set the speed?

Important note 1: even though the minimum increment is 0.1 km/h, the speed uint16 has a resolution of 0.01! So it's one magnitude off. Keep this in mind, because if you only divide your speed by 10, then you can unintentionally try to set 10x higher speed (say 36 kmh instead of 3.6 kmh).

Important note 2: if you set 1 as the uint16 speed, that's 0.01 kmh, which is below the minimum 0.3 kmh.

Question: when you say the speed was set to 2.5, is that kmh or mph? I guess you set it too low and the machine just either defaulted to a reasonable value, or you were already at that walk speed and nothing changed?

Ketec, 2024-04-02:
Will collect a log with trying different t changes tomorrow.

The machine is kmh(I think this is in bt standard?).

I tried uint16 big endian values and 1 is 2.5, 2 is 5.4, 5 was over 12km.

The Inclination jumped to 25 with value 1 (which is out of bounds so it maxes at max height).

There is also that write permisdion request you need to do o ce per session.

So it could be that the tool sends wrong data I guess.

Some various logs I captured while trying.

[Log 2024-04-03 00_14_26.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14843875/Log.2024-04-03.00_14_26.txt)
[Log 2024-04-03 00_12_49.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14843876/Log.2024-04-03.00_12_49.txt)
[Log 2024-04-03 00_06_07.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14843877/Log.2024-04-03.00_06_07.txt)

MrCsabaToth, 2024-04-03:
@Ketec all of the multi byte values are little endian. I still don't know 100% what do you mean by setting 1, but from that latest description it sounds like you set the MSB (Most significant byte) to 1, which would mean 256. Due to the 1/100 precision that actually means 2.56 km/h, so 2.5 sounds about right - given that the machine does a truncation and not rounding up to 2.6. Value 2 in the MSB means 512, which should come out to 5.1 km/h. 5 would mean 5*256 = 1280, divided by 100 -> 12.8 km/h.

So it almost seems like to me you are mistreating the little endian uint16 as big endian, and now these values almost make sense. Similarly with inclination.

MrCsabaToth, 2024-04-03:
It depends on the machine, but if you want to follow full protocol, then you'd want to request control first https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/c814af14ae59be670bdbada56c7942b3b565accf/lib/devices/gatt/ftms.dart#L109, then also signal workout start to the machine. Maybe that'll help. Make sure everything stays connected, and nothing signals workout pause or stop (both via Bluetooth FTMS or on the physical console). Request control is done on the 0x2AD9 FTMS Control Point: 
My app already does these three:
1. Request control https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/c814af14ae59be670bdbada56c7942b3b565accf/lib/devices/gadgets/fitness_equipment.dart#L655
2. Signal workout start (or resume): https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/c814af14ae59be670bdbada56c7942b3b565accf/lib/devices/gadgets/fitness_equipment.dart#L1297
3. Signal workout stop (or pause): https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/c814af14ae59be670bdbada56c7942b3b565accf/lib/devices/gadgets/fitness_equipment.dart#L1313

Ketec, 2024-04-03:
I know the elevation worked fine when I tried it with Kinomap - it tried to match it to the run height.

Start/stop worked through swift and this app too.

Likely also need to call reset at certain times to clear all settings.

It will accept speed when stopped, displays it and sets to 0 in a few seconds. Start can also just switch to instantly paused mode - but likely it had some invalid settings already and needed a reset.

MrCsabaToth, 2024-04-04:
That's good, maybe we don't have to differentiate. I'll need some time to verify this finding in the original Russian hacker code, maybe I haven't ported it over properly from C / C++, that would be the best case.

MrCsabaToth, 2024-04-08:
> Also, if we are adding more sections, perhaps it's time for preferences to show/hide them. There will be a lot of equipment specific fields.
> 
> Or a simple equipment profiles, hiding irrelevant fields.

100%. I added the resistance level (not released yet), but it'll have a show / hide preferences option. Each field I'll add will have one. Long term it'd be great to make the UI fully configurable (order items, show/hide): #179. Back in the past I tried that but it was super slow. Maybe after I refactor the state handling to RiverPod or similar.

Evgeny105, 2024-12-17:
Hi,

First of all, I want to express my appreciation for the TrackMyIndoorWorkout app (Android). It works excellently with my SVENSSON BODY LABS FURY SPIN trainer, and I’m especially pleased with how it captures all the training data and records it into files.

I have a suggestion that I believe could enhance the user experience even further. My trainer supports the FTMS Bluetooth protocol and offers 16 resistance levels. It would be fantastic if the app could implement predefined workout programs, where the resistance could automatically adjust based on a specified power, heart rate, or speed profile (i.e., a graph). This would enable users to follow structured training sessions without manually adjusting the resistance throughout the workout.

Since the app already supports controlling the trainer (start/stop), I believe this could be a relatively small addition, but it would greatly improve its functionality for users who prefer guided workouts.

I am a Python developer myself and would be happy to help with the implementation of this feature. However, I would need some assistance in understanding the structure of the project and identifying where the feature could be integrated.

Best regards, Evgeny

MrCsabaToth, 2024-12-21:
Hey @Evgeny1051 this would be fantastic. You are right that #176 laid the foundation for Control Point Support. But your request has several stages:

1. Providing Bluetooth / engine logic for speed and inclination setting / read. There are some parts in the code already, for example we poll what features the fitness machine supports.
   - Sometimes the feature bits can be bogus / mixed up
   - Write the functions in a way that it fits next to current logic. May require some partial refactoring.
2. Providing UI for the speed and inclination manual control.
   - Then we can decide when when to display the buttons
   - Where to display the buttons
3. Design a UI where someone can configure the profile of the workout. Possibly active and rest periods could alternate. There could be formal warmup and cooldown. This requires separate pages with some dialogs for some kind of a CRUD configuration. This is ticketed and some of my thoughts are there: #119
4. There can be even more advanced scenarios and needs, ticketed #400

This would be great to implement. We can go step-by-step. I can open a feature branch for you. I'm very busy, but let's chat on Slack: https://join.slack.com/t/trackmyindoorworkout/shared_invite/zt-2wnrmeeee-tTxzcF~pw9HNfNMlAFXy6w

Evgeny105, 2024-12-23:
Sadly, Slack doesn't work in my country.....
Can we communicate on this topic here, or, for example, in Telegram?

To begin with, I am studying the project. I realized, of course, that I am not an expert in Dart). I am still trying to understand the structure of interaction with the equipment. The application successfully sends stop/start commands to the trainer. It also shows on the screen information about the resistance level set on the trainer in the field with the icon “123”. 

I of course also have a serious workload with work and family. I think I need to try to figure out how to debug such apps in general, since I have no experience developing for mobile operating systems.

MrCsabaToth, 2024-12-25:
@Evgeny105 does Discord work? Maybe I should establish a Discord channel. I have Skype, Teams, and Google chat as well. I haven't used telegram yet, but I can try especially if it is possible on the desktop. I can show you around in the code.