TrackMyIndoorWorkout/TrackMyIndoorWorkout #484: Suggestion.. option to change units

Lauropy, 2024-02-05:
It would be great if there was an option to change displayed units from Km to Miles, and kph to mph

MrCsabaToth, 2024-02-06:
Hey @Lauropy, please click on the cog icon in the menu to get to the preferences. Go into the `UX Preferences`. Below the `Theme Selection` there's a `Unit System` checkbox. If it's off then it's imperial (mp/h, miles), when it's checked it's metric (km/h, meters or km). Right below that there's also a resolution checkbox: if the distance should be miles / km (low resolution) or meters / yards (high resolution). Let me know if it works.

Question: which country are you on and what locale Play Store are you using? I'm asking because the app determines the state of the Unit Type checkbox based on the Play Store locale: see #265
Right now US and UK gets set to imperial, every other locale gets metric. If in a European Country someone uses US custom ROM (happens with my twin brother) then the Play Store locale is US so he gets imperial by default. There can be flaws in this logic so please let me know.

MrCsabaToth, 2024-02-06:
In case this is just a preferences miss then the main issue is lack of documentation.

MrCsabaToth, 2024-02-11:
@Lauropy any news?

Lauropy, 2024-02-11:
I wasn't aware of the configuration option - I've never come across the term UX before and didn't know this meant "general application configuration".

So, I'm happy now with the app and this can be closed.

closed 2024-02-11 by Lauropy

MrCsabaToth, 2024-02-11:
@Lauropy That's a great feedback! UX means User Experience, but I haven't thought about a better naming. I should call it "User Interface"!