TrackMyIndoorWorkout/TrackMyIndoorWorkout #492: Supported KayakFirst ergometers

Abasz, 2024-03-09:
We have a rather old KF machine that is orange. I wonder whether that is supported?

Thanks

MrCsabaToth, 2024-03-10:
The support covers the blue colored ergometer models. You got it right that the orange ones belong to an earlier generations and their console uses different firmware and protocol compared to the blue (as far as I know). To clarify, when you say orange, those look like these?
[image]

I don't have access to orange ones (and only have access occasionally to blue ones when I'm in Hungary, last time in the winter, and maybe late summer, or fall next time). However you have a GitHub account and you look like not technically challenged, so I'm curious about a few things.

1. Does the "old" official Android app for the blue models able to handle the orange ones? You can find the app in APKPure for example: https://apkpure.com/kayakfirst-paddle-app/com.kayakfirst.e2. If yes then maybe I can muck around in the decompiled jars of that apk.
2. Plus if that app supports it then it could also help to do a bluetooth snoop of a short session with that app, so we can see how the protocol goes.
3. What do you see when you scan the device with nRF Connect? What Bluetooth services and characteristics you see? Unlike some FTMS machines unfortunately probably the workout data won't start to magically flow. But who knows, maybe you can try to attach to characteristics which are notification types and see if any stream is coming in.

If that app doesn't handle it then the road will be very rocky. We'd need to reverse engineer. I might try to reach out to some engineers, but when I was developing I was happy that I could get the blue ones working.

(BTW, if you have blue ones as well I wonder if my app works with them. The initialization phase takes many seconds but then it should work well. Plus you can pair a HRM to my app as well. Plus in the future I plan to establish an FTMS companion app with which you can convert the protocol to FTMS and so transmit to Kinomap and similar, but that will require a good amount of work.)

Abasz, 2024-03-10:
Hi Csaba!

Yes you are correct the ones on the picture (I think those are literally the exact machines I use(d) that are on the picture as I believe it is taken in the boat house of UTE :)) There is also one blue (not bull but one generation before that) now.

> Does the "old" official Android app for the blue models able to handle the orange ones? You can find the app in APKPure for example: https://apkpure.com/kayakfirst-paddle-app/com.kayakfirst.e2. If yes then maybe I can muck around in the decompiled jars of that apk.

Yes it does support it and I have decompiled the original kayak first app before (for the purpose of getting some information how the calculation of the metrics work). To my surprise after looking at your app code, I realized that it is significantly different from the blue one (hence was the question). Actually the firmware of the orange MCU only measures the delta times between impulses and the calculation of the metrics are done in the app in line with the [Physics of Ergometers](https://eodg.atm.ox.ac.uk/user/dudhia/rowing/physics/ergometer.html). For the blue ones based on your code, the calculation is done on the MCU rather than on the app and the ready calculated metrics are broadcaster.

> Plus if that app supports it then it could also help to do a bluetooth snoop of a short session with that app, so we can see how the protocol goes.

In terms of bluetooth it uses the same method as the blue ones i.e. serial BLE protocol with a hc-05 serial bluetooth module. I can connect to it with the [Serial Bluetooth Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=hu&gl=US) also I am able to receive the data (delta time at the start and end of the drive and the recovery I think).

> What do you see when you scan the device with nRF Connect? What Bluetooth services and characteristics you see? Unlike some FTMS machines unfortunately probably the workout data won't start to [...]

I have to check the UUIDs but I believe its the same BLE serial. Based on my previous test and by looking at your code I believe commands are the same (though when I was testing I did trial and error with the limited help of the decompiled java code of the apk that is... not pretty to say at least :)

> (BTW, if you have blue ones as well I wonder if my app works with them. The initialization phase takes many seconds but then it should work well. Plus you can pair a HRM to my app as well. Plus in [...]

I will try the app with the blue shortly, though as the winter season is ended we will work on the water rather then the erg :) 

Before you start putting significant effort into the orange, I wanted to give you some background for my question: 

There are two consideration:

1. I have developed a monitor that uses superior math in terms of metric calculation (ESP Rowing Monitor) but uses the same method i.e. the rotation of the flywheel (the principals are similar to the KF, KayakPro Concept2 etc. but the algorithm is significantly more robust). The math was developed (and still being refined) in the Open Rowing Monitor project (that is for Rpi) and I ported the algorithm to an MCU. Currently my monitor uses wifi and a web gui because I dont have any android/mobile dev experience. This means that I have to run a simple web server on my phone to use my monitor (that is not the best solution :)). So I was looking for an app that is could bridge this gap and yours seemed to support a lot of protocols 
2. In order for the metrics to be correct there are some physical properties that are necessary (like inertia of the flywheel, or the gear sprocket radius, the magic constant that converts the work to actual distance). In the original KF apk I was able to find these and I assumed they apply to the blue as well. But when I found you app I realized it does things differently which made me thinking that maybe the blue has different physical properties. So I wanted to find out whether you have any code specific to the orange (for which I have the answer now).

Due to the above two I was wondering whether you had any experience with the orange. But based on the above (and the fact that you dont have experience with the orange) it seems that your app simply consumes the metrics that has already calculated by the MCU of the KF.

I dont want to send you on a wild goose chase for the orange that are not in production :) Rather I wonder whether you would be willing to discuss how ESP Rowing Monitor could be compatible with your app :) I mean its clear that I could implement the FTMS protocol into ESP Rowing Monitor (actually there is a fork that has done this) and then call it a day, but there are a few consideration: 
- one of the best feature is the force curves (this shows you a curve on the drive that helps refine your technique) this is not supported by FTMS
- the recovery and drive time data that shows if one paddles asymmetrically (not supported by any BLE profile)
- garmin watches dont support FTMS not to mention that if I connect the monitor to the phone via bluetooth it will not be able to connect to the watch at the same time (this is actually the reason EPS Rowing Monitor uses CPS and CSC profiles so with a little hackery of the wheel circumference  setting and using bike activity it is compatible with these watches).

MrCsabaToth, 2024-03-10:
Yes, this is the UTE kayak club. (KayakFirst's place in Budapest was in the Csepel area whereas I reside in Óbuda and MULTI SE was branched off of UTE where I started in 1989 so I reached out to UTE and coaches were supportive and let me show up for some debug sessions). I wish I had known you when I had those two debug sessions recently. I would have tried the orange ergs BTW, but when I peeked at them I didn't see any consoles:
[image]

(So no need to test with the blue, I've done that, I also tested the two sliding knee canoe ergs downstairs. There's also a catch that the power cable is gone, I used the magnetic cable I got with my engineering sample console. I did several hundred sessions with that engineer console (which is only the console) so I can hone in on everything. First I got it working when you manually have to help my app a little and click the Free Paddle button, and was not ready for the FIBO expo 2023 April which was a missed chance. Then I finally nailed the full procedure (not requiring any tap on the console) by extra session with my engineering console + finally the two in-person debug days.)

You got it right, the blue machine's MCU does the calculations for me. I had two major challenge with it:
1. It was the first fitness machine which used serial protocol. All other machines (including non FTMS such as proprietary Schwinn 170 / 270 / 570 or Spinner Power Chrono) supply binary bytes, so data packets fit in 10-25 bytes and there's no need for new lines (which is not even a thing in binary, but the blue console even has a 20 bytes MTU which was also interesting).
2. It requires some extensive multi-step initialization which takes many seconds, and that's why I introduced a bottom bar progress, which has multi stage for KayakFirst. (This proved to be useful for other machines as well, since I'm notifying now vie the FTMS Control Point about activity start and stop and things may take a couple of seconds.)

You are also right that FTMS do not support detailed power curve or most of the time differences between left and right side. The only paddle sport in FTMS is the Rower Data, and since rower ergs don't have dual flywheels or independent left and right paddles, it's a one dimensional movement. KayakPro's Genesis Port FTMS console implements Rower Data. (I have to mention it supports the FTMS spin-down protocol for internal calibration: you paddle the flywheel to a speed range and then let it spin off and the firmware measures how fast it spins down and can calculate the loss off the whole mechanics (bearings, etc).

The only more sophisticated protocol I know (and also support) is the original non FTMS C2 Rower protocol. It was developed before the FTMS was standardized though. One quirk of that protocol is that it has quite many characteristics, but some of those can provide detailed force curves. So you may consider implementing that and then you might leverage C2's facilities to log, store, and display such detailed data.

In any other case you'd loose that, since I don't think fitness portals have a place to store that or display. One of the main motivation behind my app (besides being able to connect to fitness machines and record the workouts) was integration: upload the workouts to fitness portals directly (Strava, SUUNTO, MapMyFtiness / Under Armour, Training Peaks) or indirectly via FIT / TCX exports (such as Garmin did not provide me upload API - see https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/5#issuecomment-1332410218 and https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/5#issuecomment-881231706, or Saucony, Adidas, and a list goes on and on).

For this reason I only focus on a few metrics to persist:
Cumulative metrics:
* elapsed time
* moving time
* distance
* burned calories

Non cumulative metrics:
* speed or pace
* cadence
* power
* heart rate

So you say that the orange one would send me the flywheel pulses also in a serial format? I'd like to support that. Can you give me some code snippets in any languages (C / C++ if it's an ESP MCU I guess, or maybe Java)? I should even skip the "old" app's Java code if your method is superior!

At the same time I also want to support your ESP based project: because that could still provide an achievable solution to everyone potentially - it wouldn't be a one-off.

As I understand your ESP exposes a CSC (https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.csc_measurement.xml) and a Power Meter (https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.cycling_power_measurement.xml) profile. It'd be great if I can identify your ESP by manufacturer name and possible advertisement data. That way I'd know that even though I see Cycling sport profiles but they are really for another sport. This would not be the first time:
* Stryd foot pod exposes a Cycling Power profile because the RSC didn't provide good place for Watt data.
* The Old Danube kayak erg (Hungarian, but Viktor Glatz in Szentendre) simply attaches a speed sensor on the axle of the flywheel and a cadence sensor at the center of the paddle. Turns out that the cadence sensor can deal with the different acceleration profile of a kayak paddle shaft (vs a cycling crank arm) and pick up the cadence. Since there are wide brand and variety of very cheap speed + cadence sensor pairs, my app has a special switch in the Equipment preferences "Paddling with Cycling sensors" and then a cor [... cut ...]
If it comes to extra data like left/right balance I can imagine that I might be able to display some of these extra metrics real time but don't persist. In fact soon I'll display the resistance level that way (#337), or also steps / strokes / revolutions (#303). Maybe one day I can also add power curve or left / right balance or both?

One advantage I could see if I support your ESP is that if that supports other ergs then transitively I will as well. My app's mission "preventing fitness machines from becoming laundry drying racks" points beyond my app. Let me know your thoughts.

MrCsabaToth, 2024-03-11:
I'll extend the code changes for the Old Danube #384 so Cycling Power meters could be also used for paddling sports (kayak for now).

Note, that when someone use these sensors (speed sensor, cadence sensor, power meter) my app doesn't immediately shuttle to the recording screen, because it doesn't know what other and how many more sensors you'd want to connect to. For example in an extreme case someone can pair 4 sensors: power meter, speed sensor, cadence sensor, heart rate monitor.

Abasz, 2024-03-11:
> Yes, this is the UTE kayak club. (KayakFirst's place in Budapest was in the Csepel area whereas I reside in Óbuda and MULTI SE was branched off of UTE where I started in 1989 so I reached out to UTE [...]

They are slightly hidden, this is how the board looks like. This machine does not have a display only works over the app via BLE.

[image]
[image]

They work of a 9V barrel jack that is located at the top. Sensor is an unipolar hall sensor. It has the HC-05 BLE Serial module that is connected to the serial UART of the MCU. I've attached an FTDI programmer so I could listen to the Serial interface (this is how I was able to understand the commands and then verify from the APK).

>So you say that the orange one would send me the flywheel pulses also in a serial format? I'd like to support that. Can you give me some code snippets in any languages (C / C++ if it's an ESP MCU I guess, or maybe Java)? I should even skip the "old" app's Java code if your method is superior!

Its important to clarify that ESP Rowing Monitor does not use the data of the KF MCU. It measures signals directly from the hall sensor (via interrupt of the  ESP32). Its a totally independent from other implementation. All it needs is a hall sensor and magnets on the rotating flywheel. Actually the algorithm uses the delta time between two consecutive signal. From that point metrics are calculated by using quadratic and linear regression models fitted to the data poinst received. Drag factor (to which Kayak Pro refers to as spin down) is actually determined and calculated on every stroke, hence there is no need for specific action "calibration".

So basically it functions as a server, what it is missing is the client. Currently it has two ways to connect that is concurrent: 
1. BLE (e.g. to connect a garmin watch to record the activity). This is limited to the standard BLE sport profiles (CSC, CPS for now, but FTMS can be added quickly, though that is not supported by Garmin watches which I use so probably I would stick to CPS)
2. Wifi where a websocket server is implemented on the ESP32 that broadcasts to client (here I made a web GUI where one can enter the IP address of the ESP32 and that handles all metrics, shows the force curve as a graph etc.)

The issue is that the Wifi and web GUI approach is bit crappy (to have all the functions it requires workarounds on the user side)

> In any other case you'd loose that, since I don't think fitness portals have a place to store that or display. One of the main motivation behind my app (besides being able to connect to fitness [...]

Actually you are right about the persistence. Currently the way sessions can be persisted is
1. record via BLE and a fitness watch (this is limited to the data available in the respective Bluetooth profile).
2. at the end of the session the data can be exported to JSON from the GUI (I process that in excel). Here every information is saved (i.e. handle force values and drive and recovery data, as well as data that would be available in the CPS via BLE). Its not an ideal persistent method and not user friendly. Actually my longer term plan here is to extend the webGUI code to be able to process a previously downloaded session data and allow iterating through and compare etc. 

>(such as Garmin did not provide me upload API)

You dont need a personal one, actually they have an API, this is how connect works, you just need to imitate a full auth session, its pretty easy to imitate this actually (though sometimes they brake it which halts things for a couple of days). There are project in all languages for instance here is one in C#: https://github.com/lswiderski/yet-another-garmin-connect-client. I have an implementation in JavaScript but that does not support 2FA.

> It'd be great if I can identify your ESP by manufacturer name and possible advertisement data. That way I'd know that even though I see Cycling sport profiles but they are really for another sport.

This is a setting in the implementation that can be changed before compiling, which means there is flexibility.

So basically the initial questions are: 
1. Would you be able to add new metrics to your UI, including a chart for the force curve (like [here](https://github.com/Abasz/ESPRowingMonitor-WebGUI/blob/master/docs/imgs/ESP-Rowing-Monitor-WebGUI.jpg)), even without persisting/uploading them on the long term?
2. Would your app allow export to a json (or xml or anything that is standard) with all the session data, including the force curve data (actually its possible and rather easy to add custom fields to a FIT file, so technically all data can be added to a fit file - it will be a valid processable fit file, only fitness sites will ignore the unknown fields - only question is if it supports array like fields). 
4. I will need to check if ESP32 can allow/handle concurrent clients (so both the phone and a watch can connect to it via BLE and I can drop the Wifi). I think it should but I am not sure weather it requires complicated coding or works out of the box. My thinking here is that I would keep the CPS / CSC (and possibly FTMS) advertisement but add an extra service that broadcasts the additional data, so fitness watches can connect, but your app can connect simultaneously. Theoretically, should allow every consumer to read the data it needs: watches can read the standard BLE profile, and TMIW when sees this new service can add the extra metrics (hence differentiate between device as well).

MrCsabaToth, 2024-03-12:
I know this might not be the main stream, but I made some code changes, so the "Paddling with cycling sensors" mode (can be switched in the preferences) doesn't only affect CSC sensors, but now Power Meters as well. Can you test a custom APK if it works well both with the CSC and the Power Meter profile your ESP exposes? https://drive.google.com/file/d/1bWmpcZlOIaN6wQmldydE5a3wMKtxyEtC/view?usp=sharing

Regarding the initial questions:
1. I think it's possible to display extra data. Needs some work, maybe I can start with total strokes 
2. The export is a whole other animal because that would mean persistence.
2.1. By few metrics I guess you mean the drive and recovery, and also drag factor and the 9.1 stk/m (what is that BTW, I thought stk is strokes, but the cadence is on the display already 25). That's four fields, and currently the Record entity is very slim, I don't want to make it fat, that could 1.5x the DB size. But I guess if it's about persistence the power curve would need some extra entity anyway.
2.2. How much data does the power curve mean? How man data points and what frequency?
3. Yes, I can see that the wifi + web app might be cumbersome. Once (I don't have ETA) I have time to write the Bluetooth peripheral companion app for my app you won't need dual connection, because the companion app can augment the FTMS or other standard BT profiles.

Regarding the format it'd be great to settle on something existing if it fits the needs. For example I wonder how C2 can log that data (if it can), they have detailed power curves. Actually you might consider implementing a protocol like theirs? See https://www.concept2.co.uk/files/pdf/us/monitors/PM5_BluetoothSmartInterfaceDefinition.pdf 0x003D "C2 force curve data characteristic": it's 2 - 288 bytes separated into multiple successive
notifications. Plus maybe a few more characteristics to cover the rest. Currently I'm connecting to 4 characteristics to handle their old style Bluetooth: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/develop/lib/devices/gatt/concept2.dart

And maybe they have also some format to export that to their log. Otherwise I can accommodate something else. The current JSON export is for MapMyFitness / Under Armour mainly, and the CSV is for Schwinn AC Performance+ Blue Carbon pkg (which is an ANT+ only console) import mainly, but I already tinkered with the CSV for my own activity transfer. The bottom line is I can add something, but it'd be great to leverage something existing, for example imagine if you could take advantage of Concept2's portal, if it can store it for you and display. I didn't explore C2 extensively yet though.

(I normally like CSV and JSON better, I've had my bouts with FIT, it's a flexible messaging format, I haven't looked into custom data chunks yet, but that's doable too I guess. I branched the FIT two ways anyway for SUUNTO.)

MrCsabaToth, 2024-03-12:
That little white module is "dope" with the Hungarian tags "SZENZOR". What's DINAMO BTW, could it be recharged from the flywheel? That would be nice! KayakPro's Genesis Port console is also headless, but I need to recharge time to time. It doesn't even have any LED, and the Bluetooth Battery Level always shows 100%.

Do you have some data logged coming off of that box BTW? I'd still consider to support it if I have enough data.

Abasz, 2024-03-12:
> What's DINAMO BTW, could the gadget be recharged from the flywheel? 

That would be an AC power source, but has not been implemented in the orange one we have. I think they got the idea from the Concept2 that is able to power its monitor from the flywheel rotation (but it uses a different approach).

> Do you have some data logged coming off of that box BTW? I'd still consider to support it if I have enough data.
> Can you test a custom APK if it works well both with the CSC and the Power Meter profile your ESP exposes?

I will get back to you by the end of next week.
> 2.1. By few metrics I guess you mean the drive and recovery, and also drag factor and the 9.1 stk/m (what is that BTW, I thought stk is strokes, but the cadence is on the display already 25).

Its supposed to be distance (meter) per stroke, but I swapped the two. It should be m/stk :) thanks for noticing I will fix it.

> 2.2. How much data does the power curve mean? How man data points and what frequency?

This is a very good question. So this depends on several factors like the number of magnets on the flywheel and the length of the drive phase. For instance on the C2 it could go even higher than 200 data points (data type is float). On the KF blue (6 magnets, so 6 impulses per rotation, but shorter drive than on a row erg) I've never seen going higher than 150 (averaging around 80) 

In order for this to work over BLE fragmentation needs to be implemented (I just realized this when you asked the question). In this sense C2 protocol is good as it does exactly this, but also the BLE Serial (Serial Port Profile, which you have done for the KF I suppose) could be used. KF approach of polling is lame, I would take advantage of the already available BLE notify that puts the server into the driver seat and makes life on the consumer side much simpler. I will need to see which is a better approach. In the [Open Rowing Monitor project](https://github.com/JaapvanEkris/openrowingmonitor) the C2 protocol is implemented in javascript and it is a pain....

If need be, I can bring down the size to 2 bytes (an unsigned short) per data point (assuming that negative values are not possible, max value for a data point is around 1,300 and the resolution is lets say 50), but my preference is to keep the proper 3 decimal precision as well as a higher theoretical max value (though I see a potential good compromise here).

So when you say slim database, I see your point. For a 5k paddle where there could be 3k strokes that is like 3k * 80  * 2 byte plus the other metrics.... Here probably writing the data directly to the file (json, xml, fit which ever) and persist that can meet your slim DB requirement.

> C2 export And maybe they have also some format to export that to their log. 

Their export from the log book is junk though they can do FIT, CSV and TCX. There are several reasons 1) they dont record proper metrics during recovery 2) dont record metrics on every second only on per stroke basis 3) force curves are not persisted only displayed (i.e. its not possible to download from log book) and there are many other issues like its recorded like having data for the 15th minute then the next data point is the 17th (assuming there is a recovery in between) and no data nothing between (like HR for instance), it also does not give a shit if you do moderate rowing in between intervals. Here are two exmpales for the same workout (an interval session).
[concept2-result-85891846.csv](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14578633/concept2-result-85891846.csv)

[concept2-logbook-workout-85891846.fit.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14578639/concept2-logbook-workout-85891846.fit.txt) - note the extension git does not allow FIT so I added txt but its a fit file :)

Next steps:
1. I will get you the data for the orange machines
2. I try to test how multiple clients can connect to the ESP32 (so e.g. a garmin watch and a phone can receive data) - if this is not doable, one road is the bridge. If doable I think bridge is not really important for the ESP Rowing Monitor it self as two client (consumer) devices can just simply connect to the ESP32 and subscribe to what they are able to (e.g. Garmin to CPS, phone to the extended data stream as well as the CPS).
3. In terms of the protocol to go with, I think there are two routes: 
    a) supplement the core service (like CSC, CPS et.c) with serial profile and broadcast all metrics as a serial string as well basically (this would be the closest to the current websocket approach, of course fragmentation and end of line signal would be necessary but that is necessary already for the KF).
    b) supplement the core service (like CSC, CPS etc.) with new service(s)/characteristic(s) that have the new data. 
   I will think through which would be potentially the best (also what are the pro's and con's of each), but at a first glance the serial approach requires less work from both the consumer and ESP32 side. I will also need to keep in mind (and possibly test) some performance considerations for both RAM and CPU cycles (ESP32 is powerful for an MCU but still and MCU :))

MrCsabaToth, 2024-03-13:
> That would be an AC power source, but has not been implemented in the orange one we have. I think they got the idea from the Concept2 that is able to power its monitor from the flywheel rotation (but it uses a different approach).

Interesting! It'd be great though because then no cable needed. The newer console has a rechargeable battery, which if I recall correctly is either two 18650, or one longer?

> > Do you have some data logged coming off of that box BTW? I'd still consider to support it if I have enough data.
> > Can you test a custom APK if it works well both with the CSC and the Power Meter profile your ESP exposes?
> 
> I will get back to you by the end of next week.

Not urgent.

> Its supposed to be distance (meter) per stroke, but I swapped the two. It should be m/stk :) thanks for noticing I will fix it.

I see. So essentially this is redundant and I can derive it from the speed and the spm (cadence), no need to transmit it.

> This is a very good question. So this depends on several factors like the number of magnets on the flywheel and the length of the drive phase. For instance on the C2 it could go even higher than 200 [...]

We'd need to think about the extremes, so if someone is sprinting with high spm and the flywheel has many magnets, then how many bauds would that mean? I'm not sure how easy is it for you, but I'd consider sending binary format data, because then you can save yourself from serializing (printf) all the values, and then I would not have to parse a string either.

Not sure if you are hitting the computation limits of the ESP.

(BTW I remember back in the day I studied the DOOM source code, John Carmack used some pretty neat techniques, since at that time GPU weren't a thing (nVidia Voodoo just came out and could perform some vertex operations, triangle mesh handling and some texturing). So he resorted to use pre-computed lookup tables to compute certain things such as trigonometric functions combined with some interpolation (your lookup table just has to be good enough resolution), and also for square root or logarithm (https://stackoverflow.com/questions/64355426/how-does-doom-determine-its-cosine-lookup-table-from-an-existing-sine-lookup-tab and https://www.doomworld.com/forum/topic/126997-classic-doom-source-code-questions/). Another technique: you may consider even float -> integer logic, such as some Mandelbrot generators were utilizing (https://math.stackexchange.com/questions/3455100/is-there-a-way-to-perform-calculations-of-mandelbrot-set-using-only-integer-numb), resulting in 100x speedup because integer arithmetic is so much faster than floating point. And fractal equation has some floating operations in there.)

> KF approach of polling is lame, I would take advantage of the already available BLE notify that puts the server into the driver seat and makes life on the consumer side much simpler. I will need to see which is a better approach.

Try to use BLE notify! Polling is inferior. As you say it gives the control to my hand, and what if I'm flood you with polls? With notify you can control the flow.

> In the [Open Rowing Monitor project](https://github.com/JaapvanEkris/openrowingmonitor) the C2 protocol is implemented in javascript and it is a pain....

It's just an idea, if it's too cumbersome then ditch it (a.k.a. "Nem erőszak a disznótor"). My thought was if it's possible to implement C2 compatible protocol then maybe even C2 software can connect and be compatible.

> If need be, I can bring down the size to 2 bytes (an unsigned short) per data point (assuming that negative values are not possible, max value for a data point is around 1,300 and the resolution is [...]

If we don't follow C2 or other protocol, then we can decide on our own. There are also techniques to qunitize a floating point mantissa and value separately. Whatever fits better.

> So when you say slim database, I see your point. For a 5k paddle where there could be 3k strokes that is like 3k * 80 * 2 byte plus the other metrics.... Here probably writing the data directly to the file (json, xml, fit which ever) and persist that can meet your slim DB requirement.

I switched to Isar because it is 100x faster than SQFlite. So I'd still persist there, because it'll be probably faster than if I serialize strings into JSON or XML. Especially with SQFlite my main goal was to do as little processing as possible during recording so I would lag the devices. Since I don't have iOS build especially Apple people tend to find a super old 10 year old Android tablet at a bottom of a drawer and - shocker - it's super slow :)

The slim would mean that if I add these special fields to the main Record entity, they will be not utilized for 99% of the users and would just take up space. So I can already envision that there will be let's say a `RecordExtra` or similar entity, which can be *1:1 relationship with the Record entity.

And even if your force curve will be different from C2's, I think I'll try to find a common format so I can utilize force curve graphs for C2s as well. So then the feature wouldn't be only for Kayak First.

> Their export from the log book is junk though they can do FIT, CSV and TCX. There are several reasons 1) they dont record proper metrics during recovery 2) dont record metrics on every second only [...]

[... cut ...] d one for power curve. I guess it'll matter of your software stack). If I were you I'd try to ditch serial, especially for the power curve due to bandwidth. And have enough MTU, not just 20 bytes.

What I'll do at some point is probably I'll split our conversation into multiple tickets. For example me implementing the white box's protocol is very low priority, and trying to work with you is higher priority. Maybe even until then we can open up a discussion.

I'm excited, but I'll need free time.

Abasz, 2024-03-18:
>Interesting! It'd be great though because then no cable needed. The newer console has a rechargeable battery, which if I recall correctly is either two 18650, or one longer?

The charger is not implemented in the hardware we have. So it only works of external power. The new does not have this feature it has two 26650 or even bigger. 

>I see. So essentially this is redundant and I can derive it from the speed and the spm (cadence), no need to transmit it.

The metrics shown on the pictures are not a 1 to 1 to what is broadcasted. There are metrics that are calculated on the client side. Currently these are the data that are being via the websocket (its json string that is converted to a byte array) 

0. revTime
1. distance
2. strokeTime
3. strokeCount
4. avgStrokePower
5. driveDuration
6. recoveryDuration
7. dragFactor
8. handleForces
(9. deltaTimes) - this serves debugging and calibration purposes only

The first 4 items are seemingly equal to the data that is braodcasted via CPS but the format is different (CPS and CSC uses different resolution and can overflow, these are unsigned long long (i.e. 64bit unsigned ints).

From the above the stroke rate or the distance per stroke, as well as pace can be calculated.

I think the best would be as you suggested that there are multiple characteristics (the CPS, then on extended, then one for the force curve). I just have to check whether such setup takes more time to complete the notifications on the MCU side compared to having packed everything into one. 

There is one issue that needs consideration and that is the power data. It is possible to set ESPRM to CSC mode where no power is broadcasted. So I think power will be a permanent element of the extended data characteristics (which makes it redundant when CPS is mode is set but never mind, its a singed short anyway so 2 bytes).

> We'd need to think about the extremes, so if someone is sprinting with high spm and the flywheel has many magnets, then how many bauds would that mean? I'm not sure how easy is it for you, but I'd [...]

I was able to do multiple simultaneous connection (mobile and watch) and subscribe to the same characteristic. I dont know the performance impact yet I will do more testing but this is promising. 

Its difficult to predict extremes, but I think 200-250 should be the cap. Any way I think handle forces data will need a fragmentation in place (e.g. if MTU cannot be set to a sufficiently high value). I have some ideas, like setting up the structure where the first two bytes are the total number of fragments and the actual fragment number and the fragments are sent via notify after each other. Client will know that all fragments have arrived once the first two bytes are equal. I think this is something similar to what C2 has but the documentation is not clear on this. 

I will need to do benchmarks and I have to think through what is the best approach. I see three ways currently:

1. via simple notify I burst the data as described above (pro. simplest from the server's perspective, probably the most efficient as well, con. there is no way to know if the data was received fully by the client, my be the notify bursts are two quick, or there is no way to know whether the fragments are received in the proper order etc.)
2. via indicate (pro. simple from the server's side, its known that the packets arrived and the server can send the next once the client indicated receipt, order of the fragments can be preserved, con. I dont know the performance cost of this)
3. via an initial notify that is followed by subsequent reads (pro client is in the driver seat so data can be validated, allows the implementation of re request on failure, con. rate limiting is complicated on the server side, performance impact is unknown, server needs to handle read requests after all data is sent)

In all 3 cases the handling of data quality and the fragments needs to be considered. Fragment order can be ensured with the frament number as well as validation whether all fragments are received. 

Any way I will do some testing and see. I lean toward the indicate option if it does not have a significant performance overhead and ensures data fragment quality and "self" rate limiting by waiting for an answer from the client.

> Not sure if you are hitting the computation limits of the ESP.

The issue is the frequency of the incoming impulses via the interrupt. The MCU should be able to complete the fitting of the regression curves by the time the next impulse comes in. The limit is not BLE that currently broadcasts once per sec (which will be increased to once per stroke), or the wifi. The problem is that if something blocks the CPU for more than like 5ms (which seems an eternity but some of the iterative calculations are for the theil Sen quadratic regression fitting are pretty involving) it is possible that a new impulse comes in before the processing of the previous is completed that messes up the model.

> I'm excited, but I'll need free time.

Dont worry about this, for me this is also a hobby project. Even without your app, I would be trying to implement the new BLE profile and keep using the webGUI , if it allows me to provide alternative to the websocket server (which does not require wifi) :). So I have already profited from this conversation as I have not once thought of using BLE for extended data metrics.

Abasz, 2024-03-20:
Hi Csaba!

Please find attached the data you requested:

These are the commands that are sent from the app
[commands.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14673073/commands.txt)
On Bluetooth connect it issues a 9 and then it starts polling the data (1,2,3,4). The second 9 is when I hit start in the app.

These are the responses for each command
[responses.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14673084/responses.txt)
Little explanation on the response:
- I assume that the 1 to 4 corresponds to the drive start, drive end, recovery start, recovery end of the last known point.
- The second line of each response is the stroke count (ideally this should follow a pattern, which is not the case and that is due to the pulling method which sometimes results pulling the same stroke again, I suppose they have some kind of check for this but I was not able to identify this in the decompiled code)
- The third line is the time in micro seconds of that delta time (time the last rotation took) point (i.e. drive start drive end recovery start recovery end).

Now all metric calculation is done by the app. It uses the start and end times of the drive and recovery (i.e. how long does one rotation take at the start of the drive and at the end of the drive) to calculate metrics. I assume it determines drive and recovery based on whether the impulse delta times are increasing (signalling deceleration i.e. recovery) or decreasing (signalling acceleration i.e. drive).

Basically it calculates the drag factor during the recovery - declaration - that is used to calculate the torque, that is used together with the "magic number" - that is calculated based on the weight of the paddler and 12kg that is the assumed boat weight - the speed. Pull force is calculated based on torque and the radius of the flywheel puly. Does not give watts (think).

Supporting this machine would mean that you need to implement the full model in your app as per the [Physics of Ergometers](https://eodg.atm.ox.ac.uk/user/dudhia/rowing/physics/ergometer.html). Its using a numerical approach which is not accurate as it relies too much on the recorded timing of the input signals and a few microsecond difference could result in high errors (more on this [here](https://github.com/JaapvanEkris/openrowingmonitor/blob/main/docs/physics_openrowingmonitor.md#determining-the-angular-velocity-and-angular-acceleration-of-the-flywheel)). 

This is how I test. Basically I have an MCU that is connected to the sensor pin, which simulates impulses. Then I have an FTDI connect to the BLE UART module TX and RX pins that reads the serial.

This is the code that I run on the MCU. Basically it simulates a very steady paddling by faking impulses in varying periods (this is the main reason I know what the response means)
[main.cpp.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14673104/main.cpp.txt)

Let me know if you have any questions.

Also I read it that you have contacted KayakFirst. I was wondering whether their blue machines has the same flywheel as the orange. Specifically, if the inertia of the blue is the same as for the orange. This would be a very valuable piece of information for me. 

I was able to find inertia value for the orange in the decompiled code (as that is needed for calculating the metrics that is done in the app). Until recently, I assumed they are the same as I assumed that metrics are calculated on the app too, but now I am no longer sure...

Thanks.

MrCsabaToth, 2024-03-24:
Thanks, that's great info for the future - in case I'd want to support the orange ergo straight -, I'll branch it off into a low priority ticket later. Apparently the blue erg's console has only the "6" command as a poll, and the commands "1" - "5" are all related to initialization, configuration and startup. Command "9" is the stop.

I got some info about the flywheels: the newest blue erg are codnamed BULL, they have different flywheel characteristics than the earlier models. On top of that there was a version change between the older models around 2020. The orange ones in the UTE are pre 2020, so two generations earlier than the blue BULLs. This also means you cannot apply the same constants and characteristics.

The old Android app should be compatible with all of them, so reverse engineering that could yield the required constants.

Abasz, 2024-03-24:
I have good news on the above topic. Namely, that I was able to implement a proof of concept of the extended BLE protocol (actually the NimBle stack of the ESP32 is pretty efficient and the high-level implementation of this was simpler than I have thought). 

I have also done high-level benchmarking and the BLE stack is more performant than the Wifi meaning that disabling the Wifi and broadcasting to multiple clients even with extending data performs better. But still it is possible to enable both services at the same time (so broadcast via Wifi as well as BLE extended). I was also able to manage the additional memory requirement efficiently so that is not a bottleneck either (even with both services enabled). 

Based on the testing it is clear that only the notify version (proposed option 1. above) is capable of achieving the needed results. Basically, I’ve added two new services (lets call them extended metrics, that has additional data like recovery time, drag factor etc. and handle forces, that has the data for the curve). 

### Extended metrics

The first one is simple, very similar to the standard BLE, and uses a byte sequence to broadcast (e.g. the first two bytes are the average power – as a short –, then the next 4 bytes represent the recovery time – that is an unsigned int – etc.). 

I will need to broadcast settings and my original intend is/was to include the settings as the first byte in the extended metrics. I have thought about setting up once again a new characteristic (or even a new service) for this, but I don’t see that this makes much sense right now as there are only 3 settings. Changing of settings are done via control point OpCodes. I am convincible though 😊 as this approach does not support future expansion of the settings object (I have plans to make all rower settings editable at some point but that is a very very long term plan 😊).

###Handle forces

The handle force was a bit trickier as it needs to be flexible so I ended up using a fragmentation logic (which is similar to the C2 approach), meaning that it will chunk up the data to separate notifications (its important its not indicate, more on that below) where the first byte is always the total number of chunks to be expected, the second is the sequence number of the chunk (so client can track them efficiently). The rest within the chunk are the data in sequences of 4 bytes as a 32bit float. 

The chunks are calculated based on the MTU negotiated by the client (ESP32 stack support up to 512bytes for MTU). So, for instance the EFR connect android app sets MTU to 247 (this is max), while on my PC web browser BLE stack sets it to 255. So, when determining the chunk size and number of chunks needed to broadcast the data set this will be the base size. Note that only full floats (i.e. 4bytes) are broadcasted, meaning that partial bytes not (so it cannot happen that the first 2 bytes of the float is one notify message and the second is in the next). So basically every byte in the chunk can be converted to its proper float value. Actually, I have thought about allowing partial floats since it could be more efficient (reducing the number of chunks needed) especially for lower MTU sizes, but I think it adds extra complexity to the client as it would need to wait for all bites before it can convert as well as make handling missed packets more complicated. This way the client can be sure that when it receives a package it can be converted and added to an array. If something is missing that data will be missing in the array but would not jeopardize subsequent package conversion (note that actually based on my testing if a package was not received/registered by the client all subsequent once were lost until the next burst, so technically this is less of an issue but still).

I think this provides the necessary flexibility.

###Other considerations and tests

I have stress tested the app by setting the chunk size to 23 (basically 23 - 3 for BLE headers - 2 for the first two bytes that actually results in sending up to 18 bytes of useful data per notification, which is technically 4 float number – total of 16 bytes). Android EFR connect app was able to handle up to 15-17 packages without loss. Above that it became unstable, and the end was lost. I think this is fine. MTU of that low is not going to work anyway, I might add some checks for a minimum MTU size that the client needs to be able to handle (or check for the number of packages to be sent) otherwise handle forces are not sent but even IOS can handle higher MTU than this 23 (which the minimum per the BLE specs). 

I have tested which notification type (indicate or notify) would be the most appropriate. Indicate has so high overhead on the response that indications of more than 4-5 was lost. So that is not a viable option. 

We need to also decide on the UUID. I am leaning toward using the Concept2 UUIDs for this purpose, but since these are not on spec I think it may create issues along the way. So having something total different might make sense.

Abasz, 2024-03-24:
I just saw your response

> I got some info about the flywheels: the newest blue erg are codnamed BULL, they have different flywheel characteristics than the earlier models. On top of that there was a version change between [...]

I never tried the BULL but the blue once before that I have. Just to clarify: I understand that the orange are pre 2020, and anything that is blue is after 2020. Does that mean that anything that is after 2020 has different flywheel right? Actually, it would make sense as the metrics calculated by ESPRM on these blue machines seemed materially lower compared to on water tests so I suspected that something must be different. Thanks for asking around about this.

In terms of the ESPRM BLE API, I will be implementing this regardless whether you will support or not as I think it will be simpler on the long term to connect via BLE than to set up a hotspots and all that... probably in the up coming weeks I will create a devel branch with the proof of concept version. Also I will update its WEB Gui to support extended BLE connection, which could also serve as an example how to consume the data.

MrCsabaToth, 2024-03-26:
Great, so your protocol will be notification based and also binary!

1. Make sure that you'd serialize any values little endian way (Intel x86 is big endian, and all of the protocols I saw were little endian so far) just to be in concert with other protocols.
2. Also please select different service UUID than C2, something unique. That's the best way to distinguish your solution from others, better than relying on manufacturer name. If you'd select C2 you could also confuse any other app which is expecting a C2. Unless you implement the exact protocol of C2, but yours sounds like to be custom.

I implemented resistance level, soon will implement step count / revolution count display, and I think within a month or so I'll also move towards your extra values. Yours will be also persisted. I want to display the force curve. I may actually utilize the UI and persistence of this for C2 as well.

As for Kayak First goes: it has at least 3 flywheel generations, all of them different. The blue ergs are the latest ones, and the orange ones in UTE are the earliest one. Constants should be hidden somewhere in the depth of the Android app's apk.

Abasz, 2024-03-26:
> Great, so your protocol will be notification based and also binary!
Yes
 
> 1. Make sure that you'd serialize any values little endian way (Intel x86 is big endian, and all of the protocols I saw were little endian so far) just to be in concert with other protocols.
Yes this is the case (all BLE protocol data is LE so it made sense to keep using that)

> 2. Also please select different service UUID than C2, something unique. That's the best way to distinguish your solution from others, better than relying on manufacturer name. If you'd select C2 you [...]
I agree, I will generate one.
 
> I implemented resistance level, soon will implement step count / revolution count display, and I think within a month or so I'll also move towards your extra values. Yours will be also persisted. I [...]

Using the same UI for C2 makes sense. Essentially they are the same 

> As for Kayak First goes: it has at least 3 flywheel generations, all of them different. The blue ergs are the latest ones, and the orange ones in UTE are the earliest one. Constants should be hidden somewhere in the depth of the Android app's apk.

I dont think the blue one is in the APK. The calculations are made on the MCU (I it has a STM32 chip that has probably sufficient computing power to handle UART the interrupt and the calculations. Also by looking at the data that is broadcasted 

(6;1711448904253;0;0;0;0.00;0;1;0;1759;4.28;4.51;67;65;46;44;116;110;233;220;17837;316;411;0)

This does not have the necessary delta Times data that can be used to calculate the metrics (rather the metrics are calculated on the MCU). This indicates that the APK does not need to have the inertia value for the Blue machines. (it has this for the orange, and I found it already).

Also, In the APK there is a firmware bin file (I think its dated to 2019). When I tried to connect with the KF official app in the DB log I was able to find the following: ergo update: _"NO ergo update needed - oldDriver: 210206 (available: 191118), oldDisplay: 201008(available: 0). errodDriver: 0, errorDisplay: 0"_ the firmware bin file's name is 191118_driver.bin in the ergo_firmware folder. 

I am not sure if its not possible to decompile such firmware file unfortunately... 

dfasd

MrCsabaToth, 2024-03-26:
I think I used the developer console (which is blue one) with the APK, however if it does not login any more then I may not be able to confirm.

One note for your web app: take a look at Web Bluetooth: https://caniuse.com/web-bluetooth
I experimented with that when it was relatively new: https://github.com/gdgfresno/candle-bluetooth

(As you see some browsers don't support it. The biggest problem is the stupid mobile Safari, since on the walled garden Apple ecosystems all of the mobile browsers (even if you install mobile Firefox or Chrome on iOS) are actually Safaris. There were some 3rd party Apple Store apps which would allow Web Bluetooth for mobile Safari, but that could be complicated for an average user. The good news is that finally EU forces Apple to allow alternative browser engines https://www.theregister.com/2024/01/25/apple_eu_dma/. And that can open up easier Web Bluetooth usage via Chrome for Apple users as well. Unfortunately as a side effect of this enforcement is that some PWA features are hurt: https://www.macrumors.com/2024/02/15/ios-17-4-web-apps-removed-apple/)

If I could I'd convert my app into a PWA (Progressive Web App) so I wouldn't have to deal with the cumbersome app store release cycles and tackle the iOS and Apple platforms easier (#34). However I learned that the persistence is important and hurt the performance, so there are several things with a web app:
1. Provide efficient persistence of the recorded workout.
2. Verify that a browser can handle multiple independent connections, such as a paired HR monitor and possible 3-4 simultaneous notify BLE streams in case of a C2 classic protocol.
3. For historical workout storage I'd need a server back-end which currently I don't have.

I'm still very attracted to a PWA, if the performance issues can be overcome then the only issue would be to provide nice enough UI, which Flutter is quite good at. There are some special things such as the graphs, however if I think again: I use Syncfusion Flutter Charts, and that is based on the HTML widget under the hood, so probably I can migrate it to the HTML widget. As I'm writing this I'm thinking more and more of a pilot project!

MrCsabaToth, 2024-04-03:
Just a note from off-band conversations: there are three major type of machines based on flywheel characteristics / mechanics:
1. The orange one shown in this ticket, these are pre 2020
2. 2020-2022
3. 2022- (newest blue machines)

Abasz, 2024-04-08:
This is the draft of the BLE API documentation of ESPRM. Feel free to make any suggestions to the structure of the API. Its not published yet so there is flexibility.

I have a working example for the client side in JS (technically the code is TS and uses RxJs for the streams) for the WebGUI: [ble-data.service.ts.txt](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14903771/ble-data.service.ts.txt). please see the functions `connectToXXX` and `observeXXX` for the parsing of the broadcast data.

Both the client and the server are under testing (which is a bit slow as the weather is good so indoor training is no longer priority, but I have long simulated sessions already) but I consider them as stable for now.

### Bluetooth

ESP Rowing Monitor provides a BLE interface to get the data and metrics it measures.

It supports two concurrent BLE connection, meaning that one may connect a smart watch as well as connect to the WebGUI via BLE.

ESP Rowing Monitor supports two standard BLE profiles that allows it to be connected to smart watches and 3rd party apps:

1. Cycling Speed and Cadence Sensor profile
2. Power Meter Sensor profile

Switching between the profiles can be done through the WebGUI or via BLE Control Point using a specific OpCode (18). The implementation of these profiles complies with BLE standards, although the OpCode used is not standard.

Both BLE profiles are fully compliant with BLE standards, making them accessible by compatible devices such as smartwatches. They have been tested with Garmin smartwatches, including FR235, FR645, and FR255. This also means that the device advertisement data is either CPS or CSC (based on the selected profile).

Please note that in order for Garmin watches to work with ESP Rowing Monitor, a cycling activity must be selected. Otherwise, due to limitations of Garmin, the watch will connect but will not use the sensor data for metrics.

To obtain accurate speed and distance data, the wheel circumference must be set to 10mm when pairing the device with ESP Rowing Monitor.

In addition as of version 5.1 experimental supports for custom BLE profiles/services were added.

1. Extended Metrics Service (UUID: a72a5762-803b-421d-a759-f0314153da97)
2. Settings Service (UUI: 56892de1-7068-4b5a-acaa-473d97b02206)

#### Extended Metrics Service

This Service currently contains two characteristics:

```text
Extended Metrics (UUID: 808a0d51-efae-4f0c-b2e0-48bc180d65c3)
```

Uses Notify to broadcast the following metrics (which may be extended in the future) as an array of consecutive bytes (i.e. currently a total of 11 bytes, Little Endian):

- avgStrokePower (16bit short in Watts)
- recoveryDuration (32bit unsigned int in microseconds) // _NOTE: I am still considering if this should be changed to a 16bit unsigned short with a resolution of 2048 or 4096 (like the time for CPS profile), which would leave this with a max value of ~32/~16 seconds that should be plenty_
- driveDuration (32bit unsigned int in microseconds) // _NOTE: I am still considering if this should be changed to a 16bit unsigned short with a resolution of 2048 or 4096 (like the time for CPS profile), which would leave this witha max value of ~32/~16 seconds that should be plenty_
- dragFactor (8bit unsigned char)

New metrics are broadcasted on every stroke (after the drive ends) or at least 4 seconds (which ever happens earlier).

```text
Handle Forces (UUID: 808a0d51-efae-4f0c-b2e0-48bc180d65c3)
```

Uses Notify to broadcast the handle forces measured during the last drive phase. Full data is broadcasted once per stroke (after the drive ends) or at least 4 seconds (which ever happens earlier).

Considering that the number of measurements vary from stroke to stroke (since, among others, it depends on the number of impulses per rotation, the machine etc.) this characteristics may be chunked into consecutive notifies ("bursts") until all data is flushed. The chunk size (consequently the number of consecutive notifies within a burst) will depend on the MTU (max data size per broadcast) negotiated with the client (ESP32 supports 512 bytes, but for instance on android based on experience this is around 250).

The first byte in every Notify is the expected total number of chunks within the burst, the second is the current chunk number. Rest of the bytes in one Notify are 32bit floats in Little Endian. Every chunk can be parsed individually without data loss (i.e. the bytes of one float is never broken into two notifies, which prevents data loss on missed packages/notifies). Basically the last Notify within the burst is signaled by the fact that first two bytes of the data package are equal.

Below is an example of data where the MTU is 23 bytes (which practically means that 20 bytes would be available to transfer actual data). Considering the first two bytes are reserved, that leaves 18 bytes for the data, but since into that only 4 32bit float can be fitted basically a maximum of 4 floats per Notify can be sent.

Handle Forces values:
_[2.4188, 52.64054, 80.1877, 110.6412, 142.9242, 174.9786, 201.5447, 218.6083, 228.5825, 233.4143, 234.7116, 234.2218, 230.6765, 227.8172]_

Notifies:

1. [5,1,159,205,26,64,234,143,82,66,37,96,160,66,81,72,221,66]
2. [5,2,156,236,14,67,135,250,46,67,119,139,73,67,186,155,90,67]
3. [4,3,32,149,100,67,19,106,105,67,46,182,106,67,201,56,106,67]
4. [4,4,48,173,102,67,56,209,99,67]

[... cut ...]
    ChangeBleService = 18U,
    SetWebSocketDeltaTimeLogging = 19U, // deprecated and will be changed to delta time logging via extended BLE
    SetSdCardLogging = 20U,
```

The response to the Write is sent via Indicate and the structure follows the BLE Control Point standard (i.e. starts with the ResponseCode - 32 -, followed by the request OpCode, than the ResponseOpCode - e.g. 1 for success or 2 for an unsupported request OpCode).

Also a Notify is sent by the Settings characteristic including the new settings.

Abasz, 2024-04-22:
Added one more characterstic to the Extended Metrics. This is to allow broadcasting the measured delta times between impulses, so they can be visualised/analysed as well as replayed/simulated for calibration purposes. This is not a general use characteristic so there is no immediate need for it, rather this is a convenience method to record calibration and debugging data.

Below is the quote from the draft README

```text
Delta Times (UUID: ae5d11ea-62f6-4789-b809-6fc93fee92b9)
```

Uses Notify to broadcast the measured delta times if enabled. This serves mostly calibration/debugging purposes as the recorded delta times can be replayed and test various settings efficiently. This feature is disabled by default (meaning that this characteristic may not be visible). It can be enabled by defining `ENABLE_BLUETOOTH_DELTA_TIME_LOGGING true`. After that the actual notification of the measured delta times can be turned on or off via OpCode 19.

The measured delta times are broadcasted once sufficient elements to fill the max negotiated MTU (minus 3 for the header i.e. when the max data capacity) is reached or if 1 second since the last broadcast has passed.

Basically if the negotiated MTU is 255 then 63 delta times can be broadcasted ((255 - 3)/4 - assuming that unsigned integer is 4bytes on the system like on the ESP32). Actual frequency will depend on the number of impulses and the speed of the flywheel since.

In practice once the system measured 63 delta time value it will send Notify (or if 1 second elapses since the last Notify) to the connected clients. Please note that in certain cases this could be rather resource intensive (e.g. when there are a lot of impulses per rotation), the client should support and negotiate a minimum MTU of 100 (ESP32 NimBle stack supports up to 512bytes). If the MTU is below 100, no Notify will be sent.

The data in the Notify are 32bit unsigned integers in Little Endian.

MrCsabaToth, 2024-05-08:
Opened #505

Abasz, 2024-05-08:
Quick note: I made my mind up about the extended metrics recovery and drive duration and went with the 16bit ushort and a resolution of 4096. 

#### Extended Metrics Service

This Service currently contains three characteristics:

```text
Extended Metrics (UUID: 808a0d51-efae-4f0c-b2e0-48bc180d65c3)
```

Uses Notify to broadcast the following metrics (which may be extended in the future) as an array of consecutive bytes (i.e. currently a total of 7 bytes, Little Endian):

- avgStrokePower (16bit short in Watts)
- **recoveryDuration (16bit unsigned short in seconds with a resolution of 4096)**
- **driveDuration (16bit unsigned short in seconds with a resolution of 4096)**
- dragFactor (8bit unsigned char)

New metrics are broadcasted on every stroke (after the drive ends) or at least 4 seconds (which ever happens earlier).

MrCsabaToth, 2024-05-09:
I'll convert this issue into a discussion.