TrackMyIndoorWorkout/TrackMyIndoorWorkout #498: Application freezes and crashes with KayakFirst Blue sudden disconnect

Abasz, 2024-03-28:
When the connected KayakFirst machine suddenly disconnects the app crashes on Android (Samsung Galaxy S20FE, Android 13)

I was trying to export the logs, but after clicking export I do not see where it is saved. Can you help me with that?

Thanks

MrCsabaToth, 2024-03-28:
1. The application needs to be enhanced to the possibilities of Bluetooth devices turning off. Right now it's some pacman (catch all) exception handling, and once connected the display lists it connected even when console turns off. I think I have a ticket for this but I couldn't find it.
2. Intermittent connection black-outs and handling those with auto-reconnect would be a step further I also want to tackle sometimes (?) in the future.
3. The log export first exports the logs in a textual format and gzips it (both in memory) and then kicks off a share intent. So there must be an intent popping up where you can select share destinations (OneDrive, GDrive, GMail, etc).
4. If nothing pops up my first suspicion is if you might not have any logs, because the log level is set to NONE. Set it to INFO. In such case you'd need to see a snack bar pop up briefly which tells you there's nothing to export. It disappears in some seconds.

Once you turn on the logging to INFO I'd really be interested in the exception. (Hopefully logging catches it)

Abasz, 2024-03-28:
> If nothing pops up my first suspicion is if you might not have any logs, because the log level is set to NONE. Set it to INFO. In such case you'd need to see a snack bar pop up briefly which tells you there's nothing to export. It disappears in some seconds.

I set it to Info at the first use :) I knew I might need it. So there is no snack for the intent popping up. I tried clearing the log, and click export, then I have the snack saying that no logs. May be I am missing some permission?

MrCsabaToth, 2024-03-29:
First of all, sorry for thinking you haven't turned on the log level, the only reason is that I know the UI is rudimentary.

I don't think you need any permission simply for kicking off a share intent. So actually I could potentially remove that from the manifest: #456

Currently I'm using [share_files_and_screenshot_widgets](https://pub.dev/packages/share_files_and_screenshot_widgets) for:
* The log file export (in the Expert preferences)
* Full data export (also in the Expert preferences)
* Individual activity export (in every selected format) in the Activities list

So first I'd be curious if you can export any recorded activities in any chosen format. Because if you can then this will be something about the logs only. The logs are gathered in a table, so that's part of the native app like other data - not requiring any permissions.

You can check all the permissions depending on Android OS version and UI and proactively enabling everything. I'm super curious what causes this. If the log cannot be exported then the crash cannot be debugged much either (I'm not sure if the app store get any report about that, since I try to pacman catch everything for the purpose of logging).

Abasz, 2024-03-29:
Ok so after some back and forth I was able to export  the logs with the export all data method (simple log export was not working). 

Actually I had to set the database location and then I had a couple crashes then exported all data. Then I had more crashes and now on every once and a while the simple export logs works too (but not on every run). Not sure what its problem but not permissions. Based on the logs I do not see anything about that
[DebugLogs20240329T213014026729Z.txt.gz](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/files/14808687/DebugLogs20240329T213014026729Z.txt.gz)

Based on the logs it seems (but I am not sure) that the app hangs and repeats this error constantly with like 1ms frequency.

2024-03-28 14:46:52.047811,ERROR,DATA_HANDLER,_executeControlOperationCore,controlPoint.write; FlutterBluePlusException | writeCharacteristic | fbp-code: 6 | device is not connected; #0      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:170)
#1      KayakFirstDescriptor._executeControlOperationCore (package:track_my_indoor_exercise/devices/device_descriptors/kayak_first_descriptor.dart:179)

MrCsabaToth, 2024-03-30:
I think the reason nothing was happening for log export attempt is out of memory: `ERROR,MAIN,runZonedGuarded pacman,Out of Memory;`.

Besides that I think I need to solve this flood which inflated the whole log:
```
2024-03-28 08:46:59.461241,ERROR,DATA_HANDLER,_executeControlOperationCore,controlPoint.write; FlutterBluePlusException | writeCharacteristic | android-code: 133 | UNKNOWN_GATT_ERROR (133); #0      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:215)
<asynchronous suspension>
#1      KayakFirstDescriptor._executeControlOperationCore (package:track_my_indoor_exercise/devices/device_descriptors/kayak_first_descriptor.dart:179)
<asynchronous suspension>
#2      KayakFirstDescriptor.pollMeasurement (package:track_my_indoor_exercise/devices/device_descriptors/kayak_first_descriptor.dart:223)
<asynchronous suspension>

2024-03-28 08:46:59.462953,ERROR,DATA_HANDLER,_executeControlOperationCore,controlPoint.write; PlatformException(writeCharacteristic, device is disconnected, null, null); #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:651)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334)
<asynchronous suspension>
#2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:563)
<asynchronous suspension>
#3      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:203)
<asynchronous suspension>
#4      KayakFirstDescriptor._executeControlOperationCore (package:track_my_indoor_exercise/devices/device_descriptors/kayak_first_descriptor.dart:179)
<asynchronous suspension>
#5      KayakFirstDescriptor.pollMeasurement (package:track_my_indoor_exercise/devices/device_descriptors/kayak_first_descriptor.dart:223)
<asynchronous suspension>
```

MrCsabaToth, 2024-03-31:
Maybe for workouts the in-memory file serialization is a good idea, but for logs and full export I should possibly write to a temporary scratch file.

MrCsabaToth, 2024-03-31:
I branched off the export OOM possible avoidance into #498

MrCsabaToth, 2024-04-01:
Note to possibly enhance debug prints via article https://docs.flutter.dev/testing/code-debugging referenced by https://github.com/boskokg/flutter_blue_plus/pull/455#issuecomment-2028701644

MrCsabaToth, 2024-04-05:
I was able to figure out something for the full DB export with a new technique (I needed a full DB migration to a new phone), however I still need to figure out something for the log export still.

MrCsabaToth, 2024-04-05:
Refactored Log export too. Now we just need to refactor file export and share to use materialized files to help low mem devices, and then we can completely get rid of `share_files_and_screenshot_widgets` and rely only `share_plus`, however that will be a different issue.

MrCsabaToth, 2024-04-26:
Regarding the crash in the `_executeControlOperationCore` I can imagine multiple improvements:
1. At deeper level the Logger can be more intelligent and deduplicate completely identical entries. That can stop a flood filling the log.
2. Specifically in `_executeControlOperationCore` the exception causes in infinite loop, since the `while (chunkBeginning < commandBytes.length)` condition never fulfilled because the `chunkBeginning` doesn't advance.

commit ce05729e (2024-04-26): Avoiding endless loop of polling style data collection when the device goes off, fixes #498
closed 2024-04-26 by MrCsabaToth