TrackMyIndoorWorkout/TrackMyIndoorWorkout #505: Display extended paddling metrics

MrCsabaToth, 2024-05-08:
This is to support special console by @Abasz with proprietary protocol (#492 https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/492#issuecomment-2042362071) and also Concept2 RowErgs. Metrics include:
* Drag Factor (C2 0x0031)
* Drive Length (C2 0x0035)
* Drive Time (C2 0x0035)
* Stroke Recovery time (C2 0x0035)
* Stroke Distance (C2 0x0035)
* Stroke Power? (C2 0x0036)
* Force Curve (C2 0x003D, 288 bytes = 16 * 18)

And any extra what the Kayak First special console offers. First this will be display only and will be tested on a C2 RowErg. Then we can move onto persistence, which could be a follow-up ticket.

MrCsabaToth, 2024-05-08:
Sounds like Abasz ESP RowErg mentions avg stroke power, and also peak power

Abasz, 2024-05-08:
> Sounds like Abasz ESP RowErg mentions avg stroke power, and also peak power

Its peak force, not power. Those are different metrics. AVG stroke is basically the total power within one stroke cycle (i.e. drive + recovery).

MrCsabaToth, 2024-05-08:
> Its peak force, not power. Those are different metrics. AVG stroke is basically the total power within one stroke cycle (i.e. drive + recovery).

Thanks for the correction, I think C2 has that too. I plan to test the UI with a C2, once I have something. It may take a month or more. I'll be at Google I/O and then the week after that in NY.

MrCsabaToth, 2024-10-16:
I still haven't got to this due to job search and now starting a new job, but hopefully during the holidays I'll have time