TrackMyIndoorWorkout/TrackMyIndoorWorkout #512: 7/30;release

bschatzow, 2024-08-02:
Can not choose any device on 1.1.156 release. Screen shows up but devices immediately disappear
 Pixel 7pro. Been using several years with no issues like this.

MrCsabaToth, 2024-08-02:
Sorry to hear that, and thanks for reporting. I'll investigate this, can you turn on logging in the expert preferences (to the highest info level), reproduce this, then export the logs and send to me? Don't forget to turn off the logging after that.

You must be opted into the beta, I wonder if you exit the beta program would Google play install the 155, I never tried that.

Btw I use the app daily, but a debug version, so it must be something release only. Stay tuned.

MrCsabaToth, 2024-08-02:
Another idea: what happens if you turn off the compatible device filtering in the expert preferences? Your issue could be something with the newer version of the Flutter Blue Plus plugin, or the Life Fitness machine support I am working on.
What fitness machine are you using BTW?

bschatzow, 2024-08-02:
The compatible device filtering fixed it? On the logs, do you want the last item in expert, Info?
Just so it is clear, the devices show up for less than a second and then they disappear.

MrCsabaToth, 2024-08-02:
Yes, info level logging.
The scanning duration is 6 seconds by default. So the disappearance happens before the end of scanning? It is interesting because if the filtering logic is faulty then the affected devices should not show up at all.
What fitness machines do you have?

MrCsabaToth, 2024-08-02:
So far I have tried to reproduce on a Pixel 6, build 156 release version (proper install from play store), turned on a few devices but they stayed. Need to chase the issue more, but hopefully at least the issue is not wide spread?
[image]

MrCsabaToth, 2024-08-02:
So far I have tried to reproduce on a Pixel 6, build 156 release version (proper install from play store), turned on a few devices but they stayed. Need to chase the issue more, but hopefully at least the issue is not wide spread?
[image]

Update: fired up some more devices with the Pixel 6 test phone, involved some HRMs and recorded workouts with the Precor and then the Wahoo sensors
[image]

So this might be some settings or machine specific bug?

bschatzow, 2024-08-02:
> Yes, info level logging. The scanning duration is 6 seconds by default. So the disappearance happens before the end of scanning? It is interesting because if the filtering logic is faulty then the affected devices should not show up at all. What fitness machines do you have?

The Chrono on the spinning bikes.  Also any device it sees (hr strap) disappears as well.  With the change you recommended it works as before. If you want I can change back and send you the log files. Let me know.

MrCsabaToth, 2024-08-02:
> The Chrono on the spinning bikes. Also any device it sees (hr strap) disappears as well. With the change you recommended it works as before. If you want I can change back and send you the log files. Let me know.

Yes please: just because I cannot reproduce it is definitely a bug you can reproduce steadily, and I want to prevent other users encountering it. Turning of the filtering is not ideal, because then everything shows up in the scan starting from smart TVs, BLE locator tags (Tiles, Nuts, Apple Air Tags, ...), and many more. So the landscape is polluted with non compatible devices in the list. The intended (and default) mode is to not show incompatible devices.

So you have a Precor Spinner Power Chrono and heart rate strap(s)? Do you have anything else, like Life Fitness machines? Is this at home in a relatively isolated place, or rather in a gym setup where potentially there are many machines? (I'm asking this because I'm trying to figure out what could be the reason of this)

MrCsabaToth, 2024-08-02:
Note to self: maybe somehow the advertisement caching plays a role in this? How does something show up and then disappear? Another culprit could be FBP (Flutter Blue Plus): it was at 1.32.4 @ build 155, and 1.32.11 @ build 156

bschatzow, 2024-08-03:
It does not see my watch or the precor with the seeing changed

bschatzow, 2024-08-03:
> Yes please: just because I cannot reproduce it is definitely a bug you can reproduce steadily, and I want to prevent other users encountering it. Turning of the filtering is not ideal, because then [...]
This is at the same gym I have been at for the last 12 years.  The app never had this issue here before this release.

MrCsabaToth, 2024-08-03:
> I see a lot of stuff but not any of the precor bike with the changes you had me make. [DataExport_20240803T130420406564Z.zip](https://github.com/user-attachments/files/16481003/DataExport_20240803T130420406564Z.zip)

This is a database export primarily meant to haul data over to another device when you'd upgrade phones for example.
Could you rather click on the `Export Logs...` button? Thanks!

bschatzow, 2024-08-04:
> This is a database export primarily meant to haul data over to another device when you'd upgrade phones for example. Could you rather click on the `Export Logs...` button? Thanks!
Sorry.  Saw the wrong export.  Hopefully this still has the information in it that you need.

[DebugLogs_20240804T001354234760Z.zip](https://github.com/user-attachments/files/16484332/DebugLogs_20240804T001354234760Z.zip)

MrCsabaToth, 2024-08-04:
Thanks, I see that there are Life Fitness machines in that gym. Since that's the new support I was fiddling with that is highly likely the cause of this bug. Unfortunately there are no Life Fitness machines around me (I interact with them only few times a year when I'm in hotels for conferences, and the next time will be late September https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/462#issuecomment-2185523595).

However I'll be able to revise the code changes I made, but I'll only have time sometimes after a Monday job interview next week.

bschatzow, 2024-08-04:
No problem. Good luck with the interview. Not sure why this new code that searches for other types of machines makes the existing found devices disappear?

bschatzow, 2024-08-04:
A little more information. When I was not in the spin room the app saw my HRM plus and it stayed visible. As soon as I when in the spin room it saw the Chrono display and immediately disappeared.

MrCsabaToth, 2024-08-04:
Thanks for all the extra info. The Life Fitness machines are somewhat outlier because I can tell what type they are in a different way I used before. So I needed to change some logic around how I deem a machine supported or not. It's a good question why this would cause everything else to disappear (as far as I understood that's what happens), but we'll see.

bschatzow, 2024-08-04:
Get your new job and then go back to this!  Let me know if there is anything else you need from me to troubleshoot this.  Thanks again.
Bill

MrCsabaToth, 2024-08-05:
Suspicious commit ba0d17aaab312aee93d6d24b661f799f8322b66d

MrCsabaToth, 2024-08-05:
I don't see how that commit would cause everything to disappear. If for some reason all of the scanned devices' status would change to connectable = false that would cause them to disappear, however in the logs I see everything is connectable. Could you maybe record a video how this happens? Maybe I'll spot something.

bschatzow, 2024-08-06:
Not sure that this will give you anything? The devices show for less than a second. Most of the time I don't see anything.  With the setting changed I see lots of devices, but the Precor devices never show up.

bschatzow, 2024-08-06:
https://github.com/user-attachments/assets/9240678a-4b53-4b5d-b85e-6946c9549ad5

This is from home with Device filter off

bschatzow, 2024-08-06:
At home with device filter on.  I'll try to get both tomorrow from the gym.

https://github.com/user-attachments/assets/2c9eed2f-8be7-4fab-b8b1-be7ba97d6c14

bschatzow, 2024-08-06:
https://github.com/user-attachments/assets/09a00fa2-f2c5-4040-908d-66f72d395434

Disappearing devices

bschatzow, 2024-08-06:
No filter.  Tried to show the disappearing first and then no filter.  File was to big to to upload.  52 M.   If you need it, I can try again tomorrow.  By the way with no filter it finally shows up if I keep searching.

MrCsabaToth, 2024-08-06:
> No filter. Tried to show the disappearing first and then no filter. File was to big to to upload. 52 M. If you need it, I can try again tomorrow. By the way with no filter it finally shows up if I keep searching.

If you have that too big one can you share it on some cloud drive or any other way?

MrCsabaToth, 2024-08-06:
> screen-20240806-062704.mp4
> 
> Disappearing devices

That's a great shot. So to be clear, you pressed the scan button only once? I'm thinking of the scan button press can cause it somehow: it clears the list and then it doesn't reload anything. Maybe I need to hand over a custom build to you after all to see clearer if this is an issue caused by some FBP (Flutter Blue Plus) change.

Have you got any major Android OTA update? My 2022 Motorola Razr just got the Android 14 not so long ago. What phone are you with and what Android version?

bschatzow, 2024-08-06:
On the disappear I pressed only once.  On the no filter, I had to press multiple times before the precor display showed up.  I am on the latest Android Beta 15.x  Pixel 7 pro.  Not sure exact number.  I'll send it later to you. 

What I tried to capture was with filter, watch it disappear.  Turn off filter, hit scan button and you see it show up.  Problem is that the file is over 50Mb.

MrCsabaToth, 2024-08-06:
The one what you could upload here also shows the disappearance. It seems that they are displayed for almost 2 seconds. I wonder what happens if you increase the scan duration in the User Experience preferences from the default 6 seconds to 15 seconds. Would that make the list stay a little longer?

bschatzow, 2024-08-06:
I'll try it tomorrow and let you know

bschatzow, 2024-08-07:
No difference with the setting changed to 15. Disappears before you can select it. No filter works for now.

MrCsabaToth, 2024-08-07:
> No difference with the setting changed to 15. Disappears before you can select it. No filter works for now.

So if I understand correctly the disappearance delay is the same (1-2 seconds)? This means that the bug is not triggered when the scan ends, it's something else.

bschatzow, 2024-08-07:
Disappearance appears to me to be the same. Did not actually time it, but it seemed very quick.

bschatzow, 2024-08-08:
Today when I went to pair (filter off) it did not see the Chrono display. Turned filter on and Chrono was there for a couple of seconds. Turn filter off refreshed and Chrono found.

MrCsabaToth, 2024-09-18:
I haven't been able to reproduce this yet. Sometimes the opposite happens: there's nothing to display almost until the 6 second scan duration ends, at which point the device(s) show up.

bschatzow, 2024-09-19:
I have seen this as well, I just refresh it and lots of devices show up.

bschatzow, 2024-10-01:
Is there any more information that you need on this?

MrCsabaToth, 2024-10-02:
Thanks for asking. I traveled to my new job orientation and I made changes to the code which might affect some scanning behavior, although I'd suspect it'd mostly affect how devices are identified, but other situations could be complex.

This is the most suspicious change which may have affect: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/commit/84d4a0b224a19e220fdf22d79cac9696a243f6e0
This is another one: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/commit/ac0079c49cbe23ce5724ad286d540604aea3d256

I'll make a beta release, then kindly re-test. I'll make the release possibly next week or so. Other than that I'd need a method to reproduce the bug on my end which I never succeeded. Now that should also wait until you retest a new build. Let's hope the best. You already tried increasing the scan time.

bschatzow, 2024-10-02:
One of the first things you told me to try was increasing the scan time.  This did not help.

MrCsabaToth, 2024-10-07:
Beta build 157 is going through usual Play Store inspection and soon will start roll out

MrCsabaToth, 2024-10-14:
New build 158 is coming because of #523 but that has also crucial Flutter Blue Plus changes which can potentially affect scanning.

bschatzow, 2024-10-14:
When it is released, I'll test it and let you know if this fixes my issue as well.

MrCsabaToth, 2024-11-12:
There were more Flutter Blue Plus changes, eventually I'll have new beta release.

bschatzow, 2025-03-04:
Have you finished the beta for testing?

MrCsabaToth, 2025-03-04:
Yes, I had two releases this year: build 159 on Jan 1st to celebrate the new year and build 160 on Jan 26. Let me know if you experience the issue still. I'll have more releases, one in the spring for sure.

bschatzow, 2025-03-04:
Just tried to install the 1/26 beta and get "app not installed as package
conflicts with an existing package."

bschatzow, 2025-03-05:
The January beta fixes my issue.

closed 2025-03-05 by bschatzow

MrCsabaToth, 2025-03-05:
Let's hope it stays like that

MrCsabaToth, 2026-03-14:
I cannot look it up right now but it exposes the HR as standard Bluetooth HRM profile. But if the pixel 4 watch showed up in scans it must have exposed some kind of a standard profile, I am super curious what was that. You click on the tile and it shows extra info such as BT service UUID. Without the specific app, what was the icon in the tile, that's also telling: heart means HRM profile, bike means FTMS indoor bike, or power meter, or CSC, running means FTMS treadmill or RSC