TrackMyIndoorWorkout/TrackMyIndoorWorkout #513: Add Support for YOSUDA rower

MangoScango, 2024-08-04:
I have a cheap rower I got off [Amazon](https://www.amazon.com/gp/product/B0C1VGRX2P) that does support FTMS, but the app gives the error unable to infer sport of rower.

Here's the output of nRF Connect, is it possible to support? App logs attached as well: [DebugLogs20240804T202337505731Z.txt.gz](https://github.com/user-attachments/files/16488837/DebugLogs20240804T202337505731Z.txt.gz)

[image]

MrCsabaToth, 2024-08-05:
Question: when you are scanning what sport icon do I display next to that rower? Is it a question mark?

Normally at scan I can only see what important services the machine implements, 0x1826 means it's FTMS. On top of that FTMS machines implement a service data, which helps my app to distinguish what type of FTMS is it exactly:
[image]

In this case possibly (?) the machine doesn't give a clue to me, so I must connect to it to tell it apart what type of device is it. Fortunately I thought about this scenario (even though I don't have such a device). For some reason the Bluetooth protocol stack errors out when I perform this connection to enumerate the characteristics within the 0x1826 FTMS service. Ideally I'd be able to see the 0x2AD1 Rower Data and then determine that it's a rower device type. Although after that I'd offer you to pick an exact sport, because such an ergometer can be a rower, a kayak ergometer (like my Kayak Pro or Kayak First ergometers), a canoe ergometer, or even a swim (!) ergometer. The app doesn't get to that point doe to the crash bubbling up from the guts of FlutterBluePlus, the call stack:

```
discover,device.discoverServices; FlutterBluePlusException | setNotifyValue | android-code: 17 | GATT_INSUFFICIENT_RESOURCES; #0      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:280)
<asynchronous suspension>
#1      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:273)
<asynchronous suspension>
#2      DeviceBase.discover (package:track_my_indoor_exercise/devices/gadgets/device_base.dart:213)
<asynchronous suspension>
#3      FitnessEquipment.discover (package:track_my_indoor_exercise/devices/gadgets/fitness_equipment.dart:668)
<asynchronous suspension>
#4      FitnessEquipment.connectOnDemand (package:track_my_indoor_exercise/devices/gadgets/fitness_equipment.dart:514)
<asynchronous suspension>
#5      FindDevicesState.goToRecording (package:track_my_indoor_exercise/ui/find_devices.dart:628)
<asynchronous suspension>
#6      FindDevicesState.build.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:track_my_indoor_exercise/ui/find_devices.dart:995)
<asynchronous suspension>
```

MrCsabaToth, 2024-08-05:
Another developer ran into that https://stackoverflow.com/questions/78819399/flutter-blue-plus-how-to-handle-gatt-insufficient-resources
I filed an issue https://github.com/boskokg/flutter_blue_plus/issues/955

MangoScango, 2024-08-05:
I'll hop over there and offer to help troubleshoot. Later today I'll try editing my settings with a hex editor to force in the info for the rower in the device usage section to see if that gets me past it. It does show as a question mark now, sorry forgot to add that screeshot before.
[image]

MrCsabaToth, 2024-08-05:
I can build a custom APK with a custom `FlutterBluePlus` version where I add a proposed try-catch around that iOS compatibility `setNotify`. The thing is that it doesn't even if it's a brand rower the console and electronics may be outsourced to a shop where the implementation may not follow some common expected behavior. This goes even as far as some Rowers botch (!) the FTMS Rower Data protocol: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/291#issuecomment-1163798003

MrCsabaToth, 2024-08-05:
Hmmm, your rower does provide service data! I should rather look into why I cannot identify it as rower

MangoScango, 2024-08-05:
I can grab a HCL snoop log with it in a working app, if that would help.

MrCsabaToth, 2024-08-05:
> I can grab a HCL snoop log with it in a working app, if that would help.

No need. It looks like to me they errornously implemented the service data. The first byte there 1 means it's FTMS. The next two bytes encode the machine type in a bit flag manner. Almost looks like they mistaken the byte order and serializing those two byte in a big endian way instead of little endian. As far as I see all bluetooth multi byte data are little endian.

Since the bits 6-15 are reserved for future use, when I detect your machine it'll be clear that it's a mistake and I'll correct it. This will avoid the need of the preliminary service discovery and I can identify the type right away. There will be still a sport picker the first time as I mentioned, but once you pick one the app will remember that (in the device usage record - what you can also override if needed). The question will be what happens after that, did they implement the Rower Data according to the specifications or not. We'll see.

MangoScango, 2024-08-05:
So I was able to edit my Device usage table and get it recognized as a rower, but when I try to start a workout I get the error "Problem connecting to Unknown Generic Rower Ergometer". Heres the logs: [DebugLogs20240805T220322699745Z.txt.gz](https://github.com/user-attachments/files/16503194/DebugLogs20240805T220322699745Z.txt.gz)

[image]

MrCsabaToth, 2024-08-05:
What's happening is even though we could get over that machine type identification hurdle, the app still has to do the service discovery at some point. And it seems like no matter when that happens, that iOS compatibility `setNotify` still ruins the day. So I'll produce an apk for you which will have the try-catch, plus the identification workaround so you don't have to hack Isar.

MangoScango, 2024-08-05:
Sounds great, thanks for all your work on this!

MrCsabaToth, 2024-08-06:
Here is a GDrive share link of the custom build: https://drive.google.com/file/d/1p-sFwMV2R5AcBMgqzYm3FcZQzEmmmYTN/view?usp=drive_link

MangoScango, 2024-08-06:
All appears to be working now, didn't even need to pick the sport at discovery time. Debug logs Incase there's anything interesting in there [DebugLogs_20240806T114140298473Z.zip](https://github.com/user-attachments/files/16509722/DebugLogs_20240806T114140298473Z.zip)

[image]

MrCsabaToth, 2024-08-06:
Awesome! The only thing that you are on a custom build, so if ever you decide to hop back on the main build, you'd need to export the database and then import it (since there's no back-end, because there's no login either). First the fix will be only in a beta build and then only later in the production.

MrCsabaToth, 2024-08-06:
Let me know if the values seem right and sync OK to whatever platform of your choice. One thing stands out a little: 103 kCal seems a little high for 0.1 mile ~1 minute row. The machine seems to report several cumulative metrics: distance, stroke count, and calories.

The machine may not follow your workout schedule and when you start the next workout it starts from the previous point you left off (in the log it's 2965m distance, 2554 stroke count (looks almost too much for that distance) and 3669 kCal. This latter number looks way too much for that distance, but also correlates with the same distance / kCal ratio we see with your workout snippet.
My app provides two types of "tuning" if the values are not desired: power tuning and calorie tuning. Power tuning would be specifically important if the calorie was not communicated by the machine and my app would numerically integrate the calories from the wattage. This is not the case, but someone can also modify the calories. That works the following way:

1. After the workout while you are still on the measurement screen, or you navigate to the activity list and open the accordion for details and click on the flame icon.
2. You enter what calories would you expected instead of the reported one. Let's say for this workout above instead of 103 calories you expect 10.
3. The app calculates a 10% calorie ratio (very aggressive) and registers it for the particular device (by Bluetooth ID).
4. From then on all workouts on that particular machine will have this multiplier ratio (x0.1 in this case for 10%) applied.

You can view and edit all calorie tunes if you click on the flame icon not at the bottom of the workout accordion, but rather the icon in the circular menu. That lists all calorie tune entries.

You have also choice to calculate calories based on your heart rate. For that you need to configure your age, weight, gender, and optionally VO2Max. I apply a formula from a publication, however that also yields very high calories when I tried. But it can also be tuned with the same manner as the non heart rate one.

You can actually have to separate tunes: one for the machine, and one for a heart rate monitor. Which one is applied depends on if you turn on HR based calorie counting or not.

MangoScango, 2024-08-06:
Good catch with the calories! I don't intend to trust the calories from this thing anyway, but the display on the machine counts 0.4 calories per stroke, but the app is counting 4 calories per stroke. So the calibration feature already implemented should be enough to handle that.

MrCsabaToth, 2024-08-06:
Interesting. If you look at the standard https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.rower_data.xml (line 164-184), the energy features indicated by the C9 feature flag: Total Energy, Energy Per Hour, Energy Per Minute are clearly all have "resolution of 1". I believe that is true for probably all sports (Indoor Bike, and other data profiles). So I'm not sure where does the 10 divider comes from. There are some metrics which have 1 or 2 digit fractional resolutions, such as speed, distance.

If you look at the logged raw data, these are 5 bytes, -8 .. -4 (if we use Python style backward indexing) index. The first two are the total calories, the rest is 255d out which means invalid data (not supplied), but needs to be there due to the specification of C9.

Another possibility could be if there a byte offset in the data by error, but it wouldn't pass my sanity check then and the full packet just gets dropped. If I can distinguish your machine based on manufacturer name and being a rower I can inherit from the generic rower descriptor in my source and add this division by 10 for this specific device. Maybe I'll look into that.

MangoScango, 2024-08-07:
So going for a full workout on the rower, there was some unusual behavior with the calorie count with a dedicated heart rate monitor paired. While I was actively using the machine, the calorie count would drop to zero, until I would pause at which point the inflated calorie count from the machine would be displayed. However once I started using the machine again, the calorie count dropped back to zero. When I saved my workout, the count was at 0, and it was saved as 0 as well in the activity log. I made a recording of this behavior and exported logs again:

https://github.com/user-attachments/assets/af37b039-a90c-4fe0-99e5-0c05954cbbd6
[DebugLogs_20240806T232058108876Z.zip](https://github.com/user-attachments/files/16517113/DebugLogs_20240806T232058108876Z.zip)

MrCsabaToth, 2024-08-07:
Did you turn on the heart rate based calorie counting when that happened? Did you change any of the workout on-stage related settings? I'll extract this into another issue. Or we have an issue where the elapsed time flips between 0 and regular values depending on if the athlete is moving or not: #449 

I see one more data problem on the original workout screenshot: 0.01 mile distance over 1:05 pedalling with 1:40 / 500m average pace. The calculation would be ~1.1 / ~1.66 * 500m = ~330m = 0.2 miles. So the mileage is 1/20th what it should be?

MangoScango, 2024-08-07:
Yes, HR based calorie counting was enabled. I cleared the app data and tested again, that behavior only occurs when its turned on, with the rower specifically. If I don't have that turn on and use the calorie tuning feature, results are as expected. I have a treadmill that it works fine with.

After doing some more testing with the rower, I believe the distance is also off by a factor of 10. Here is a sample of the last test I did, app displays 0.03km, display on the rower showed "0.32", and all of my other tests gave similar results. Note the display on the rower doesn't give clear units, but the manufacturer recommended app, Kinomap  recorded a similar test as 310m, so I am pretty confident is is kilometers.
[DebugLogs_20240807T141938605706Z.zip](https://github.com/user-attachments/files/16530115/DebugLogs_20240807T141938605706Z.zip)
[image]

MrCsabaToth, 2024-08-07:
That's very interesting that Kinomap picks up the distance properly. I'll need to revisit and debug why this happens with my code. But since my app is used with many other rowers and kayak ergometers, this may mean that I indeed need to recognize your case and customize a version of my standard Generic Rower descriptor for Yosuda. Yosuda doesn't give any manufacturer data, and the Bluetooth name "H-181-..." doesn't seem to be too specific, so I guess I'll apply the workaround when I see a rower with no manufacturer data and the screwed up service data bytes. I'm dealing with a plugin issue now, so this will happen a week from now.

MrCsabaToth, 2024-09-18:
So far this workaround as I described should work without specifically detecting YOSUDA. The workaround focuses on the flipped service bytes.

MrCsabaToth, 2024-09-18:
@MangoScango let me know if any more issue emerges, we can reopen or file a follow-up issue

closed 2024-09-18 by MrCsabaToth