TrackMyIndoorWorkout/TrackMyIndoorWorkout #514: Average showing no information

bschatzow, 2024-08-09:
While trying to resolve my sensor disappearing issue, I have been playing with other settings and observing the changes.  I turned on average and the information is blank (see attached).

https://github.com/user-attachments/assets/ef9eba1b-c47f-4730-ae0b-3fad44872480

MrCsabaToth, 2024-08-26:
I have I have two problems lately:
1. The graphing library doesn't compile (regardless if I want to display graphs or not), Syncfusion promises fix mid September. I'm applying a local patch, but I would not release anything with that.
2. My AC was flooding my apartment two weeks ago. I'm renting, so the management has to deal with it. I was proactive and ended up ripping out some carpet (this was the 4th time we got flooded for various reasons - 2 times upstairs washer, one time upstairs toilet, and upstairs sink, so the carpets are well done). The thing is that management still haven't came around. Why this matters is that a lot of stuff from my room is scattered in the living room, bedroom, and now it's impossible to use my test fitness machines. At the spinning studio I won't be able to experiment.

bschatzow, 2024-08-26:
I understand. Life issues sometimes gets in the way of doing what we want to do.  On your AC it sounds like a clogged condesent drain.

MrCsabaToth, 2024-09-15:
The AC situation is solved last week and stuff is back in my closet so I can roll out my equipment. Hopefully this week. I just need time.

MrCsabaToth, 2024-09-18:
I tried to change these parameters in combination (while "Use heart rate based calorie counting" was on, but I don't think that influences this error on a kayak ergometer):
1. Workout Mode: Individual / circuit training
2. Time display mode: elapsed / moving / HIIT
3. Stage mode: instantly on stage or not
4. Statistic type: none / avg / max / alternating

The error showed up when I turned off the Stage mode. The instant on stage means that you don't have to signal with an extra click to the app that you are starting the workout. The app will automatically wait until it senses moving and starts the workout as soon as it senses any, and that's automatic and no click needed. This is supposed to be the default.

When it's turned off then the app is supposed to display the play button on the top right, signaling that you need to indicate when you really mean to be "on stage" and work out. Right now it's the stop button because somehow the app enhances into a state which is HF way between measurement, but in reality it's waiting off stage for you to nudge it on stage, and that's why the stats are also not calculating because it's not really measuring inside the workout yet.

However this is only my experience with a kayak ergometer, let me know if this comes with instant on stage 'on' as well for you.

Also I cannot find which bug was about the "Use heart rate based calorie counting" being on.

MrCsabaToth, 2024-09-18:
I tried some combination on the CHRONO as well and it seemed there as well that the Instant On-Stage mode turned off triggered the error.

This means that a simple workaround is to enable that to the default "on" ("checked") state to avoid the bug, until the non-instant on-stage mechanism is fixed.

MrCsabaToth, 2024-09-18:
This is essentially a UX issue.
1. The mechanism and state should be indicate in the top-right in one way or another and controlled from there and not from within the menu
2. The feature could be also more intelligent:
   - Before going on stage it could already record, and when the athlete signals to be on stage it could automatically save a workout postfixed with "warmup".
   - Once the effective workout is over and it is saved the cool-down could be automatically recorded the same fashion: with a "cool-down" postfix.
   - The only problem would be if an athlete would go on-and-off stage multiple times. Maybe an extra settings could control this behavior

bschatzow, 2024-10-01:
Do you need more info from me?

MrCsabaToth, 2024-10-02:
@bschatzow could you confirm my hypothesis in the previous comments? Here is how:
1. When you switch the instant on stage to on (checked, the default) then you don't experience the bug?
2. When you turn it off, can you put yourself on stage by signaling that in the menu? Once you put yourself on stage manually the stats should show.

bschatzow, 2024-10-02:
I have never touched the instant on stage.  Not sure where this even is.  I'll try and check it later when I get to the gym.  In any case, I have never gotten the average stat to show.

bschatzow, 2024-10-02:
I do not see where to turn this on / off

bschatzow, 2024-10-02:
Found it. And it works.

closed 2024-10-02 by bschatzow