TrackMyIndoorWorkout/TrackMyIndoorWorkout #519: Add apk releases on GH

ippocratis, 2024-09-15:
I don't use play store.
I like to use the git repos as installation source.
I see that the is an apk option in the download section in https://trackmyindoorworkout.github.io/ but  I'd kinda awkward to use apkpure as host.

Thanks

MrCsabaToth, 2024-09-15:
I'll consider that.
If someone has a Samsung the app is also in the Galaxy Store, or in the Huawei AppGallery in case of a Huawei device (#208). I have plans to get into the OPPO App Market, Xiaomi GetApps, Vivo App Store, realme APP store (#388), and HONOR store (#374).
Providing a direct apk download is a good idea. I'm filing an issue: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout.github.io/issues/36

Until then APKPure updates automatically and often more recent than the other app stores (only exception is the Google Play Store which is the most recent main distribution channel).
There are two more apk website, which I need to update because they lag behind as well:

1. https://www.apkfiles.com/apk-609581/track-my-indoor-workout
2. There was a 3rd one I forgot, I'm looking

MrCsabaToth, 2024-11-04:
Even though we can point to the binary from the website, the best place of the releases will be this (the app) repository, because I can even tie them to the specific release tag. I'll add the last two beta right now, and from now on this should be part of the release. We'll figure out later where to link the releases. It could be even from the changelog, but maybe it's easier from some other menu

MrCsabaToth, 2024-11-04:
This will make the whole app more professional! https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/releases

MrCsabaToth, 2024-11-12:
I'll close this because the releases is established and the website facing link is filed in that repository

closed 2024-11-12 by MrCsabaToth