TrackMyIndoorWorkout/TrackMyIndoorWorkout #535: Crash with message `Scudo ERROR: invalid chunk state when deallocating address` on version 1.1.158

dwalkes, 2024-11-30:
Was working great until this week, then started crashing as soon as I attempt to open.

Tried clearing cache, doesn't help.

Tried uninstall, reinstall, this fixed the issue.

I'm guessing it had something to do with the large amount of stored workouts I had, I was up to 1.1GB.  Feel free to close this if it isn't useful, just thought I'd share in case anyone else sees this.

Output from adb logcat:
```
11-30 16:38:24.261 24752 24752 I indoor_exercise: Using CollectorTypeCMC GC.
11-30 16:38:24.267 24752 24752 E indoor_exercise: Not starting debugger since process cannot load the jdwp agent.
11-30 16:38:24.272 24752 24752 D nativeloader: Load libframework-connectivity-tiramisu-jni.so using APEX ns com_android_tethering for caller /apex/com.android.tethering/javalib/framework-connectivity-t.jar: ok
11-30 16:38:24.283 24752 24752 D CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10246; state: ENABLED
11-30 16:38:24.283 24752 24752 D CompatibilityChangeReporter: Compat change id reported: 242716250; UID 10246; state: ENABLED
11-30 16:38:24.297 24752 24752 W ziparchive: Unable to open '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.arm64_v8a.dm': No such file or directory
11-30 16:38:24.297 24752 24752 W ziparchive: Unable to open '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.arm64_v8a.dm': No such file or directory
11-30 16:38:24.299 24752 24752 W indoor_exercise: Entry not found
11-30 16:38:24.299 24752 24752 W ziparchive: Unable to open '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.en.dm': No such file or directory
11-30 16:38:24.299 24752 24752 W ziparchive: Unable to open '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.en.dm': No such file or directory
11-30 16:38:24.299 24752 24752 W indoor_exercise: Entry not found
11-30 16:38:24.299 24752 24752 W ziparchive: Unable to open '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.es.dm': No such file or directory
11-30 16:38:24.299 24752 24752 W ziparchive: Unable to open '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.es.dm': No such file or directory
11-30 16:38:24.300 24752 24752 W indoor_exercise: Entry not found
11-30 16:38:24.300 24752 24752 W ziparchive: Unable to open '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.xxhdpi.dm': No such file or directory
11-30 16:38:24.300 24752 24752 W ziparchive: Unable to open '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.xxhdpi.dm': No such file or directory
11-30 16:38:24.300 24752 24752 W indoor_exercise: Entry not found
11-30 16:38:24.301 24752 24752 D nativeloader: Configuring clns-4 for other apk /data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/base.apk:/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.arm64_v8a.apk:/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.en.apk:/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.es.apk:/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.xxhdpi.apk. target_sdk_version=34, uses_libraries=, library_path=/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/lib/arm64:/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/base.apk!/lib/arm64-v8a:/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.arm64_v8a.apk
11-30 16:38:24.308 24752 24752 V GraphicsEnvironment: Currently set values for:
11-30 16:38:24.308 24752 24752 V GraphicsEnvironment:   angle_gl_driver_selection_pkgs=[com.android.angle, com.linecorp.b612.android, com.campmobile.snow, com.google.android.apps.tachyon]
11-30 16:38:24.308 24752 24752 V GraphicsEnvironment:   angle_gl_driver_selection_values=[angle, native, native, native]
11-30 16:38:24.308 24752 24752 V GraphicsEnvironment: dev.csaba.track_my_indoor_exercise is not listed in per-application setting
11-30 16:38:24.308 24752 24752 V GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
11-30 16:38:24.321 24752 24768 D vulkan  : searching for layers in '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/lib/arm64'
11-30 16:38:24.322 24752 24768 D vulkan  : searching for layers in '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/base.apk!/lib/arm64-v8a'
11-30 16:38:24.322 24752 24768 D vulkan  : searching for layers in '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.arm64_v8a.apk!/lib/arm64-v8a'
11-30 16:38:24.323 24752 24768 D vulkan  : searching for layers in '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.en.apk!/lib/arm64-v8a'
11-30 16:38:24.323 24752 24768 D vulkan  : searching for layers in '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.es.apk!/lib/arm64-v8a'
11-30 16:38:24.323 24752 24768 D vulkan  : searching for layers in '/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.xxhdpi.apk!/lib/arm64-v8a'
11-30 16:38:24.343 24752 24770 D nativeloader: Load /data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.arm64_v8a.apk!/lib/arm64-v8a/libflutter.so using ns clns-4 from class loader (caller=/data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/base.apk): ok
11-30 16:38:24.360 24752 24752 W indoor_exercise: type=1400 audit(0.0:223969): avc:  denied  { read } for  name="max_map_count" dev="proc" ino=36382365 scontext=u:r:untrusted_app:s0:c246,c256,c512,c768 tcontext=u:object_r:proc_max_map_count:s0 tclass=file permissive=0 app=dev.csaba.track_my_indoor_exercise
11-30 16:38:24.469 24752 24752 D com.llfbandit.app_links: Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=dev.csaba.track_my_indoor_exercise/io.flutter.embedding.android.FlutterActivity bnds=[440,315][640,627] }
11-30 16:38:24.469 24752 24752 D FlutterArchivePlugin: onAttachedToEngine - IN
11-30 16:38:24.469 24752 24752 D FlutterArchivePlugin: doOnAttachedToEngine - IN
11-30 16:38:24.469 24752 24752 D FlutterArchivePlugin: doOnAttachedToEngine - OUT
11-30 16:38:24.469 24752 24752 D FlutterArchivePlugin: onAttachedToEngine - OUT
11-30 16:38:24.469 24752 24752 D [FBP-Android]: [FBP] onAttachedToEngine
11-30 16:38:24.471 24752 24752 D [FBP-Android]: [FBP] onAttachedToActivity
11-30 16:38:24.493 24752 24752 D CompatibilityChangeReporter: Compat change id reported: 279646685; UID 10246; state: DISABLED
11-30 16:38:24.500 24752 24752 I ContentCaptureHelper: Setting logging level to OFF
11-30 16:38:24.504 24752 24752 D CompatibilityChangeReporter: Compat change id reported: 237531167; UID 10246; state: DISABLED
11-30 16:38:24.532 24752 24752 W isarworker: type=1400 audit(0.0:223970): avc:  denied  { search } for  name="vendor" dev="tmpfs" ino=2 scontext=u:r:untrusted_app:s0:c246,c256,c512,c768 tcontext=u:object_r:mnt_vendor_file:s0 tclass=dir permissive=0 app=dev.csaba.track_my_indoor_exercise
11-30 16:38:24.532 24752 24752 W isarworker: type=1400 audit(0.0:223971): avc:  denied  { search } for  name="vendor" dev="tmpfs" ino=2 scontext=u:r:untrusted_app:s0:c246,c256,c512,c768 tcontext=u:object_r:mnt_vendor_file:s0 tclass=dir permissive=0 app=dev.csaba.track_my_indoor_exercise
11-30 16:38:24.532 24752 24752 W isarworker: type=1400 audit(0.0:223972): avc:  denied  { search } for  name="vendor" dev="tmpfs" ino=2 scontext=u:r:untrusted_app:s0:c246,c256,c512,c768 tcontext=u:object_r:mnt_vendor_file:s0 tclass=dir permissive=0 app=dev.csaba.track_my_indoor_exercise
[... cut ...]
11-30 16:38:25.043 24826 24826 F DEBUG   :       #78 pc 000000000047772c  /data/app/~~YYscSDNlZMeLNGDyzBZ04w==/dev.csaba.track_my_indoor_exercise-j1z6Tb_1BGX2R-YGmAMu2w==/split_config.arm64_v8a.apk!libflutter.so (offset 0x9c8000) (BuildId: 77711dc3e28ead87416b4e79bb5815b725651d28)
11-30 16:38:25.043 24826 24826 F DEBUG   :       #79 pc 000000000006efbc  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+204) (BuildId: 1d36f8ae6e0af6158793abea7d4f4f2b)
11-30 16:38:25.043 24826 24826 F DEBUG   :       #80 pc 0000000000060d60  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 1d36f8ae6e0af6158793abea7d4f4f2b)
```

MrCsabaToth, 2024-12-14:
So sorry to hear that you experienced crashes like that. Do you know how many workouts did you have approximately? I'd need to reproduce this somehow.
It looks a little different than the OOM (out of memory) I experienced with the full DB export (#500)

MrCsabaToth, 2024-12-14:
If you have any full DB export maybe I can base off a stress test on that.

dwalkes, 2024-12-14:
@MrCsabaToth sorry I didn't have any backup and didn't have any way to access anything in this state other than the logcat as the app was crashing immediately. I should have used a backup phone but I wanted to make sure it would work after reinstall. I think I had just under 200 workouts logged.

dwalkes, 2024-12-14:
I should mention I started eating low storage messages from my phone shortly after so I think this was probably the issue.

MrCsabaToth, 2024-12-15:
That could be the reason. BTW the next release (159) will remove the Floor support. For more than a year I kept around both the Sqlite (under Floor) and the isar database around in case someone is late upgrading. Now this is over and I yanked out the Sqlite, this hopefully will help with space somewhat. I think I had close to 200 or more activities before.

I'll think what the app could do to not crash and how to handle full storage. This can be as hard as OOM.