TrackMyIndoorWorkout/TrackMyIndoorWorkout #537: "virtual power" for bicycle

opendever, 2024-12-20:
some bicycle app needed power data, 
so the function calculates "virtual power" based on speed sensor or heartrate sensor data,than broadcasted on BLE,
e.g. [vpower](https://github.com/oldnapalm/vpower)

cagnulein, 2024-12-20:
@opendever i don't want to stamp the @MrCsabaToth 's feet but if you want you can do already with https://github.com/cagnulein/qdomyos-zwift/

opendever, 2024-12-20:
> @opendever i don't want to stamp the @MrCsabaToth 's feet but if you want you can do already with https://github.com/cagnulein/qdomyos-zwift/

I tried for a while, but it was a bit complicated for me.

cagnulein, 2024-12-20:
@opendever Let me know if I can help you

opendever, 2024-12-20:
> @opendever Let me know if I can help you

I haven't found the "virtual power" function in qdomyos-zwift yet (-_-)

cagnulein, 2024-12-20:
@opendever simply set your heart rate sensor in the heart rate belt name setting and qz will auto calcolate the power, you don't need to enable anything else

MrCsabaToth, 2024-12-21:
So my app is calculating power data (#64, #240, #287). To transmit that is a whole other ballgame: the phone would need to be put into a so called Bluetooth "peripheral" mode. I haven't found a Flutter plugin yet which would be able to transmit data. All of them I saw so far only augmented the services and characteristics to advertise. #389

MrCsabaToth, 2024-12-22:
@opendever I have a workaround if you are a developer, has Android debug skills, and also have Linux or able to adapt a Linux python script to your platform (Windows, MacOS): #390 https://github.com/TrackMyIndoorWorkout/FTMSServer/

This requires:
1. Tethering your phone via USB
2. Starting the Python server part of that solution
3. Port forward via `adb`
4. Configuring the local server port in my app

With this the augmented FTMS compatible endpoint will be transmitting with your Laptop's wifi's Bluetooth chip and Linux's BlueZ Bluetooth tech stack. I was able to successfully connect Kinomap to that augmented Bluetooth peripheral.

If not that workaround, the most likely solution is a separate app which is specialized for formulating this augmented peripheral (like what the Linux BlueZ server script does in https://github.com/TrackMyIndoorWorkout/FTMSServer/). Then somehow connect the Track My Indoor Workout to that application and bridge the data with a constant data flow, while both applications stay active and not get sidelined or "killed" by the phone manufacturers logic. Possibly even needs background process which is a nightmare right now from app privilege and store acceptance perspective. So a lot of engineering, refactoring, and other things are involved. But it's definitely something I desire.