TrackMyIndoorWorkout/TrackMyIndoorWorkout #540: File import doesn't seem to work

MrCsabaToth, 2025-01-02:
Maybe it's newer version of the file picker: I'm picking file from my Google Drive and the file picker doesn't fill in the path.

MrCsabaToth, 2025-01-02:
I wonder if this is related to #319 commit "Custom file import restriction for CSV files doesn't work, doesn't let import CSV files" https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/commit/6157af1ebe8b795b89bd9aec2fc78decfabbb6f5

MrCsabaToth, 2025-01-02:
https://github.com/miguelpruivo/flutter_file_picker/issues/1658

MrCsabaToth, 2025-01-02:
Maybe it's this (permission problem): https://github.com/miguelpruivo/flutter_file_picker/issues/1652

vicajilau, 2025-01-08:
Mmm... That's weird. Can you give me the version of Flutter, dependency, platform and code you use? I will try to replicate your problem 🤗

MrCsabaToth, 2025-01-09:
Latest stable, the code is this repository and I am trying to import a workout. However I don't want to blame only the picker: after picking a file from Google drive didn't work, I downloaded the CSVs locally. I got the path of those files and if I paste those into the dialog (essentially shortcutting the file picker), then I get a file permission error (cannot access file). Which is crazy, it's the device download folder. As you can see in the repo's Android manifest I even have some permissions declared, however when I check the app's permissions in Android I don't see anything storage related. I wonder if this is because I bumped the target API level to 34? I'm not sure what to do.

MrCsabaToth, 2025-01-12:
I tried the https://pub.dev/packages/flutter_document_picker/example but it didn't work, but I got the dreaded missing plugin exception:
```
I/flutter (29648): MissingPluginException(No implementation found for method pickDocument on channel flutter_document_picker)
I/flutter (29648): trace:
I/flutter (29648): #0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:332:7)
I/flutter (29648): <asynchronous suspension>
I/flutter (29648): #1      FlutterDocumentPicker.openDocument (package:flutter_document_picker/flutter_document_picker.dart:11:21)
I/flutter (29648): <asynchronous suspension>
I/flutter (29648): #2      ImportFormState.build.<anonymous closure> (package:track_my_indoor_exercise/ui/import_form.dart:94:36)
I/flutter (29648): <asynchronous suspension>
```
and this time I cannot fix that because the plugin didn't even register itself in the GeneratedPluginRegistrant.java! So I couldn't even add a workaround in the proguard config, and probably that's the reason for the error.

MrCsabaToth, 2025-01-13:
Tried to restrict storage permissions to earlier SDK versions only because the AI assistants (Gemini, Perplexity) told me that they can interfere with the Android SAF type of access what the file picker use. So far that didn't help.
Possibly we need to reintroduce permission handler, but what permission should we use?

MrCsabaToth, 2025-01-13:
Looks like https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/commit/06307971aee8c01668980904bd4f1fbbd4e111c3 fixed the issue. Another issue kept interfering with the Android build and probably that's why the change didn't apply originally when I tried.
The issue was that I left a section at the end of the `pubspec.yaml` for the Android-only heart rate monitoring:
```
  plugin:
    platforms:
      android:
        package: heart_rate_flutter
        pluginClass: HeartRateFlutter
```
That somehow kept the framework from generating the `GeneratedPLuginRegistrant.java` and other Gradle related changes and upkeeps (like adding a new task for a new package or other related files of that build).

MrCsabaToth, 2025-02-09:
This is still faulty to some extent

MrCsabaToth, 2025-02-09:
https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/commit/fc77b0ee871a418b637fe75275b67bf1adb2b19b with the 8.3.1 file picker has an important fix regarding this: https://github.com/miguelpruivo/flutter_file_picker/issues/1700 and https://github.com/miguelpruivo/flutter_file_picker/issues/1404

MrCsabaToth, 2025-02-15:
It works now.

commit 03103d8d (2025-01-13): Reverting back to file_picker and removing more dormant flutter_document_picker because the issue seems to be fixed #540
closed 2025-02-15 by MrCsabaToth