TrackMyIndoorWorkout/TrackMyIndoorWorkout #543: User reports distance discrepancy on Schwinn IC4

MrCsabaToth, 2025-01-13:
Reported by Stacey Celano:

> I recently started using your app, "Track my Indoor Workout" to export data from my Schwinn IC4 to Strava. I am having a problem with the distance and mph metrics that I am hoping you can help me with.

> The bike metrics are set to mph for speed and miles for distance. The app states that it is recording in mph and miles. However, while the mph metric is correct in the app, the distance is not. It is recording miles as if the bike is set to kilometers (for example, if I have ridden 5 miles according to the bike's screen, the app tells me I have ridden 3.1 miles, the exact conversion as if the bike were in kilometers). So all of my miles are only 62% what they should be.

> See attached sample of a ride on the bike vs what the app shows for a half-mile ride (the app shows correct average speed of 12.92 mph for 2.5 minutes - which should be a distance of .5 miles as is shown by the bike's screen - but the app incorrectly shows the distance as .31 miles.)

> Can you let me know if there is a fix for this? I have tried toggling the switch for "unit system" in the preferences tab, but I have the same problem whether that box is checked or unchecked.

[image]

[image]

MrCsabaToth, 2025-01-13:
I'm asking Stacey to record short workouts in both imperial unit system and metric unit system. Turn on logging for each separately and export the logs, and also the workouts themselves, in CSV format.

From her screenshot there's definitely a discrepancy. If she spinned for ~2.5 minutes with 12.29 mph average, then that would be 12.29 / 24 ~= 0.51 miles instead of 0.31 (60 / 2.5 = 24).

MrCsabaToth, 2025-01-17:
Stacey did a spectacular job by trying all possible 4 combinations of the app being in imperial or metric mode and the bike console being in imperial or metric mode:

 - bike in MILES, app unit system in MILES (box unchecked)
Ride B - bike in MILES, app unit system in KM (box checked)
Ride C - bike in KM, app unit system in MILES (box unchecked)
Ride D - bike in KM, app unit system in KM (box checked)

| Workout | Console units | App units | Console distance | Bike distance   | time  | avg speed | Match |
|---------|---------------|-----------|------------------|-----------------|-------|-----------|-------|
| Ride A  | Miles         | Miles     | 4.93 mi (20:04)  | 3.00 mi (20:13) | 20:01 | 14.71 mph | NO    |
| Ride B  | Miles         | Km        | 5.00 mi (20:02)  | 4.92 km (20:01) | 20:01 | 24.12 kmh | NO    |
| Ride C  | Km            | Miles     | 8.44 km (20:05)  | 5.17 mi (20:02) | 20:01 | 15.78 mph | YES!  |
| Ride D  | Km            | Km        | 8.26 km (20:04)  | 8.13 km (20:03) | 20:03 | 24.89 kmh | YES!  |

MrCsabaToth, 2025-01-17:
This would mean that when the bike console is in km mode the app functions fine regardless of it is in miles or km mode. We know that the Bluetooth FTMS Indoor Bike Data is such a standard that is always communicating the speed in km/h units. The application does not know which mode the console is in! This would also corroborate that something is actually bad with the console when it's in miles mode.
However! The IC4 console neither supplies calories nor distance. So this means that the application is the one which computes the distance in all cases. Looking into the workout data it seems that the speeds are always int he right range (around 25kmh) so I don't understand how the app computes bad distance.
The core calculations don't depend on unit system: the FTMS Bike Data is in km/h and the application's native persistence unit is also km/h, so there's no conversion needed. When the distance is estimated the app converts the km/h into m/s and then multiples that with the fractional time since the last packet. This logic is absolutely unit system independent and is always the same for all combinations. We perform conversion only for displaying the metrics.
If there was any problem with the accumulation logic (for example it would slowly bleed some time and therefore would loose distance gradually with the exact 1.609 rate, then that would come with all the four ride combinations.
However the ride exports show that the speed is correct for all rides.

[Ride A_Activity_1-15-2025_07-55-54.csv](https://github.com/user-attachments/files/18450386/Ride.A_Activity_1-15-2025_07-55-54.csv)

[Ride B_Activity_1-15-2025_08-23-36.csv](https://github.com/user-attachments/files/18450391/Ride.B_Activity_1-15-2025_08-23-36.csv)

[Ride C_Activity_1-15-2025_08-56-20.csv](https://github.com/user-attachments/files/18450396/Ride.C_Activity_1-15-2025_08-56-20.csv)

[Ride D_Activity_1-15-2025_09-23-07.csv](https://github.com/user-attachments/files/18450397/Ride.D_Activity_1-15-2025_09-23-07.csv)

MrCsabaToth, 2025-01-26:
I'm dealing with serious Bluetooth library issues #547. Until I sort that out and could debug this a workaround could be a recalculate feature: this can at least fix the distance after the workout since the speed seems to be correct in all CSVs. The recalculate is an existing feature but only visible at debug, not even in the beta. So I need to move it out of the debug guard for all users to access

dwalkes, 2025-02-06:
I have the same Schwinn IC4 and I have noticed a discrepancy on my bike as well, sorry I didn't report it.  I only really pay attention to the miles reported by the app.  I didn't think it was this far off though, I'll pay more attention on my next ride.

MrCsabaToth, 2025-02-09:
The file import works now, I imported the all four workouts, and I can confirm that the recalculate feature could fix the A and B activity:

[Activity_1-15-2025_04-55-54.csv](https://github.com/user-attachments/files/18722961/Activity_1-15-2025_04-55-54.csv)

[Activity_1-15-2025_05-23-36.csv](https://github.com/user-attachments/files/18722962/Activity_1-15-2025_05-23-36.csv)

dwalkes, 2025-02-14:
Here's what I noticed in km mode

[image]
Bike shows 18.81 km
[image]
App shows 11.57 miles (exact match would be 11.68miles)

After switching to miles mode

[image]
Bike shows 11.34 miles

[image]
App shows 11.2 miles

So it looks like I'm getting a match with the bike in either KM or miles modes with my app in miles mode.

MrCsabaToth, 2025-02-14:
Thanks so much for checking. So essentially you could not reproduce. There must be something else causing this, some settings I need to figure out.