TrackMyIndoorWorkout/TrackMyIndoorWorkout #549: Being able to convert the resistance level by a conversion table rules (mainly Peloton usage)

MrCsabaToth, 2025-01-30:
Great idea advised asked by Bob Keown:
"Is there a way to add a feature in your app for Peloton riders using third party, FTMS capable bikes?  The only thing missing in your app to make this a great solution for those wanting to use it for Peloton classes is to create a resistance conversion factor (call it Fitness Class Conversion) and allow the user to place the resistance levels in a conversion factor table.  For example, there are two widely used conversion resistance conversion tables for Schwinn/Bowflex users to use when taking a Peloton class (The one I use is Peloton Resistance Level 30=Schwinn Resistance Level 3, Peloton 35=Schwinn 6, etc.).  If the user can prepolulate this data into a table and then the conversion shows up in the app when their Schwinn is set at Resistance 3, then Peloton 30 also shows up in the app."

I can see this working well, I've seen conversion tables. Obviously this is a display only feature so far, it won't control resistance level. We could provide well known presets for well known bikes for Peloton.

When people want to customize that could be maybe with SyncFusion DataGrid Editing https://help.syncfusion.com/flutter/datagrid/editing

If it's a completely free configuration it'd be a non trivial view:
* A list of integer pairs
* The numbers should be monotone increasing in their own column
* User would edit with spinner combine edit means (integers)
* User should be able to delete and add rows, possibly also reorder

MrCsabaToth, 2025-01-30:
Hmmm, something is not totally clear to me: on the conversion table here it states "at 100 rpm" https://www.lambertslately.com/how-to-convert-peloton-resistance-schwinn-ic4. And then gives the conversion table. So if someone spins with 60 rpm, 80 rpm, 120 rpm it require different conversion rules? Are the Peloton number related to Peloton bike resistance levels, or rather some kind of an effort / exertion - which would rather be proportional to the Watts, which is the combination of the resistance level and the cadence?
We need to do more research https://www.reddit.com/r/SchwinnIC4_BowflexC6/comments/10l3tka/ic4_to_peloton_conversion/

MrCsabaToth, 2025-03-05:
Bob's other takes to my questions:

"I believe that two other apps (the QZ app and mPaceLine) have somehow integrated the Peloton’s publicized output = cadence x resistance (google Peloton Cadence Resistance Output) information to create profiles within their apps that allows those apps to provide peloton resistance information.  I have used the QZ app on several echelon bikes, a Proform and NordicTrack bike, a Schwinn IC4 and now on my Bowflex IC Bike SE.  Eack bike had the ability to produce a wattage output. When researching what the equivalent Peloton resistance was for each bike, when applying the app to the bike and running the class, the converted Peloton resistance seemed to always matchup in the app.  Both those apps are somehow taking the output produced on those bikes,matching that output to the widely publicized Peloton output charts, and then having their app display the converted resistance.

Because many people use cadence sensors and heart rate monitors to reproduce a rudimentary Peloton experience (example most Sunny Bikes), the downside to many of those setups is that they don’t provide a wattage output that can be accessed by a third party app.  For those riders, having an ability to input a value in an app that they feel is equivalent to Peloton’s level 30, 35, etc. would be ideal because it takes the guess work out of it during the class. 

Where your app could be very ideal is to have that same ability and allow users to input a perceived resistance level for Apple Fitness classes.  If I had the ability to input on the front end what I perceived to be an easy, moderate, hard, and all out effort as called out by their instructors, then I again have an ability to set up a perceived level threshold prior to the class instead of during it. Other YouTube spin classes also use similar user-perceived effort levels to guide users through a class.

So to answer your question, I would just provide a feature in your app to let the user enter those values and provide some instructions on how to input the values plus provide the IC4 to Peloton’s conversion plus an Apple Fitness+ as examples of this capability. Then let the users come up with their own values and post them on a wiki or Facebook page.  You will be surprised how creative people get.  Plus that might avoid any potential litigation from Peloton if there was a formula involved that could be patented."

---

"This table may shed some more light (https://www.reddit.com/r/pelotoncycle/wiki/index/resistancechart/). I have read several of these sites as well and the Lambert's Lately site seems to be referenced quite a bit.  When I take my Bowflex and spin up to 80 cadence, I can see that my wattage is approximately 112.  From there you will see that my resistance level on my Schwinn is reporting at Resistance Level 12.  If I kept the same resistance level 12 on the same bike and then spin up to 100 rpm, my wattage will be around 160-165. When comparing my Schwinn metrics to the Lambert's Lately table, it was spot on.  That is basically how these charts are formulated.  You could use either 80 or 100 or both as a guideline.

But again, seeing that many bikes don't have the capability to provide output (Sunny, Joroto, Yosuda, Pooboo, any bike without metric reporting console), allowing the user to input whatever data they choose into a table  provides so much more flexibility for the user (and widens your customer base significantly)."

---

"When in a Peloton class and when using a non-Peloton bike, the instructor provides two callouts that the rider must follow - cadence and resistance.  The cadence is the easy part. Many people just buy a Bluetooth cadence sensor and place it on the crank arm of their bike.  This cadence, btw, can be displayed on the screen in the Peloton app.  That cadence readout in the Peloton app makes it fairly easy to follow along. The resistance is where it gets tricky.  The Peloton app will only display a resistance range (for example level 25-35 resistance). For bikes that are not “smart bikes”, the rider must guess by feel what a 25-35 resistance level feels like with no real basis for determining that.  

For riders with smart bikes such as the Schwinn IC4 or Bowflex C6, Echelon, Merach, Joroto, Nordictrack, etc. those bikes have a console that shows the bike’s cadence and resistance level, while some also provide a wattage output probably using an algorithm that you mentioned taking cadence, resistance, and most likely the riders age, sex, weight that was originally input in the riders app or console during setup.  I really can’t confirm if the weight, sex, age is actually used as a factor for determining wattage, but I do know that the bike must produce a wattage output number either in an app or the bike’s console in order to determine what the conversion factor is between a Peloton bike and the bike you are riding. The weight of the bike is a non-factor.

From my understanding, for calibration purposes, Peloton has published what the expected output levels should be for their bikes and this is determined by the what the wattage readout is at 80 and 100 cadence for resistance levels 25, 30, 35, 40, 45 and 50.  All the rider has to do now to figure out what the Peloton equivalent resistance level for their bike is to look at that Peloton calibration table to see what the wattage output is for cadence 80, resistance level 25 (for example : 56-60 watts).  Then spin up to 80 cadence and turn their resistance knob until they get an output between 56-60 watts and then write the resistance number down where their bike is currently set at.  Keep repeating that process for level 30, 35, etc and before you know it, your inexpensive spin bike produces a cadence and resistance very similar to a Peloton bike."

MrCsabaToth, 2025-03-05:
Conclusion and Apple Fitness

Q: "So it is indeed a resistance level to another scale conversion in case of Peloton.
In case of Apple Fitness is it rather wattage ranges to the perceived effort?"

"For the purposes of using your app with Apple Fitness, correct.  The user on Apple Fitness has to determine on their own what their Easy, Moderate, Hard, and all out efforts are.  You could also use heart rate zones instead of wattage to identify what those efforts are and in Apple Fitness case, would probably be more accurate than wattage.  Again, this is data the rider would need to pre-populate in the app, just like the resistance conversion table if the rider was doing Peloton instead of Apple Fitness."

MrCsabaToth, 2025-03-05:
Bob's example for a rowing machine, we should keep that in mind when designing:

[image]

[image]

[image]

[image]

cagnulein, 2025-03-05:
QZ developer here if you have any questions.

MrCsabaToth, 2025-03-10:
> QZ developer here if you have any questions.

Hey @cagnulein, actually, could you describe what you do in QZ about Peloton levels? When Bob writes "other apps (the QZ app and mPaceLine) have somehow integrated the Peloton’s publicized output = cadence x resistance (google Peloton Cadence Resistance Output) information to create profiles within their apps that allows those apps to provide peloton resistance information. I have used the QZ app on several echelon bikes, a Proform and NordicTrack bike, a Schwinn IC4 and now on my Bowflex IC Bike SE. Eack bike had the ability to produce a wattage output. When researching what the equivalent Peloton resistance was for each bike, when applying the app to the bike and running the class, the converted Peloton resistance seemed to always matchup in the app. Both those apps are somehow taking the output produced on those bikes,matching that output to the widely publicized Peloton output charts, and then having their app display the converted resistance."

cagnulein, 2025-03-10:
Qz translates the resistance levels of the bike to peloton resistance
levels. For each bike there is a conversion function

Roberto Viola
Software engineer and open source enthusiast
http://robertoviola.cloud