TrackMyIndoorWorkout/TrackMyIndoorWorkout #553: No average for rpms

bschatzow, 2025-04-04:
I just noticed that the Rpms does not show an average. The screen on the bike does.  All other sensors are working correctly. 

[image]

bschatzow, 2025-04-04:
Not sure what I am suppose to change @dosu?  The data is on the bike but not getting to the app.  It did work in the past.

bschatzow, 2025-04-06:
I have no idea what this is telling me to do? I am connecting to the Chrono display.  The information is on the bike's chrono display but not in the app.

MrCsabaToth, 2025-04-06:
Sorry Bill, just ignore the bot for now. It tried to help and be smart.

MrCsabaToth, 2025-04-06:
I think I found the bug. We are far from AGI (Artificial General Intelligence) if an AI specialized in coding (=not general) cannot find that bug.
I'll release a beta today and promote it to production soon next week.
Thanks for the good find! Now I just don't understand how it works for me? Or maybe I just didn't notice because I am focusing on watts only.

MrCsabaToth, 2025-04-07:
The fix is one character. The #552 feature introduced it. It's good that I didn't introduce a show/hide logic for all the metrics yet. A logical expression's clause was missing a negation.

MrCsabaToth, 2025-04-07:
Fix release published, beta 162

bschatzow, 2025-04-07:
Confirmed as fixed.  Thanks for the quick fix.

closed 2025-04-07 by bschatzow