TrackMyIndoorWorkout/TrackMyIndoorWorkout #564: Allow upload and save of virtual workout (FIT)

MrCsabaToth, 2025-05-11:
Requested by Sergei.
Currently the app uploads Ride and Run in non virtual, because the app was originally geared for GPS interpolation. It'd be actually good to allow for virtual version of those workouts for upload / save.

Sergei's Strava specific tips:

## First

"I've used the  "Export original" feature of Strava to download the fit file was created by your app
and then uploaded it back but with "activity_type" API field set to "virtualride" as well as
the "trainer" API parameter was set to true. The activity was correctly created with the type VirtualRide as expected, and a road bike was not hooked to it.

I looked into the app's source code and I notices in the file upload/strava/upload.dart the parameter "activity_type" is completely missed and "trainer" is hardcoded to false. This API description (https://developers.strava.com/docs/reference/#api-Uploads-createUpload) does not mention the "activity_type" parameter, but this one https://developers.strava.com/docs/uploads/ suggests to (optionally) set it or, even better, set the parameter "sport_type"."

## Second

"I think, sending the "sport_type" API parameter may still be necessary, since the fit file's sub_sport is seemingly ignored: I tried to modify the field value (the file originally had the value of 0 ) to different values, like 5 (spin), 6 (indoor) but still got "Ride" and my bike linked at Strava.  However, when I tried to set the "sport" field to 4 (equipment) + sub_sport 6, I got the Strava activity type "Workout". That was also not what I wanted.

Only when I set the activity_type API parameter along with the parameter tracker=true, the uploaded original fit file created an activity marked as VirtualRide and a default real bike was not linked to it."

MrCsabaToth, 2025-05-18:
Need to be careful with the SUUNTO FIT, which is restrictive.

dvboxer, 2026-11-13:
I'd also like to have this implemented. As I use the app not only with my Schwinn 800IC but also with my RowERG concept 2 please make “virtual” workout the default (it usually _is_ virtual anyway). Strava doesn't let you change the workout type to virtual unless using Sergei's trick which is unsuitable for daily use. When faking GPS data the workout can then be set to non-virtual instead, likewise when uploading to SUUNTO FIT if they have a problem with that. By the way thank you very much for the app!